zigpy/zha-device-handlers #3374: [Device Support Request] IKEA inspelning power monitoring smart plug

gllmlbrt, 2024-09-20:
### Problem description

New IKEA inspelning power monitoring smart plug pairs fine and has all the entities in Hass. However power and voltage are reported divided by factor 10.

### Solution description

An adjustment of the voltage and power (as a result energy calculation) via a quirk? 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>
[image]
[image]

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

cristiangauma, 2024-09-21:
I've just bought this smart plug, and the values appear to be correct on my Home-Assistant instance.

[image]

parrel, 2024-09-21:
I just bought this plug as well, and I have the same problem.

@cristiangauma can you share your firmware version?

saversux, 2024-09-21:
I had the same problem, updated the firmware with ZHA OTA to 0x02040045 and this fixed the issue. It displays the correct voltage now

parrel, 2024-09-21:
@saversux is there a way to force an update?

saversux, 2024-09-21:
If you have Tradfri OTA activated it should display the update in the updates section.

You need to set this in your configurations.yaml

``

zha:
  zigpy_config:
    ota:
      ikea_provider: true

``

parrel, 2024-09-21:
I have that but for some reason it says 0x02040034 is the latest version.

gllmlbrt, 2024-09-21:
> I had the same problem, updated the firmware with ZHA OTA to 0x02040045 and this fixed the issue. It displays the correct voltage now

I read on other threads about Z2MQTT that the new firmware fixes the voltage indeed, but still has the wrong Power. Can you confirm whether it fixes both, by testing a power reading you know for sure ?

saversux, 2024-09-21:
@gllmlbrt just tested it with a USB-C charger that has a display for the power output.
The power output in watts of the plug matches the reported number on the charger.

Additional learning: it seems to push updates very frequent, much faster than Tuya TS011F.

[image]

gllmlbrt, 2024-09-21:
Just got it upgrded to 0x02040045 and it all displays as expected. 
Will clost the issue.

MattWestb, 2024-09-21:
The OTA is **only** in Dirigera feed.
If like updating with local file download this from the index of Dirigera OTA feed:
https://fw.ota.homesmart.ikea.com/files/inspelning-smart-plug-soc_release_prod_v33816645_02579ff4-6fec-42f6-8957-4048def87def.ota

Meta data in the signed file:

```
IKEA of Sweden
INSPELNING Smart lug 
20240801
E2132
www.ikea.com
2.4.45
```

cristiangauma, 2024-09-21:
Sorry, I forgot to confirm that yes, once I added the smart plug to homeassistant, a firmware upgrade notification appeared asking to upgrade to 0x02040045. The screenshot that I shared before was *after* upgrading. Before upgrading I didn't check the entities, as the first thing that I did is to upgrade to the latest firmware.

folfy, 2024-09-22:
THe Z2M community seems to be on to the topic of the bad power reading already, I also noticed that it's off only for loads >=1000W. Seems to be an auto-ranging feature of the plug, see [my comment on Z2M thread](https://github.com/Koenkk/zigbee2mqtt/issues/23961#issuecomment-2366733453).

So it seems we need to check/update the ac_power_divisor on every update of the power-value, and reflect it in the shown value accordingly.

gllmlbrt, 2024-09-22:
Re opened for visibility until the power values are 100% confirmed to be OK.

jasimancas, 2024-10-09:
I have the same problem, I have updated the firwmare and from 22.7V I have gone to 227V, that part corrected (Voltage), but the part of W (Power) when it passes 999V makes an erroneous division.

I know it is reported, but the interpretation error comes from the way the device works? or from the way zigpy manages the data? I mean, it doesn't follow a standard (or at least the same as other sockets).

It's more for knowledge :)

Thanks!!

maceddy, 2024-10-10:
Same issues. Is there a way to update the firmware without Tradfri? I paired directly in ZHA.

MattWestb, 2024-10-10:
Configure OTA updates for ZHA then  you can update all IKEA devices from ZHA.

WCAve, 2024-10-11:
It seems like there is some kind of divisor that needs to be taken into account for measurements above 1000w
https://github.com/Koenkk/zigbee2mqtt/issues/23961
It's fixed in z2m now or so it seems. 

Otherwise, the newest firmware does fix the incorrect voltage reading, but not the readings above 1000w. Seems like it would be as simple as multiplying the power rating by the multiplier that is supplied by the plug.

MattWestb, 2024-10-11:
Need adding binding and reporting to the quirk and  it shall working OK.

JRascagneres, 2024-10-28:
Am also having the 1000W issue. Temporarily worked around with a template. I assume we just need a quirk writing?

jutter88, 2024-11-16:
Is there any progress in solving the power measurement issue in ZHA for 
values >999 watt? Running the latest Inspelning firmware that corrected 
the voltage reading.

Nbird67, 2024-12-04:
Some news about this issue ? (1000W)

BMAN1995, 2024-12-16:
Here is what I found to be a fix for the >1000W issue. Not sure if this is a permanent fix as I have only been using it for a couple of days. A firmware update will most likely revert this and you will need to perform it again. I have not tested this though.

You will need:
- ZHA Toolkit (Can be downloaded from HACS)
- Latest firmware (0x02040045 as I am writing this)

This can be done from within developer tools as well but I stored it in an automation for easy access in the future.

`alias: IKEA INSPELNING Report
description: ""
triggers: []
conditions: []
actions:
  - action: zha_toolkit.conf_report
    metadata: {}
    data:
      ieee: 6c:fd:22:ff:fe:16:c9:cb
      endpoint: 1
      cluster: 2820
      attribute: 1541
      min_interval: 2
      reportable_change: 1
      max_interval: 5
mode: single`

Just create a new automation and go into YAML-editing mode and paste the above. Then go back into the normal editing mode.

You will need to change the IEEE to your specific plug, you can find yours under "Zigbee info" on the plug device page.
After entering the correct IEEE just click on the three dots and hit "Run" (Run might not be correct as I am not using english).

[image]

[image]

MattWestb, 2024-12-16:
It shall working only if the power is going up or down it can being the "old" power factor is being used for the first report but i think we can living with it.

Ping: @puddly and @TheJulianJES do you both have some thinking implanting reporting if this cluster / attribute for IKEA devices so we getting it working OK ?

BMAN1995, 2024-12-16:
> It shall working only if the power is going up or down it can being the "old" power factor is being used for the first report but i think we can living with it.
>
> Ping: @puddly and @TheJulianJES do [...]

I was specifically looking for that during my testing with 5 different plugs, but I never had that issue. You are right that it could happen.

MattWestb, 2024-12-16:
Its only theoretic and have not seen it then sniffing IKEA Dirigera but if its happening its only for one reading **if** its happening.

Bra jobbat !!!

Nbird67, 2024-12-27:
Can this fix be added to a quirk ?

xoferz, 2025-01-04:
Not only is my voltage off, but amps*volts do not make watts:
Current 3.3 A
Power 527 W 
Voltage 22.9 V

220V system, but neither 3.3*229 nor 3.3*22.9 result in 527W.
~500W seems reasonable, it is connected to a battery charger that has 500W on its label.

Peter376, 2025-01-05:
Same >999W issue here. F/W xxx45, newest per Jan 2025. 
Can we assume that IKEA is aware of this issue and that another F/W update is in the works?

folfy, 2025-01-06:
> Not only is my voltage off, but amps*volts do not make watts:

That's because for AC, true power P = U * I * cos(phi), where cos(phi) only ever can be considered almost 1 for resistive loads like a space heater for example. Apparent power S = 229V*3.3A = 756 VA, which would result in a cos(phi) of 0.697, which depending on your load could be very plausible. If you wanna know more, check out [wikipedia](https://en.wikipedia.org/wiki/AC_power#Active,_reactive,_apparent,_and_complex_power_in_sinusoidal_steady-state).

Since we don't know cos(phi) from the plug (some other plugs like the Aqara Smart Plug report this attribute too), we cannot reliably cross-check the values by the plug between each other. We can be pretty sure the voltage should be way higher and the scalar is not up-to-date (even in the US with their lower voltage), and it's possible to say if the values are *likely* off, if cos(phi) < 0.1 at higher loads like 100W+, and also cos(phi) cannot be greater 1 ofc, but for the most part it's still assumptions, i.e. a very hacky solution that will still cause issues for ppl.

So to fix this properly, we have to figure out how to set reporting intervals like @BMAN1995 already showed how to do via ZHA toolkit as a workaround, and implement the same solution in a proper quirk somehow. Unfortunately all this seems way easier in Z2M still, and while the issue also doesn't seem to be completly fixed there, I'm rly considering switching over the way better device support and expandability overall :/

MattWestb, 2025-01-06:
@puddly and @TheJulianJES This is still one issue that no devs have looking on and its one of the most speed devices and its need attention for the devs then normal users is not having the knowledge implanting the reporting and how it shall being done in the best way (very likely IKEA specific for not messing with other devices).
If not having the device ordering online at your IKEA store then its released all over the world and its many users (including me) that can helping testing the solution then its implanted.

Thanks in advance.
MW

MattWestb, 2025-01-06:
By the way **P** of reporting order in your PR.
I think it can also being done in the sequence the binding its being added to the controller (i think new IKEA remotes with device binding is doing it like that).

folfy, 2025-01-07:
@MattWestb @puddly I agree, the order of configuring reporting shouldn't rly matter, it's just values/attributes you're setting on the device (i.e. it doesn't keep track of any order changing them).

@puddly I can try patching my HA / ZHA setup and having a look / provide logs and feedback. Actually I even got a second ZBDongle and IKEA inspelning plug, that I can throw into my HA testing container and provide access for you guys, if it'd help ya.

I'd say min_interval of divisor/multiplier should be in-line / ident with what is set for the actual attributes? I could look into it / try to fetch it with ZHA toolkit during the week.

Nbird67, 2025-01-12:
The ZHA toolkit fix from @BMAN1995  seems to do the trick on my side with min_interval: 2
Should I push min_interval to 0 ?

I tried with 2 Inspelning (dishwasher and dryer) next step will be on a washing machine and cooking oven

maceddy, 2025-01-17:
> Here is what I found to be a fix for the >1000W issue. Not sure if this is a permanent fix as I have only been using it for a couple of days. A firmware update will most likely revert this and you [...]

I get an error on the automation (I installed the toolkit and rebooted HASS)..

The automation "IKEA INSPELNING Report" (automation.ikea_inspelning_report) has an unknown action: zha_toolkit.conf_report.

This error prevents the automation from running correctly.

BMAN1995, 2025-01-18:
> I get an error on the automation (I installed the toolkit and rebooted HASS)..
>
> The automation "IKEA INSPELNING Report" (automation.ikea_inspelning_report) has an unknown action: [...]

You can try running it from Developer Tools instead of inside an automation. 
Under the tab "Actions", search "Configure Attribute Reporting".
When you have filled all the required fields hit "Perform Action".

[image]

maceddy, 2025-01-19:
I feel so stupid. If only I had RTFMed better. I missed the 'add the zha_toolkit: to the configuration.yaml and reboot Home Assistant' bit in the instruction....

The automation works now.

maceddy, 2025-01-21:
> I want to use the automation script but don’t know what must be added to the configuration.yaml to get zha toolkit to work, can someone help what must be included in the configuration?

Literally just this (I put it at the end of my configuration.yaml:

`zha_toolkit:`

[image]

Off course after installing the zha toolkit from HACS.

jutter88, 2025-01-21:
Thanks for the information! 
I have successfully managed to run the automation and now my Inspelning finally shows the correct power consumption if > 999 Watts. Great!
Now I am confidence to buy some more IKEA Inspelning smart plugs.

MiLeX84, 2025-01-27:
> I have that but for some reason it says 0x02040034 is the latest version.

Any luck getting the update to 0x02040045, as mind also still shows 0x02040034 and being up to date.

closed 2025-01-27 by puddly

Nbird67, 2025-01-27:
Closed ? The >1000W problem was resolved ?

TheJulianJES, 2025-01-27:
The bugfix should likely land in Home Assisstant Core 2025.2.0. On that version, the plug needs to be reconfigured once using the UI (or re-paired completely). It *should* work fine then.

folfy, 2025-01-28:
@Nbird67 Yes, I had tested the PR, it's working

abenedykt, 2025-02-05:
Just installed HA Core 2025.2.0 update, pressed that Reconfigure button on a plug entity, and now it reports proper power usage. Awesome job :)

piwy84, 2025-03-13:
sorry

> Just installed HA Core 2025.2.0 update, pressed that Reconfigure button on a plug entity, and now it reports proper power usage. Awesome job :)

sorry, bit of a noob here, where is that 'Reconfigure' button you are referring to? (do you have printscreen maybe)
(same problem here, firmware 0x02040045 on the plugs, voltage ok, power readings (W/ kW) not ok)
Thanks in advance

xoferz, 2025-03-13:
> sorry, bit of a noob here, where is that 'Reconfigure' button you are referring to? (do you have printscreen maybe) (same problem here, firmware 0x02040045 on the plugs, voltage ok, power readings [...]

[image]

piwy84, 2025-03-13:
Thank you @xoferz now let's hope for some correct power readings

benriou, 2025-03-15:
@TheJulianJES 
I just bought few IKEA plugs and I'm having the same issue on 2025.3.3 with ZHA integration : values below 1000w are not shown. 
I haven't applied the ZHA_Toolkit trick. 
Is the fix supposed to be integrated out-of-the-box using 2025.3.3 ?

puddly, 2025-03-15:
@benriou Try to econfigure the device with the "reconfigure" button on the device page.

benriou, 2025-03-15:
just tried to reconfigure : no success.

puddly, 2025-03-15:
Reconfiguration failed (in which case, this is an issue with your network and not that device)? Or the device is not reporting properly loads below 1000W?

MattWestb, 2025-03-15:
First you need let the device being updated before trying getting it working OK with ZHA and all other system  then the factory firmware was not working OK and IKEA was releasing one (more correct more) update for the device for getting it working OK.
The production firmware is 0x02040045 ) 2.4.45.

benriou, 2025-03-15:
My firmware is 0x02040045. 

Reconfiguration is a success but I have no value < 1000 watts.

MattWestb, 2025-03-16:
Then deleting the device and restart HA and adding it new after HA is stable and do one reconfigure then its in the system.
I finding one toaster or weather boiler for testing then its safe and easy letting running for one short time.
Was doing little laundry after breakfast and its looks realistic:

[image]
Standby power is 0.0 - 0,4 W and normal running then heating is 2kW (but the engine is piking very high then starting but not then its running). 
My very expensive EVE energy thread was burning the fuse  the doing log laundry and they was taking it back then it was rated higher but the hard and software protection is not working do the other is now on the refrigerator.

benriou, 2025-03-16:
I did what you've suggested, and...
TADA !

Thank you @MattWestb 

[image]

piwy84, 2025-03-19:
> just tried to reconfigure : no success.

steps i tried: update firmware (unplugging-replugging seemed to initiate that) 
updated HA to most recent version
reconfigure of the plugs in the HA zigbee page
this resulted in readings that, according to me are correct

trankillity, 2025-04-24:
> > I have that but for some reason it says 0x02040034 is the latest version.
> 
> Any luck getting the update to 0x02040045, as mind also still shows 0x02040034 and being up to date.

I am also stuck on `0x02040034` with ZHA/Zigpy saying it's "Up-to-date". Have had OTA configured for ~2 days now and rebooted HA a few times to try and force the update, but no luck.

Due to being stuck on this firmware, the Power values are still reporting incorrectly for > 999W.

MattWestb, 2025-04-24:
> Thanks, but it is configured correctly. If you don't have OTA configured then the Firmware state is just "Unknown". Mine is "Up-to-date".

= the system have one OTA provider for 4476 but its not saying from where is getting the information and if its updated or not.

The response: `manufacturer_ids=(4476,)) was recently contacted, using cached response`  is saying its not online / updated information. 
Did you restarting HA after doing the changes ??

The DIRIGERA OTA file is the right then looking in the data setting in it (at 44C90).
But you need putting in the warning lines for getting the local OTA file working (its not easy getting OK then its truncated and need HA restart if doing changes).
My lines looks like this:
```
        - type: advanced
          warning: I understand I can *destroy* my devices by enabling OTA updates from
            files. Some OTA updates can be mistakenly applied to the wrong
            device, breaking it. I am consciously using this at my own risk.
          path: /config/zigpy_ota
```
I dont knowing if somthing is changed in ZHA but it was working for some months updating new INSPELNING so shall working (thes is out of stock @ WestBhf in Vienna so cant testing).

One way is trying ZHA-toolkit then it can doing more tricks like updating the feed without restarting HA so not using old cashed info.

trankillity, 2025-04-26:
> The response: `manufacturer_ids=(4476,)) was recently contacted, using cached response` is saying its not online / updated information. Did you restarting HA after doing the changes ??

Yes. As mentioned, I have restarted HA multiple times.

> One way is trying ZHA-toolkit then it can doing more tricks like updating the feed without restarting HA so not using old cashed info.

I have just tried using ZHA Toolkit and got a slightly different error.

```
2025-04-27 06:04:23.639 DEBUG (MainThread) [zigpy.ota] Failed to load provider Tradfri(url='https://fw.ota.homesmart.ikea.com/DIRIGERA/version_info.json', manufacturer_ids=(4476,))
  File "/usr/local/lib/python3.13/site-packages/zigpy/ota/__init__.py", line 402, in get_ota_images
  File "/usr/local/lib/python3.13/site-packages/zigpy/util.py", line 501, in replacement
  File "/usr/local/lib/python3.13/site-packages/zigpy/ota/__init__.py", line 377, in _load_provider_index
  File "/usr/local/lib/python3.13/site-packages/zigpy/ota/providers.py", line 201, in load_index
  File "/usr/local/lib/python3.13/site-packages/zigpy/ota/providers.py", line 262, in _load_index
```

I did notice that there was an SSL certificate issue with the `https` address the other day, while the `http` address worked fine. Could this be the thing preventing it from working properly?

puddly, 2025-04-26:
Can you enable ZHA debug logging, click the "check for updates" button in Home Assistant, and then download the ZHA debug log by disabling debug logging?

The update entity will only be "up-to-date" if your device's firmware matches what IKEA's feed contains.

trankillity, 2025-04-27:
Sure thing. [Here's the log file](https://filebin.net/1b7iecz8vsoj0wfg).

Steps I took to generate:
* Enable logging
* Reboot HA
* Press "Check for Updates" under the HA config -> Updates menu
* Wait 15-20 seconds
* Disabled logging

[3 more replies]