zigpy/zha-device-handlers #3376: [BUG] Switch module behaving Differently on ZHA than Z2M (K4411C)

jafar-atili, 2024-09-20:
### Bug description

I have moved recently from Z2M to ZHA and I noticed different behavior of a dimmer switch module.

I have a [Bticino K4411C](https://www.zigbee2mqtt.io/devices/4411C_L4411C_N4411C_NT4411C.html) dimmer switch, When it was paired with z2m, I could turn off or on the light from any side of the switch (physically)

With ZHA, the light (device) ONLY turn on from the bottom side of the switch and turn off from the upper side of the switch.

How could the device get configured differently on Z2M than on ZHA? How can I fix it (I prefer Z2M behavior).

### Steps to reproduce

Nothing special, just pair the device on ZHA

### Expected behavior

The device should be toggled on both side of the switch.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

shimon88, 2025-01-21:
same here, really annoying. of course we could just install the switch u side down but if it can be done sw side it is better

jafar-atili, 2025-01-22:
Also when opening the zigbee network the devices led turns to pink but with z2m its green

Seems that devices can distinguish between Z2M and ZHA coordinator and behave differently.

I saw some developer statement on the legrand website mentioning that this (no toggle) behavior is set by the device if it recognize a 3rd party coordinator.

closed 2025-08-10 by github-actions[bot]