zigpy/zha-device-handlers #3377: [Device Support Request] Problem with TRV801Z - Tuya radiator thermostats

joe2002, 2024-09-21:
### Problem description

Hello Tuya community, I have a problem controlling Tuya radiator thermostats via Home Assistant. It is a TRV801Z which is sold by Aliexpress, Ecobuy there. Is there already a corresponding configuration for this? The function of setting the temperature is possible. But everything else, like setting it to "Auto", "Off", "On" does not work.Thank you very much for your help.

Hallo Tuya Gemeinde, ich habe ein Problem bei der Steuerung von Tuya Heizkörperthermostate über Home Assistant. Es handelt sich um TRV801Z die von Aliexpress, dort Ecobuy verkauft werden. Gibt es dafür bereits eine entsprechende Konfiguration? Die Funktion von Temperatur setzten ist möglich. Aber alles andere, wie stellen auf "Auto", "Aus", "An" gehen nicht. Vielen Dank für eure Hilfe 

### Solution description

I would like to be able to control the thermostat completely via automation and/or button in Home Assistant.

Ich würde das Thermostat gerne vollständig per Automation und/oder Button in Home Assistant steuern können.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

tco99ttocs, 2024-11-15:
would appreciate if this device would be added:)

pro-jul, 2024-11-15:
Would also help me a lot!

copernic-us, 2024-11-16:
There's an external converter for it. Check it on:
https://github.com/DonaldChung-HK/z2m-moes-trv-801-converter/blob/main/moes_valve_ZTRV801.js

tco99ttocs, 2024-11-16:
the question is now, how could this convertred to an tha compatible quirk?!

pro-jul, 2024-11-25:
Hi guys,
Thank you very much for the tip. I followed the instructions and also managed to successfully connect the device to Zigbee2mqtt.
Unfortunately the only settings I can change via the user interface are “display_brightness” and “screen_orientation”. I also don't get any sensor values or current settings, neither in zigbee2mqtt nor in mqtt. No different result even after repeated re-pairing.
I suspect that the configuration file does not match the thermostat? (vendor name is: "_TZE204_qyr2m29i")

Thank you.

gielk, 2025-02-03:
I have the Moes TRV801Z device integrated into my Zigbee2MQTT setup. I purchased it via AliExpress, expecting it to be supported by ZHA. However, I discovered that it isn’t currently compatible. I would prefer to manage this device through ZHA to streamline my setup. If anyone has experience or knowledge about integrating this device into ZHA, your assistance would be greatly appreciated. I’m also willing to provide data or logs from my Zigbee2MQTT configuration to aid in developing support for this device in ZHA.

prairiesnpr, 2025-02-17:
Please try using the quirk for `_TZE204_rtrmfadk`, it looks to be the same.

closed 2025-08-25 by github-actions[bot]