zigpy/zha-device-handlers #3382: [BUG] Paulmann 501.34 Wall Switch Button 2 and 4 long presses do not work

itsmicash, 2024-09-23:
### Bug description

@redfusion83 @drueppela @MattFromGer @TheJulianJES 

coming from this thread: [Issue](https://github.com/zigpy/zha-device-handlers/issues/2662)

Hi guys,

I have issues with button A0 and B0 (or button 2 and 4)
all button presses work except long press and press and realease on these 2 buttons. hope someone can support me.

Button 2 long press & release
[image]

Button 2 long press
[image]

Button 4 long press & release
[image]

Button 4 long press
[image]

I'm really struggling and don't know what to do. 

### Steps to reproduce

123

### Expected behavior

123

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

thwgmx, 2024-10-07:
Same problem Herr…

gexxx05, 2024-12-14:
Yeah, i have the same problem, but you can create a custom trigger as a workaround for this:

1. In Home Assistant go to Developer Tools and Events, listen for Event "zha_event" and make a long press on the off button you want. After the press, stop listening and save the two results. One has the command "move" (Long press) and one has the command "stop" (Realese after long press).

2. Copy all the content from data section and remove the unique_id part.
[image]

3. In your automation create a custom trigger with event type "zha_event" and copy the data from above.
[image]

4. Now you should be able to dimm your lights.

itsmicash, 2024-12-14:
@gexxx05 
thanks for the hint. in the meanwhile i migrated to z2mqtt, which does the job much better for the paulmann switch and in general for all my other devices. I hope your suggestion helps others with that issue.

closed 2025-08-10 by github-actions[bot]