zigpy/zha-device-handlers #3384: [Device Support Request] Tuya _TZ321C_fkzihax8 presence sensor

NH-Networks, 2024-09-23:
### Problem description

Not supported, only motion is shown but not working 

{"id":2,"type":"Router","ieeeAddr":"0xa4c1385406d5e8f6","nwkAddr":4964,"manufId":4417,"manufName":"_TZ321C_fkzihax8","powerSource":"Mains (single phase)","modelId":"TS0225","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,4,5,1280],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0225","manufacturerName":"_TZ321C_fkzihax8","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":1,"hwVersion":1,"dateCode":""}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0024cb3e7a","zoneState":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":65,"stackVersion":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-1401274594},"lastSeen":1725843436923}

### Solution description

See link for working Z2MQTT converter 

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

https://github.com/Koenkk/zigbee2mqtt/issues/23913

kaischmerer, 2024-10-24:
Try this one: https://github.com/ExcelluxZ2MZHA/Excellux-ZigBee2mqtt/blob/main/mtd085_z2m1.4.0.js

bveina, 2024-11-08:
i also am trying to integrate this sensor into HA.

My signature for this device is:

```
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 74,
    "maximum_incoming_transfer_size": 80,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 80,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0500"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ321C_fkzihax8",
  "model": "TS0225",
  "class": "zigpy.device.Device"
}```

i didnt think that zigbee2mqtt files could be drop in replacements for a zha quirk. i did put [mtd085_z2m1.4.0.js](https://github.com/ExcelluxZ2MZHA/Excellux-ZigBee2mqtt/blob/main/mtd085_z2m1.4.0.js) in my zha quirk folder but it still did not work. 

any help would be appreciated.

yaminahacs, 2024-11-20:
What is the progress on this?

supersonical, 2025-05-05:
i have also bought a TZ321C WENZH, which is not supported. Would be interesting if works bc for the pricepoint its really good price.

darklion84, 2025-11-04:
Any progress guys?

chilloutman, 2025-12-18:
I have no idea what I'm doing with this stuff, but I did try to convert that `.js` for zigbee2mqtt into a quirk. It does show a bunch of attributes on the device page now, but I can't get any updates from the device. I don't know if the device needs to be enrolled/onboarded first for it to start sending updates, or if the updates are failing...

```
from enum import Enum

import zigpy.types as t
from zigpy.zcl.clusters.measurement import OccupancySensing

from zigpy.quirks.v2 import EntityType
from zigpy.quirks.v2.homeassistant import UnitOfLength, UnitOfTime
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass

from zhaquirks.tuya.tuya_motion import TuyaOccupancySensing
from zhaquirks.tuya.builder import TuyaQuirkBuilder

class DebugMode(t.enum8):
    Off = 0
    On = 1

class ScenePreset(t.enum8):
    Custom = 0
    Toilet = 1
    Kitchen = 2
    Hallway = 3
    Bedroom = 4
    LivingRoom = 5
    MeetingRoom = 6
    FactoryDefault = 7

class SensorState(t.enum8):
    On = 0
    Occupied = 1
    Unoccupied = 2

class StatusIndication(t.enum8):
    Off = 0
    On = 1

class DistanceReportMode(t.enum8):
    Normal = 0
    OccupancyDetection = 1

(
    TuyaQuirkBuilder("_TZ321C_fkzihax8", "TS0225")
    # Occupancy mapping (adjust dp_id if your device uses 112 for sensor state)
    .tuya_dp(
        dp_id=112,
        ep_attribute=TuyaOccupancySensing.ep_attribute,
        attribute_name=OccupancySensing.AttributeDefs.occupancy.name,
        converter=lambda x: x == 1,
    )
    .adds(TuyaOccupancySensing)

    # --- Numeric datapoints ---
    .tuya_number(
        dp_id=101,
        attribute_name="entry_sensitivity",
        type=t.uint16_t,
        min_value=10,
        max_value=100,
        step=10,
        unit="%",
        translation_key="entry_sensitivity",
        fallback_name="Entry sensitivity",
    )
    .tuya_number(
        dp_id=102,
        attribute_name="entry_distance",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DISTANCE,
[... 144 lines cut ...]
        dp_id=111,
        attribute_name="scene_preset",
        enum_class=ScenePreset,
        translation_key="scene_preset",
        fallback_name="Scene preset",
    )
    .tuya_enum(
        dp_id=114,
        attribute_name="status_indication",
        enum_class=StatusIndication,
        translation_key="status_indication",
        fallback_name="Status indication",
    )
    .tuya_enum(
        dp_id=120,
        attribute_name="distance_report_mode",
        enum_class=DistanceReportMode,
        translation_key="distance_report_mode",
        fallback_name="Distance report mode",
    )

    .skip_configuration()
    .add_to_registry()
)
```