zigpy/zha-device-handlers #3385: [BUG] smart plugs polling frequency too high

Apfelstruhdl, 2024-09-24:
### Bug description

Hi,

I have a bigger ZHA network running with about 100 devices and multiple smart plugs that i hoped to be able to use as routers and track the power consumption around the house.

I fear, that the network traffic the zigbee network is almost at a maximum (reports of ~85% utilization occured multiple times). I already moved the wifi away from the zigbee channel.
Besides the power report on the zigbee channel I find it hard with the standard tools to get a feeling for the traffic that is going on in the network and to distinguish actual traffic that the network generated from noise on the channel.
Is there anyway to just monitor the message or data rate of the coordinator?

Back to the routers. They are
14x TS011F by _TZ3000_gjnozsaz
3x TS011F by _TZ3000_w0qqde0g

both are using the quirk
Quirk: zhaquirks.tuya.ts011f_plug.Plug

So far I understood, that there are push messages by the devices and I think in this case it is cluster 0x0702 for which I can also read and set the respective parameters that influence the pushing of messages by the device:
min_interval: 60
max_interval: 900
reportable_change: 10
tries: 3

this seems to be only energy (kWh delivered).

The cluster 0x0b04 seems to provide the values for voltage, current and power. There is no setting for the reporting (push) interval, so I guess that these values are collected by issuing pull request, correct?

How can I influence the frequency of these pull messages? I really would love to better understand and configure the traffic on the network. Or do I misunderstand sth?

Thanks a lot for your help!

### Steps to reproduce

Check the polling interval of the above mentioned smart plugs and try to change the frequency.

### Expected behavior

Option to alter the polling interval per device or globally.

A monitor that reports the overall traffic and traffic by device

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

Apfelstruhdl, 2024-11-11:
I really think that this is an issue. Why is there no option to configure the polling interval. Can somebody point me to the source code, that controls the behavior? I tried going through the classes but it was not very intuitive where the actual polling is implemented.

I have a smart card, that sorts all zigbee devices based on the last seen timestamp. This list is the only easy way I found to get a feeling for the number of messages that are on the network. an indeed the smart plugs are updated in intervals of seconds and are thus very likely causing the overall network to get jammed.

atmezferix, 2025-01-08:
I too am facing exactly the same issue my network is just saturated with updates from Smartplugs, is this Home assistant that is constantly polling these devices for updates in the Energy dashboard or Utiliity Meters, there must be somewhat to control this?
I see that Zigbee2Mqtt has a reporting screen where these updates can be configures, why is there nothing in ZHA ?

I see there is (from what I can tell) and almost undocumented section.. ie. System Options to Allow polling or new devices to be automatically added.  Disabled Polling there made no difference at all. so not sure what they does.

I have a total of 108 devices 17 of them are TS011F and there are other smartplugs that are the same.

I'm using an SMlight SLZB06p7 Dongle and have all sorts of issues, buffer_full is frequently seen in the logs and a dongle restart or  HA restart is my only option to resolve.  The system is just not reliable at all and is frequently having all zigbee devices stop responding until I take action.

Apfelstruhdl, 2025-01-14:
I think, that many users are affected by this without knowing it, as there are no indicators accessible to identify the cause of unreliable zigbee networks. Using the power plugs as a backbone routing infrastructure is a choice that probably many thought of.

I really think, that this should get attention. The solution is probably also not that hard as you would only need to make the time variable of the polling interface configurable. If this is a global implementation or specific to the plug, I couldn't figure this out from the source code.

Staartvin, 2026-02-02:
I'm also suffering from this issue. I manually adjusted the min. and max. reporting intervals of some of my devices using zha-toolkit, but these values don't seem to persist across a reboot. It looks after a HA reboot, they go back to reporting every few seconds.

Apfelstruhdl, 2026-10-09:
Follow-up with measurements from my network (HA 2026.5.4, `zha` 1.3.1, zhaquirks 1.2.0, EmberZNet 8.0.3 / ZBDongle-E, 120 devices). 19× TS011F (`_TZ3000_gjnozsaz`, `_TZ3000_w0qqde0g`), quirk `zhaquirks.tuya.ts011f_plug.Plug`.

**These plugs report everything on their own, so the polling is redundant, not just extra traffic.**

`zigpy.zcl` debug log, 10 minutes (22:25–22:35):

| | count / 10 min |
|---|---:|
| Requests sent by the coordinator (all devices) | 714 |
| … of which `Read_Attributes` to TS011F (0x0B04 + 0x0702 polls) | **622 (87 %)** |
| Unsolicited `Report_Attributes` from the TS011F plugs: | |
| 0x0B04 `rms_current` (0x0508) | 605 |
| 0x0B04 `active_power` (0x050B) | 417 |
| 0x0B04 `rms_voltage` (0x0505) | 43 |
| 0x0702 `current_summ_delivered` (0x0000) | 50 |

So every attribute that `PolledElectricalMeasurement` and `PolledSmartEnergySummation` read every 30–45 s already arrives via attribute reports, for power and current every few seconds.

Where the polling comes from (zha 1.3.1, `zha/application/platforms/sensor/__init__.py`):
- `PolledElectricalMeasurement` matches every EM cluster except the models in `ReportingElectricalMeasurement` (`VZM31-SN`, `SP 234`, `outletv4`, `INSPELNING Smart plug`).
- `PolledSmartEnergySummation` explicitly matches `models={"TS011F", ...}`.

There is no user-facing setting. The only workaround is disabling the Power / Summation entities, and I can't do that for summation because it feeds the Energy dashboard. I've now disabled Power on 17 of 18 plugs. Current and voltage keep updating from reports. Coordinator APS unicast TX (EZSP counters, 5-min windows) went from ~94/min to **~55/min**. Part of that is the network settling after a restart, so treat it as an upper bound.

**Correction to my original report:** the "~85 % utilization" I quoted came from the ZHA energy scan. On the network's own channel that mostly measures our own traffic during the scan, not airtime usage. The better indicator, `PHY_CCA_FAIL_COUNT`, was only ~0.3 % of MAC transmissions (est. < 10 % airtime), so the channel was not saturated. The instability I saw (Tuya devices not responding after coordinator restarts) turned out to be a routing problem (`ZIGBEE_DELIVERY_FAILED` while their reports still arrived). Enabling `source_routing` fixed it. Polling probably makes that phase worse: every poll to an unreachable plug can trigger a route discovery broadcast, and route discoveries were ~5× higher right after restarts. But polling was not the root cause. The point of this issue is that the polling is redundant.

**Suggestion:** treat TS011F plugs that report (these two manufacturers at least) like `ReportingElectricalMeasurement`, and drop `TS011F` from `PolledSmartEnergySummation`. Alternatively, have the quirk expose a feature flag that opts out of EM/summation polling. Happy to test a PR or provide full logs and diagnostics.

aarcange, 2026-11-08:
I did solve the flood by disabling the measurement some time ago. I was looking if there was a better solution.

```
--- ts011f_plug_slow_reporting.py
+++ ../ts011f_plug.py
@@ -1087,8 +1087,8 @@
                     Groups.cluster_id,
                     Scenes.cluster_id,
                     TuyaZBOnOffAttributeCluster,
-                    #TuyaZBMeteringClusterWithUnit,
-                    #TuyaZBElectricalMeasurement,
+                    TuyaZBMeteringClusterWithUnit,
+                    TuyaZBElectricalMeasurement,
                     TuyaZBExternalSwitchTypeCluster,
                 ],
                 OUTPUT_CLUSTERS: [],
@@ -1268,8 +1268,8 @@
                     Scenes.cluster_id,
                     TuyaZBOnOffAttributeCluster,
                     Time.cluster_id,
-                    #TuyaZBMeteringClusterWithUnit,
-                    #TuyaZBElectricalMeasurement,
+                    TuyaZBMeteringClusterWithUnit,
+                    TuyaZBElectricalMeasurement,
                     LightLink.cluster_id,
                     TuyaZB1888Cluster,
                     TuyaZBE000Cluster,
```