zigpy/zha-device-handlers #3388: [Device Support Request] Avatto TRV16 Smart Radiator Thermostat

ulf78, 2024-09-25:
### Problem description

I just bought an Avatto TRV16 thermostat. I was able to connect it to ZHA, but it does not show any controls.

### Solution description

I would like to see this device being supported by zigpy/ZHA with reading and setting heating parameters.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_rxntag7i",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
 "data": {
    "ieee": "**REDACTED**",
    "nwk": 15646,
    "manufacturer": "_TZE200_rxntag7i",
    "model": "TS0601",
    "name": "_TZE200_rxntag7i TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,
    "manufacturer_code": 4417,
    "power_source": "Battery or Unknown",
    "lqi": null,
    "rssi": null,
    "last_seen": "2024-09-25T07:37:07",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": {
        "logical_type": 2,
        "complex_descriptor_available": 0,
        "user_descriptor_available": 0,
        "reserved": 0,
        "aps_flags": 0,
        "frequency_band": 8,
        "mac_capability_flags": 128,
        "manufacturer_code": 4417,
        "maximum_buffer_size": 66,
        "maximum_incoming_transfer_size": 66,
        "server_mask": 10752,
        "maximum_outgoing_transfer_size": 66,
        "descriptor_capability_field": 0
      },
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZE200_rxntag7i",
      "model": "TS0601"
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "update.tze200_rxntag7i_ts0601_firmware",
        "name": "_TZE200_rxntag7i TS0601"
      }
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

ManuelYuberoDelgado, 2024-11-19:
Please, I also need this device to be supported by HA, would it be possible to add it?
Thank you very much in advance.

GrahamWeatherup, 2024-11-25:
I need this too and am new to home assistant. It's taken me a year to get round to setting this up I also have the TRV16 but as it's also listed as TS0601 albeit with _TZE200_bvu2wnxz instead of _TZE200_rxntag7i but as they are both Avatto I'm hoping it's the same.. enough.

Did you see that [Zigbee Device Compatibility Repository - Avatto_TRV06](https://zigbee.blakadder.com/Avatto_TRV06.html) points to [Custom Quirk](https://github.com/zigpy/zha-device-handlers/issues/1818)? It seems to have a lot of information about creating and editing quirks too. I have no experience in doing this yet or even installing a quirk but it appears to be a good place to start.

closed 2025-01-16 by TheJulianJES