zigpy/zha-device-handlers #3390: [Device Support Request] TS0601 _TZE204_ya4ft0w4

Margravegit, 2024-09-28:
### Problem description

This is a ZX-M100-24G motion sensor (v1.2)
The device appears in ZHA but only the status of the firmware is shown, not any of the sensors or configuration. There are a few quirks available for different versions of the same device, some of the controls work but haven't managed to get everything working.

### Solution description

Add support for the device in ZHA.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_ya4ft0w4",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>
[... 624 lines cut ...]
      }
    }
  }
}
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

thiagoanegreiros, 2025-01-03:
I have the same issue. There was another thread opened here
https://github.com/zigpy/zha-device-handlers/issues/3359

But I don't know why it got closed because the issue still persists.

Somebody said there this
I got mine working at a bare minimum by using the following instructions and the below quirk. It doesn't support changing the modes or settings but it does report presence.

https://fixtse.com/blog/zy-m100-full-zha-support

But it does not work properly for this sensor as it stays on the "detected" forever sometimes... maybe this is a starting point

prairiesnpr, 2025-01-03:
> But I don't know why it got closed because the issue still persists.

A quirk was merged and is available in HA 2025.1

thiagoanegreiros, 2025-01-03:
But it does not cover _TZE204_ya4ft0w4

prairiesnpr, 2025-01-03:
You are correct, it does cover `_TZE200_ya4ft0w4`, have you tried adding `_TZE204_ya4ft0w4` to the existing quirk?

thiagoanegreiros, 2025-01-03:
Yes! I added but it did not work as well =[

prairiesnpr, 2025-01-03:
> Yes! I added but it did not work as well =[

You tried this one? https://github.com/zigpy/zha-device-handlers/blob/812441104427e74e13a8168ea916597f87534d72/zhaquirks/tuya/ts0601_motion.py#L127

z2m has then both using the exact same converter.

thiagoanegreiros, 2025-01-03:
I need to copy this file into my custom quirks and add on line 128?

prairiesnpr, 2025-01-03:
Yes, but you need to add this at the top.

```python

from zigpy.quirks import DEVICE_REGISTRY
# other existing imports from custom quirks

# remove upstream quirk from registry
del DEVICE_REGISTRY._registry_v2[("_TZE204_ya4ft0w4" , "TS0601")]

# add custom quirk...

```

thiagoanegreiros, 2025-01-03:
Ok... I just replaced _TZE200_ya4ft0w4 to _TZE204_ya4ft0w4 and it seems to
work. I will monitor the behavior just to make sure because it seemed to me
that it was giving false positives.

thiagoanegreiros, 2025-01-04:
Hi! As I told before this is throwing some false positives

Since 8 there is no one home. There are two places eith this same sensor.

One is working fine. The one that is with false positives is on a hot place with big natural light. Maybe the issue is the heat?

[image]

prairiesnpr, 2025-01-04:
> One is working fine. The one that is with false positives is on a hot place with big natural light. Maybe the issue is the heat?

If you have one working and the other not, that seems like a device or an environmental issue. The quirk is just going to update occupancy based on what the device reports.

mi3lix9, 2025-01-04:
Hello guys, 
I am new in Zigbee, actually, I got shucked when it wasn't working at first 😂
Could you please help with a tutorial or something so I can run my sensor? 
TS0601
by _TZE200_ya4ft0w4

closed 2025-01-04 by TheJulianJES

TheJulianJES, 2025-01-05:
This should work with Home Assistant Core 2025.1.1 (or later).

Ewyckoff, 2025-01-12:
This worked for me thanks for this. I did the same thing as thiagoanegreiros and just changed the 200 to 204 nothing else. What is that other part you were saying needs to be added to the top ? should I ignore that because mine is working already

> Yes, but you need to add this at the top.
> 
> ```python
> from zigpy.quirks import DEVICE_REGISTRY
> # other existing imports from custom quirks
> 
> # remove upstream quirk from registry
> del DEVICE_REGISTRY._registry_v2[("_TZE200_ya4ft0w4" , "TS0601")]
> 
> # add custom quirk...
> ```
> 
> This assumes you keep the 200, if you just change 200 to 204 such that there isn't a built in quirk replaced by a custom quirk, you can ignore the above.

prairiesnpr, 2025-01-12:
This should be part of the current HA release. No custom quirk should be required.

stefvdw, 2025-01-23:
I just receiver my sensor and so far it works great in ZHA. My only issue is that binding this sensor to a group does not work for me. I see no cluster to bind to the group in the interface:

[image]

Ewyckoff, 2025-01-23:
> I just receiver my sensor and so far it works great in ZHA. My only issue is that binding this sensor to a group does not work for me. I see no cluster to bind to the group in the interface:
> 
> [image]

Try to re add it maybe.  Not sure what's going on there but I have 3

[image]