zigpy/zha-device-handlers #3402: [Device Support Request] TS0601 by _TZE204_s139roas Miucda/Avatto temp sensor

ArturZurawski, 2024-10-07:
### Problem description

I bought [Miucda/Avatto ZigBee E-ink temperature and humidity sensor](https://aliexpress.com/item/1005007336893299.html). I was able to connect it to ZHA, but it does not show any readings.

### Solution description

I would like to be able to read temp, humidity and battery level from this device

### Screenshots/Video

<details><summary>Screenshots/Video</summary>
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
    "signature": {
      "node_descriptor": {
        "logical_type": 2,
        "complex_descriptor_available": 0,
        "user_descriptor_available": 0,
        "reserved": 0,
        "aps_flags": 0,
        "frequency_band": 8,
        "mac_capability_flags": 128,
        "manufacturer_code": 4417,
        "maximum_buffer_size": 66,
        "maximum_incoming_transfer_size": 66,
        "server_mask": 10752,
        "maximum_outgoing_transfer_size": 66,
        "descriptor_capability_field": 0
      },
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZE204_s139roas",
      "model": "TS0601"
    },
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

pflorek, 2024-11-10:
Hi @ArturZurawski , I'm triaging them too. Please test the quirk also #3503

ArturZurawski, 2024-11-12:
Seems to be working. Thank you.

PR #3503 (2024-11-24): Add Tuya temp/humidity sensor `_TZE204_s139roas` variant
variant has e-ink screen and cr2430 battery. seems to work like the variant without screen. tested by copying to zha.custom_quirks_path

closes #3402

```python
"""Tuya temp and humidity sensors."""

from zhaquirks.tuya.builder import TuyaQuirkBuilder

(
    TuyaQuirkBuilder("_TZE204_s139roas", "TS0601")
    .tuya_temperature(dp_id=1, scale=10)
    .tuya_humidity(dp_id=2)
    .tuya_battery(dp_id=4)
    .skip_configuration()
    .add_to_registry()
)

```

## Proposed change

## Additional information

## Checklist

- [x] The changes are t [... cut]
closed 2024-11-12 by ArturZurawski