zigpy/zha-device-handlers #3416: [Device Support Request] TS0601 TZE204_mwomyz5n (BEOK TGM50-ZB)

mkozlina, 2024-10-11:
### Problem description

I bought BEOK TGM50-ZB wall thermostat but it is not recognised by HA. The device is listed as [supported on Zigbee2MQTT](https://www.zigbee2mqtt.io/devices/TGM50-ZB.html). I believe it needs a quirk but I don't know where to start. I am an experienced programmer so writing a quirk shouldn't be a problem but I would appreciate some guidelines. 

### Solution description

Full support for the device in HA.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_mwomyz5n",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information
[... 595 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

hdh75, 2024-10-13:
Hi @mkozlina, happy to see someone else is also investigating that.

I've attempted adding this device to `class Beok(EnchantedDevice, TuyaThermostat)` in https://github.com/jacekk015/zha_quirks/blob/main/ts0601_thermostat_avatto.py

After the quirk is applied, I see these entities in HA:
- a sensor with no value
- two configuration number entities (min and max temperature) that don't have any effect on the device
- a switch that does not do anything
- a number entity (temp offset) without effect
- a climate entity that does not show any values. Here I can toggle 'heating' and 'off' which turns on or off the device's display

maybe this points in the right direction

mkozlina, 2024-10-16:
@hdh75 great! I'll give it a try today afternoon. Tnx! I was able to get in contact with Beok support and I asked them to provide cluster specification, if possible.

mkozlina, 2024-10-24:
@hdh75 I finally found time for this. The quirk works fine except 'heat_state'. It needs to be inverted. I also cleaned and renamed it. 
[beok_thermostat.zip](https://github.com/user-attachments/files/17502376/beok_thermostat.zip)

Marco02Arduino, 2024-10-24:
Hi, I'm also trying to use some beok smart thermostat with ZHA, it's all working fine, but the temperature controls are reported with fahrenheit, beeing european I would prefer celsius degree as my primary temperature configuration, there's a way to change that?
PS: Thank you for the quirk, when I ordered the thermostat I assumed I would use Z2M to control them, but being new to the HA world I prefer ZHA which seems simpler

mkozlina, 2024-12-02:
I am closing this because the quirk works fine. When I find time I will look into adding it to the official list.

closed 2024-12-02 by mkozlina

bagbyte, 2024-12-15:
@mkozlina @hdh75 I'm currently using this quirk: https://github.com/zigpy/zha-device-handlers/issues/3416#issuecomment-2434427006

If I turn it off, the floor keeps being warm, is that normal?
I've tried to set the temp to 15 (the min) and to turn it off as well, but the morning after the floor was still warm...

I'm afraid during summer the thermostat will keep working even if I will turn it off....is there a command to verify the current state of the valve?

Jerseyman99, 2024-12-22:
I have a Samotech Thermostat model TGM-50 which identifies as a  _TZE204_mwomyz5n I have installed the BEOK quirk, which exposes most of the entities that I need , alas the recorded temperature is inverted, ie the hotter the floor gets the colder the temperature recorded is. @mkozlina reported this as an issue but from later post I assumed it has been fixed. Has anyone got any advice on this or can point me to different quirk?

tux1337, 2024-12-23:
I can confirm that all the main features are working with the quirk in this thread. Thank you for that. 

The features like setting the backlight / anti frost / child lock / factory reset are somehow inside of the quirk but not exposed in Homeassistant. Can you give me a hint what is needed to access this features? Zigbee2MQTT describe them here: https://www.zigbee2mqtt.io/devices/TGM50-ZB.html

tux1337, 2024-12-23:
> I'm afraid during summer the thermostat will keep working even if I will turn it off....is there a command to verify the current state of the valve?

You should be able to see when the devices changed to heating mode in HA. There is a log in the zigbee device view.

mkozlina, 2024-12-26:
> @mkozlina @hdh75 I'm currently using this quirk: [#3416 (comment)](https://github.com/zigpy/zha-device-handlers/issues/3416#issuecomment-2434427006)
> 
> If I turn it off, the floor keeps being warm, is that normal? I've tried to set the temp to 15 (the min) and to turn it off as well, but the morning after the floor was still warm...
> 
> I'm afraid during summer the thermostat will keep working even if I will turn it off....is there a command to verify the current state of the valve?

@bagbyte, floor heating is a very slow process and it can take some time for it to cool down or heat up. If you want to be 100% sure if the thermostat is off, measure voltage on the valve output.

mkozlina, 2024-12-26:
> I have a Samotech Thermostat model TGM-50 which identifies as a _TZE204_mwomyz5n I have installed the BEOK quirk, which exposes most of the entities that I need , alas the recorded temperature is [...]

@Jerseyman99 it seems to me that you have connected the valve to the wrong output. The thermostat that I have, has NO and NC contacts and my valve needs to be connected to the NO contact.

mkozlina, 2024-12-26:
> I can confirm that all the main features are working with the quirk in this thread. Thank you for that.
>
> The features like setting the backlight / anti frost / child lock / factory reset are somehow [...]

@tux1337 I didn't bother with exposing these in HA directly. You can access them via the 'Manage Zigbee device' window:
[image]

Jerseyman99, 2024-12-26:
> @Jerseyman99 it seems to me that you have connected the valve to the wrong output. The thermostat that I have, has NO and NC contacts and my valve needs to be connected to the NO contact.

Thanks for the response Marko, however the thermostat is for Electric Underfloor Heating, so no valves just a sensor

tux1337, 2024-12-26:
> @tux1337 I didn't bother with exposing these in HA directly. You can access them via the 'Manage Zigbee device' window …

super nice. Thanks a lot for the hint. 
Having the backlight controllable via UI would be nice. Then I could use an automation for it. My problem is that I need to turn off the light during the night in sleeping room but want to turn it on in the morning again. 
That’s why I was searching for the controls.

bagbyte, 2024-12-29:
> @bagbyte, floor heating is a very slow process and it can take some time for it to cool down or heat up. If you want to be 100% sure if the thermostat is off, measure voltage on the valve output.

@mkozlina is there a way to read the voltage from HA? Or do I need to check with a voltmeter?

oarchi1957, 2024-12-30:
le quirk:  book_thermostat.py a bien fonctionné pour moi ! Merci @hd75 !

wranglatang, 2024-12-30:
Thanks so much for shaing this configuration! I just installed a TGR85-ZB and connected up to Home Assistant via ZHA and has no entities created despite it being reported as supported on [the website ](https://zigbee.blakadder.com/Beok_TGR85-ZB.html). I have installed your beok_thermostat.py on my HA and now i have the entities for the climate and i can control it!! Thank you @mkozlina! 

Two things I've noticed, is that some entities dont work, I have a switch that I'm not sure what it does and then there seems to be a max and minimum entity to set a threshold which dont seem to work. I've never setup a custom quirk so not sure where to start.

tux1337, 2024-12-30:
> Two things I've noticed, is that some entities dont work, I have a switch that I'm not sure what it does and then there seems to be a max and minimum entity to set a threshold which dont seem to work. I've never setup a custom quirk so not sure where to start.

I was also struggling with that. The Switch is for enabling/disabling the child lock.

kickartz, 2025-01-05:
> @tux1337 I didn't bother with exposing these in HA directly. You can access them via the 'Manage Zigbee device' window: [...]

Hi everyone,

I’m using the same thermostat, and the quirk is working well overall. However, I’m struggling with toggling the backlight. I couldn’t find any relevant attributes or options to adjust it. 

In addition, I don’t see the "Manage Zigbee Device" section, where you can view Clusters, bindings, and so on, like in the reference picture.

One thing to note is that my Home Assistant instance is set to German. Could this be affecting the visibility of this section or its functionality?
Could someone kindly provide a step-by-step guide on how to control the backlight? Ideally, I’d like to automate it so the backlight turns off in the bedroom between 10 PM and 8 AM.

Thanks in advance for your help! 😊

mkozlina, 2025-01-05:
> super nice. Thanks a lot for the hint. Having the backlight controllable via UI would be nice. Then I could use an automation for it. My problem is that I need to turn off the light during the night [...]

I have the same problem and backlight control wouldn't help because even at the lowest setting it's too bright. My plan is to create an automation that will switch off the thermostat in the evening and then switch it back on when the temperature drops below the setpoint or in the morning.

mkozlina, 2025-01-05:
> > @bagbyte, floor heating is a very slow process and it can take some time for it to cool down or heat up. If you want to be 100% sure if the thermostat is off, measure voltage on the valve output.
> 
> @mkozlina is there a way to read the voltage from HA? Or do I need to check with a voltmeter?

No, you need to check with voltmeter. You can also use an electric tester in a form of a screwdriver.

mkozlina, 2025-01-05:
> Hi everyone,
>
> I’m using the same thermostat, and the quirk is working well overall. However, I’m struggling with toggling the backlight. I couldn’t find any relevant attributes or options to adjust [...]

The "Manage Zigbee Device" option is on the device menu:
[image]

kickartz, 2025-01-06:
Thank you, I've found it. 

However, I still can't control the brightness. I suppose I'll switch to z2m.

Turning it off at night is a workaround, but no solution.

kickartz, 2025-01-06:
> > > > I can confirm that all the main features are working with the quirk in this thread. Thank you for that.
> > > > The features like setting the backlight / anti frost / child lock / factory reset are somehow inside of the quirk but not exposed in Homeassistant. Can you give me a hint what is needed to access this features? Zigbee2MQTT describe them here: https://www.zigbee2mqtt.io/devices/TGM50-ZB.html
> > > 
> > > 
> > > @tux1337 I didn't bother with exposing these in HA directly. You can access them via the 'Manage Zigbee device' window: ![image](https://private-user-images.githubusercontent.com/13692876/398724915-18d8db6e-d0d5-4def-9a42-e38aaa29af42.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYwOTk4MDQsIm5iZiI6MTczNjA5OTUwNCwicGF0aCI6Ii8xMzY5Mjg3Ni8zOTg3MjQ5MTUtMThkOGRiNmUtZDBkNS00ZGVmLTlhNDItZTM4YWFhMjlhZjQyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA1VDE3NTE0NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI3ZjIyMWExOWRjY2U2ZGU0MGU2OWFmNjAzMmUwMTQ2MmU4YWNiYjBlNDNjM2Y0YTdiYWYwMWVlZWYxOTVjMjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.DXo0_7zvqDBFdGA9CPLZCJbpADDEWqPWBXCQQtn7iD8)
> > 
> > 
> > Hi everyone,
> > I’m using the same thermostat, and the quirk is working well overall. However, I’m struggling with toggling the backlight. I couldn’t find any relevant attributes or options to adjust it.
> > In addition, I don’t see the "Manage Zigbee Device" section, where you can view Clusters, bindings, and so on, like in the reference picture.
> > One thing to note is that my Home Assistant instance is set to German. Could this be affecting the visibility of this section or its functionality? Could someone kindly provide a step-by-step guide on how to control the backlight? Ideally, I’d like to automate it so the backlight turns off in the bedroom between 10 PM and 8 AM.
> > Thanks in advance for your help! 😊
> 
> The "Manage Zigbee Device" option is on the device menu: <img alt="image" width="505" src="https://private-user-images.githubusercontent.com/13692876/400231359-f533c035-e038-4cf2-8302-34cc716b2110.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYxNzQwNjMsIm5iZiI6MTczNjE3Mzc2MywicGF0aCI6Ii8xMzY5Mjg3Ni80MDAyMzEzNTktZjUzM2MwMzUtZTAzOC00Y2YyLTgzMDItMzRjYzcxNmIyMTEwLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA2VDE0MjkyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBkYjBhYTQ2ZDI4NGIxNGUwNzQ0MDRmYzU5MmE0ZGUwZTFjZWY0NzI5ZWIxOGIxZWNkODc3Njc5MjJiNzRhNjgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.wZoV2Rc4-9FrF95eiFIEmukvsuBHm409vydTzq_M_-c">

Thank you, I've found it.

However, I still can't control the brightness. I suppose I'll switch to z2m.

Turning it off at night is a workaround, but no solution.

tux1337, 2025-01-06:
There is a setting to power of the backlight, then it will go on when you touch it. I thought to use this in the night and put it to some brightness level in the morning. But for this we need the setting as a ha control to use it in automation. 
I tried to adapt the quirk but I had not 100% success. I am not so deep into this python stuff, so maybe someone could help out.

Jerseyman99, 2025-01-10:
For those with the displayed actula temperature going down when heating to a set point, the issues is related to a thermostat setting, not the quirk. For those with a Samotech  TGM-50, check out the Advanced Option setting (magnifying glass needed!) under setting 02  change the setting to OUT then it will start using the underfloor heat sensor,

NAGL95, 2025-01-14:
Can anyone share original firmware of this thermostat (beok TGM50-ZB-EP)?

jenswei, 2025-01-18:
_Firmware_
The original fimware is not publicly available I assume because no firmware update file has been published by BEOK. Also it seems no one yet tried to extract the firmware from the chip.

_Temerature drop phenomenon_
I also noticed the temperature drop with the internal temp sensor. A possible explanation might be due to the relais switching on and heating up just a little, the convection pulls fresh air along the internal temp sensor on the underside, so it then catches more of the air temperature and less of the radiated heat from the walls and furniture.

_Backlight remote control_
By the way, using the BeokManufCluster and the attribute 'backlight (id: 0x046e)', a remote setting of the backlight with modes [0] off with touch activation, [1] low, [2] med, [3] high works fine! Would be great to have this as a menu within the regular device settings.

_Presets for manual, eco, ..._
Also, the 'preset_mode' with [0] manual, [1] schedule, [2] manual schedule, [3] eco is not yet exposed to the device settings, can be found in the 'mode (id: 0x0404)' attribute.

Thank you so much @mkozlina and @hdh75 for having the most important functions already supported now, the TGM50 is a wonderful device in terms of functionality, ease of use and sleek design.

tux1337, 2025-01-26:
> _Backlight remote control_
> By the way, using the BeokManufCluster and the attribute 'backlight (id: 0x046e)', a remote setting of the backlight with modes [0] off with touch activation, [1] low, [2] med, [3] high works fine! Would be great to have this as a menu within the regular device settings.

@mkozlina: can you help to bring this settings to a HA control. This would help me aswell and make the device perfect for my use case.