zigpy/zha-device-handlers #3420: [Device Support Request] TS0601_TZE204_edl8pz1k electroc heater

rrmt23, 2024-10-12:
### Problem description

Please provide support for the device **TS0601_TZE204_edl8pz1k** 
I found something similar zhaquirks/tuya/ts0601_electric_heating.py

### Solution description

I found support for this device in z2m, maybe this will make things easier. ( https://www.zigbee2mqtt.io/devices/TS0601_floor_thermostat.html and https://github.com/Koenkk/zigbee2mqtt/issues/22079 )

```
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZE200_edl8pz1k', '_TZE204_edl8pz1k']),
    model: 'TS0601_floor_thermostat',
    vendor: 'TuYa',
    description: 'Zigbee thermostat for electric floors',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
    exposes: [
        e.climate()
            .withSystemMode(['off', 'heat'], ea.STATE_SET)
            .withPreset(['manual', 'auto'])
            .withRunningState(['idle', 'heat'], ea.STATE)
            .withSetpoint('current_heating_setpoint', 5, 35, 0.5, ea.STATE_SET)
            .withLocalTemperature(ea.STATE).withDescription('Floor temperature')
            .withLocalTemperatureCalibration(-9, 9, 0.1, ea.STATE_SET).withDescription('Calibration floor temperature sensor'),
        e.deadzone_temperature().withValueMin(0).withValueMax(5).withValueStep(1).withDescription('Floor temperature'),
        e.child_lock(),
        ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'system_mode', tuya.valueConverterBasic.lookup({'heat': true, 'off': false})],
            [2, 'preset', tuya.valueConverter.tv02Preset()],
            [16, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [24, 'device_temperature', tuya.valueConverter.divideBy10],
            [27, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration2],
            [36, 'running_state', tuya.valueConverterBasic.lookup({'heat': tuya.enum(0), 'idle': tuya.enum(1)})],
            [40, 'child_lock', tuya.valueConverter.lockUnlock],
            [102, 'local_temperature', tuya.valueConverter.divideBy10],
            [103, 'deadzone_temperature', tuya.valueConverter.raw],
            [110, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [109, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [108, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [107, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [106, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [105, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [101, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDP],
        ],
    },
    whiteLabel: [{vendor: 'ELECTSMART', model: 'EST-120Z'}],
};
module.exports = definition;
```

Thank you for your respectful and useful work!

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

rrmt23, 2024-10-12:
@Kastorsky
May be you can help please?

Kastorsky, 2024-10-12:
hi, what happened?

rrmt23, 2024-10-14:
> hi, what happened?

Hi @Kastorsky I saw in the thread https://github.com/Koenkk/zigbee2mqtt/issues/22079#issuecomment-2066835429 
you made a converter for z2m. Can I ask you to adapt your code for ZHA?
The device is absolutely the same.

jacekk015, 2024-11-03:
You can check
[ts0601_thermostat_electsmart.py](https://github.com/jacekk015/zha_quirks/blob/main/ts0601_thermostat_electsmart.py)
at
https://github.com/jacekk015/zha_quirks

closed 2025-12-31 by github-actions[bot]