zigpy/zha-device-handlers #3421: [Device Support Request] TS0601 _TZE204_vawy74yh Smoke detector

Keffu3636, 2024-10-12:
### Problem description

Please provide support for the device TS0601 _TZE204_vawy74yh

### Solution description

Looks like it is working with Z2M - https://www.zigbee2mqtt.io/devices/ZSS-HM-SSD01.html

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Keffu3636, 2024-10-15:
Managed to patch something together and I'm getting attribute values now . Anyone with deeper knowledge please comment if I'm on the right track :) Thanks!

Sources used:
https://github.com/Koenkk/zigbee-herdsman-converters/blob/b405404c2d8db02d94feb3a114c9baf2bd2fcf4e/src/devices/moes.ts#L400
and most of the code I took from here:
https://github.com/zigpy/zha-device-handlers/issues/3249#issuecomment-2282969644
Many thanks to the authors.

```
import zigpy.types as t
from typing import Any, Type
from zigpy.profiles import zha 
from zigpy.quirks import CustomDevice, CustomCluster
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    OnOff,
    Ota,
    PowerConfiguration,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.security import IasZone
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya.mcu import TuyaMCUCluster
from zhaquirks.tuya import (
    TuyaManufCluster,
    DPToAttributeMapping,
    EnchantedDevice,
    TuyaNoBindPowerConfigurationCluster,
)

ZONE_TYPE = 0x0001

class TuyaSmokeSensorCluster(TuyaMCUCluster):
    """Tuya manufacturer cluster for smoke sensor."""

    # Copy the attributes from the base class
    attributes = TuyaMCUCluster.attributes.copy()

    # Update the attributes with the custom ones for this smoke sensor
    attributes.update(
        {
            0xEF01: ("smoke", t.uint8_t, True),  # Smoke status
            0xEF09: ("self_test", t.uint8_t, True),  # Self-test result (Enum)
            0xEF0E: ("battery_state", t.uint8_t, True),  # Battery state (raw value)
            0xEF0F: ("battery", t.uint8_t, True),  # Battery level (raw value)
            0xEF10: ("silence", t.Bool, True),  # Silence alarm (On/Off)
        }
    )

    dp_to_attribute = {
        1: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "smoke",
        ),
        9: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "self_test",
        ),
        14: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "battery_state",
        ),
        15: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "battery",
        ),
        16: DPToAttributeMapping(
            TuyaMCUCluster.ep_attribute,
            "silence",
        ),
    }

    data_point_handlers = {
        1: "_dp_2_attr_update",
        9: "_dp_2_attr_update",
        14: "_dp_2_attr_update",
        15: "_dp_2_attr_update",
        16: "_dp_2_attr_update",
    }

class TuyaIasZone(CustomCluster, IasZone):
    """IAS Zone."""

    _CONSTANT_ATTRIBUTES = {ZONE_TYPE: IasZone.ZoneType.Fire_Sensor}

class MoesSmokeSensor(EnchantedDevice):
    """ZHA quirk for Moes ZSS-HM-SSD01 smoke sensor."""

    signature = {
        MODELS_INFO: [("_TZE204_vawy74yh", "TS0601")],
             ENDPOINTS: {
                 1: {
                    PROFILE_ID: zha.PROFILE_ID,
                    DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                    INPUT_CLUSTERS: [
                        Groups.cluster_id,
                        Scenes.cluster_id,
                        TuyaMCUCluster.cluster_id,
                        Basic.cluster_id,
            ],
            OUTPUT_CLUSTERS: [
                Time.cluster_id,
                Ota.cluster_id,
            ],
        },
    },
}

    replacement = {
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Basic.cluster_id,
                    Scenes.cluster_id,
                    TuyaNoBindPowerConfigurationCluster,
                    TuyaSmokeSensorCluster,
            ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }

```
[image]

Using ZHA-toolkit for entities, let's see how it is going to work.

pulsar895, 2024-10-20:
Hello,

I've just made a pull request for the smoke detector to be seen: https://github.com/zigpy/zha-device-handlers/pull/3447

I hope that others a little more at ease than me will know how to add the battery status...

Keffu3636, 2024-10-20:
Sorry, but what does it mean exactly?

Keffu3636, 2024-10-25:
After HA restart it start to populate values correctly. It works, but I don't have much confidence in it :)

closed 2024-12-09 by TheJulianJES

Keffu3636, 2024-12-16:
Thanks, but what exactly must I do now to get it working?

Keffu3636, 2024-12-27:
Removed my device from HA, emptied my custom quirks folder including __pycahce__, full restart of HA and then tried to add the device again. It still does not seem to work. 

<details><summary>Details</summary>
<p>

2024-12-27 16:20:28.996 DEBUG (MainThread) [zigpy.application] Received a packet: ZigbeePacket(timestamp=datetime.datetime(2024, 12, 27, 14, 20, 28, 996436, tzinfo=datetime.timezone.utc), priority=0, src=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x1A31), src_ep=1, dst=AddrModeAddress(addr_mode=<AddrMode.NWK: 2>, address=0x0000), dst_ep=1, source_route=None, extended_timeout=False, tsn=0, profile_id=260, cluster_id=0, data=Serialized[b'\x18\x04\x01\x04\x00\x00B\x10_TZE204_vawy74yh\x05\x00\x00B\x06TS0601'], tx_options=<TransmitOptions.NONE: 0>, radius=28, non_member_radius=0, lqi=108, rssi=None)
2024-12-27 16:20:28.996 DEBUG (MainThread) [zigpy.zcl] [0x1A31:1:0x0000] Received ZCL frame: b'\x18\x04\x01\x04\x00\x00B\x10_TZE204_vawy74yh\x05\x00\x00B\x06TS0601'
2024-12-27 16:20:28.997 DEBUG (MainThread) [zigpy.zcl] [0x1A31:1:0x0000] Decoded ZCL frame: Basic:Read_Attributes_rsp(status_records=[ReadAttributeRecord(attrid=4, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='_TZE204_vawy74yh')), ReadAttributeRecord(attrid=5, status=<Status.SUCCESS: 0>, value=TypeValue(type=CharacterString, value='TS0601'))])
2024-12-27 16:20:28.998 INFO (MainThread) [zigpy.device] [0x1a31] Read model 'TS0601' and manufacturer '_TZE204_vawy74yh' from <Endpoint id=1 in=[groups:0x0004, scenes:0x0005, None:0xEF00, basic:0x0000] out=[ota:0x0019, time:0x000A] status=<Status.ZDO_INIT: 1>>
2024-12-27 16:20:28.998 INFO (MainThread) [zigpy.device] [0x1a31] Discovered basic device information for <Device model='TS0601' manuf='_TZE204_vawy74yh' nwk=0x1A31 ieee=a4:c1:38:35:2e:01:ae:2c is_initialized=True>
2024-12-27 16:20:28.998 DEBUG (MainThread) [zigpy.application] Device is initialized <Device model='TS0601' manuf='_TZE204_vawy74yh' nwk=0x1A31 ieee=a4:c1:38:35:2e:01:ae:2c is_initialized=True>
2024-12-27 16:20:28.998 DEBUG (MainThread) [zha] Emitting event raw_device_initialized with data RawDeviceInitializedEvent(device_info=RawDeviceInitializedDeviceInfo(ieee=a4:c1:38:35:2e:01:ae:2c, nwk=0x1A31, pairing_status=<DevicePairingStatus.INTERVIEW_COMPLETE: 2>, model='TS0601', manufacturer='_TZE204_vawy74yh', signature={'manufacturer': '_TZE204_vawy74yh', 'model': 'TS0601', 'node_desc': {'logical_type': <LogicalType.EndDevice: 2>, 'complex_descriptor_available': 0, 'user_descriptor_available': 0, 'reserved': 0, 'aps_flags': 0, 'frequency_band': <FrequencyBand.Freq2400MHz: 8>, 'mac_capability_flags': <MACCapabilityFlags.AllocateAddress: 128>, 'manufacturer_code': 4417, 'maximum_buffer_size': 66, 'maximum_incoming_transfer_size': 66, 'server_mask': 10752, 'maximum_outgoing_transfer_size': 66, 'descriptor_capability_field': <DescriptorCapability.NONE: 0>}, 'endpoints': {1: {'profile_id': 260, 'device_type': <DeviceType.SMART_PLUG: 81>, 'input_clusters': [4, 5, 61184, 0], 'output_clusters': [25, 10]}}}), event_type='zha_gateway_message', event='raw_device_initialized') (1 listeners)
2024-12-27 16:20:28.998 DEBUG (MainThread) [zha] (ZHAGatewayProxy) handling event protocol for event: RawDeviceInitializedEvent(device_info=RawDeviceInitializedDeviceInfo(ieee=a4:c1:38:35:2e:01:ae:2c, nwk=0x1A31, pairing_status=<DevicePairingStatus.INTERVIEW_COMPLETE: 2>, model='TS0601', manufacturer='_TZE204_vawy74yh', signature={'manufacturer': '_TZE204_vawy74yh', 'model': 'TS0601', 'node_desc': {'logical_type': <LogicalType.EndDevice: 2>, 'complex_descriptor_available': 0, 'user_descriptor_available': 0, 'reserved': 0, 'aps_flags': 0, 'frequency_band': <FrequencyBand.Freq2400MHz: 8>, 'mac_capability_flags': <MACCapabilityFlags.AllocateAddress: 128>, 'manufacturer_code': 4417, 'maximum_buffer_size': 66, 'maximum_incoming_transfer_size': 66, 'server_mask': 10752, 'maximum_outgoing_transfer_size': 66, 'descriptor_capability_field': <DescriptorCapability.NONE: 0>}, 'endpoints': {1: {'profile_id': 260, 'device_type': <DeviceType.SMART_PLUG: 81>, 'input_clusters': [4, 5, 61184, 0], 'output_clusters': [25, 10]}}}), event_type='zha_gateway_message', event='raw_device_initialized')
2024-12-27 16:20:28.999 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE204_vawy74yh TS0601 (a4:c1:38:35:2e:01:ae:2c)
2024-12-27 16:20:29.001 DEBUG (MainThread) [zha.application.discovery] Discovering entities for device: a4:c1:38:35:2e:01:ae:2c-_TZE204_vawy74yh TS0601
2024-12-27 16:20:29.001 DEBUG (MainThread) [zha.application.discovery] Attempting to discover quirks v2 entities for device: a4:c1:38:35:2e:01:ae:2c-_TZE204_vawy74yh TS0601
2024-12-27 16:20:29.001 DEBUG (MainThread) [zha.application.discovery] Device: a4:c1:38:35:2e:01:ae:2c-_TZE204_vawy74yh TS0601 is not a quirks v2 device - skipping discover_quirks_v2_entities
2024-12-27 16:20:29.008 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (a4:c1:38:35:2e:01:ae:2c, 1, <ClusterType.Server: 0>, 0, 4, '_TZE204_vawy74yh', datetime.datetime(2024, 12, 27, 14, 20, 28, 998210, tzinfo=datetime.timezone.utc)) params: FOREIGN KEY constraint failed
2024-12-27 16:20:29.648 DEBUG (MainThread) [zha] Emitting event device_fully_initialized with data DeviceFullInitEvent(device_info=ExtendedDeviceInfoWithPairingStatus(ieee=a4:c1:38:35:2e:01:ae:2c, nwk=0x1A31, manufacturer='_TZE204_vawy74yh', model='TS0601', name='_TZE204_vawy74yh TS0601', quirk_applied=False, q [... cut ...] 7>)}, neighbors=[], routes=[], endpoint_names=[EndpointNameInfo(name='SMART_PLUG')], pairing_status=<DevicePairingStatus.INITIALIZED: 4>), new_join=False, event_type='zha_gateway_message', event='device_fully_initialized')
2024-12-27 16:20:39.685 WARNING (Recorder) [homeassistant.components.recorder.entity_registry] Cannot migrate history for entity_id `update.tze204_vawy74yh_ts0601_pusivara` to `update.magamistuba_smokealarm_pusivara` because the new entity_id is already in use

</p>
</details>

prairiesnpr, 2024-12-27:
This will be available in the next HA release, 2025.1.

Keffu3636, 2024-12-27:
OK, Thanks!