zigpy/zha-device-handlers #3431: [Device Support Request] Avatto TRV06 _TZE200_yqgbrdyo

tttopuz, 2024-10-15:
### Problem description

I recently bought new Avatto TRV06 that ZHA had trouble identifying. I got the _TZE200_yqgbrdy0. I have added this model number to the custom quirk ts0601_trv_me167.py as specified in this thread https://github.com/zigpy/zha-device-handlers/issues/1818 and it seems to be working.

Some options are not supported and have some errors, but in general the TRV is controllable.

### Solution description

Not all options are available and there are some errors in the device info screen. Is it possible to get support for this?

### Screenshots/Video

[image]

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0301",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0004",
        "0x0005",
        "0x0201",
        "0x0204",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0006"
      ],
      "output_clusters": []
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0x000d",
      "input_clusters": [
        "0x000d"
      ],
      "output_clusters": []
    },
    "4": {
      "profile_id": "0x0104",
      "device_type": "0x0000",
      "input_clusters": [
        "0x0006"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZE200_yqgbrdyo",
  "model": "TS0601",
  "class": "ts0601_trv_me167.ME167"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

ganjavodoo, 2024-10-22:
I just bought 7 thermostatic heads from the AVATTO brand, recognized under ZHA by _TZE200_yqgbrdy0.
I created a custom_zha_quirks/ folder where I placed the ts0601_trv_me167.py file, adding the model _TZE200_yqgbrdy0. (screenshot attached);
[image]
It doesn't seem to be recognized; I manage to pair the head, but I don't have access to anything (screenshot attached).
I can't figure out what mistake I made."
[image]

Here is my configure.yaml for your reference.
[image]

Could you help me?
Thank you in advance."

tttopuz, 2024-10-22:
I have the following in my configuration.yaml. The folder reference is different.

[image]

Also, you need to wait 15 min after pairing to have the entities pop up magically for some reason. It also took me a couple of tries.

ganjavodoo, 2024-10-22:
Thank you, I just managed to complete the integration, but I don't see the battery sensor.
[image]

And what is the purpose of 'number temperature'?
[image]

Thanks

tttopuz, 2024-10-22:
The quirk is not fully compatible, so some values are not known. That's why I put it here, to ask for support.

Also, for me some values magically starting popping up after 12 hours, so give it some time.

Number temperature is to offset to the c

tttopuz, 2024-10-22:
t

closed 2025-01-16 by TheJulianJES