zigpy/zha-device-handlers #3442: [Device Support Request] Skydance WZ-SPI RGBW LED Strip Controller

kim366, 2024-10-18:
### Problem description

The device connects to Home Assistant correctly. Firmware and diagnostic information is shown. It is, however, not possible to control the color/brightness/pattern of the LED strip.

Product Page: https://iskydance.com/index.php?c=product_show&a=index&id=2439

### Solution description

Control LED strips in the same way as other LED strip controllers already supported by ZHA.

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_8fffc3kb",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

dav972, 2024-10-23:
We can't know the dates for the updates? We can't know the dates for the updates?

KrolikROJER, 2025-03-01:
i can add support device, but settings work for:
* bright
* state: on/off
"that's all, "(c)

create `wz_spi.js` and put in external converters

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const {
    light
} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    fingerprint: tuya.fingerprint('TS0601', ['_TZB210_3zfp8mki', '_TZB210_gj0ccsar', '_TZE204_8fffc3kb']),
    model: 'TS0601',
    vendor: 'Tuya',
    description: 'Zigbee RGB+CCT light',
    // fromZigbee: [tuya.fz.datapoints],
    // toZigbee: [tuya.tz.datapoints],
	fromZigbee: [tz.light], //
    toZigbee: [tz.light],  
    exposes: [
		e.light(),
		// e.switch(), duplicate in e.light
        // e.brightness(), not supported
        // e.color(), not supported
        // Here you should put all functionality that your device exposes
    ],
    meta: {
        tuyaDatapoints: [
            [1, 'state', tuya.valueConverter.onOff], // <- mapped the datapoint
            [3, 'brightness', tuya.valueConverter.scale0_254to0_1000], // <- mapped the datapoint
            [4, 'color_temp', tuya.valueConverter.scale250_454to0_1000], // <- mapped the datapoint
            [14, 'power_on_behavior', tuya.valueConverter.powerOnBehaviorEnum],
        ],
    },
			   
    extend: [
        light({
            colorTemp: {
                range: [153, 500]
            },
            color: {
                modes: ['hs'],
                applyRedFix: true,
                enhancedHue: false
            }
        }),
    ],
    configure: async(device, coordinatorEndpoint) => {
        device.getEndpoint(1).saveClusterAttributeKeyValue('lightingColorCtrl', {
            colorCapabilities: 29
        });
    },
    // },
    // meta: {
    // // All datapoints go in here
    // tuyaDatapoints: [],
    // },
    // extend: [
    // A preferred new way of extending functionality.
    // ],
    onEvent: tuya.onEventSetTime,
    // configure: tuya.configureMagicPacket,
	configure: async (device, coordinatorEndpoint) => {
        device.getEndpoint(1).saveClusterAttributeKeyValue('lightingColorCtrl', {colorCapabilities: 29});
    },
    whiteLabel: [
        tuya.whitelabel('Skydance', 'WZ5_dim_2', 'Zigbee & RF 5 in 1 LED controller (DIM mode)', ['_TZB210_3zfp8mki']),
        tuya.whitelabel('Skydance', 'WZ-SPI', 'Zigbee & RF 5 in 1 LED controller (DIM mode)', ['_TZE204_8fffc3kb']),
        tuya.whitelabel('QA', 'QADZC5', '5 in 1 LED controller', ['_TZB210_gj0ccsar']),
    ],
};

module.exports = definition;
```

THK-artjom, 2025-03-21:
This relates to request of:

https://github.com/Koenkk/zigbee2mqtt/issues/23991

bedoron, 2025-04-30:
+1 it would be great to have it

james-callahan, 2025-06-26:
FWIW there is some info over at https://github.com/Koenkk/zigbee2mqtt/issues/23991

howels, 2025-10-01:
After looking for a ZigBee solution I have moved to WLED to control addressable strips, it's a versatile solution.

VBart, 2025-10-31:
+1 waiting for it

Rudis1261, 2026-04-03:
Hey, please see this, maybe try it out and add a comment if it helped you get your device to work, I only have the one strip type to test with sadly:
https://github.com/zigpy/zha-device-handlers/pull/4901

Deniz-KK, 2026-08-21:
<html><head></head><body><p>I have been working on this device (<code>_TZE204_8fffc3kb</code> / <code>TS0601</code>, Skydance WZ-SPI, firmware <code>0x0000004a</code>) for the past two days and have mapped most of it. Sharing everything here so nobody has to start from zero — including the one part that still does not work.</p>
<h2>Datapoints (verified on the device)</h2>

DP | Function | Type | Range / notes
-- | -- | -- | --
1 | On/Off | bool | works both directions
2 | Work mode | enum8 | write only — this device never reports DP 2, not even on a data query
3 | Brightness | uint16 | 10–1000, works both directions
46 | Strip length | uint16 | max 1000
47 | Pixel count | uint16 | max 1000
53 | Pixel count | uint16 | see note on segments below
61 | Colour data | raw, 12 bytes | see format below
101 | Chip type | enum8 | factory default 1 = TM1809
102 | Colour order | enum8 | BRG = index 4 on my strip

<p>The configuration datapoints (46, 47, 53, 101, 102) stay empty until the Tuya data query spell is sent. With <code>TuyaQuirkBuilder(...).tuya_enchantment(data_query_spell=True)</code> they all populate on "Reconfigure".</p>
<h2>DP 61 colour format</h2>
<p>12 bytes, big-endian, with a fixed frame:</p>
<pre><code>00 01 00 14 00 | HH HH | SS SS | VV VV | 00
                 hue     sat     value
</code></pre>
<ul>
<li><strong>Hue</strong>: 0–360, in degrees (not 0–254)</li>
<li><strong>Saturation</strong>: 0–1000</li>
<li><strong>Value</strong>: 0–1000</li>
</ul>
<p>Verified against real frames — for example full red is <code>000100140000000003e803e800</code>, green (120°) is <code>0001001400007803e803e800</code>. On the wire the payload carries the usual Tuya length prefix (<code>0c</code> = 12), so the raw frame is 13 bytes.</p>
<p>The device <strong>does</strong> send DP 61 spontaneously when the colour is changed with the RF remote, roughly every 1–2 seconds while the colour wheel is being dragged. I confirmed this by mapping DP 61 to a read-only sensor first, before attempting any write support.</p>
<h2>Note on pixel count</h2>
<p>The controller always divides the strip into exactly 20 segments (pixels ÷ 20 = pixels per segment), which is documented in the manufacturer manual. If the pixel count is not a multiple of 20, the remainder shows the colour of the last segment. On my 280-pixel strip the factory value of 300 put roughly one segment past the physical end — irrelevant for solid colours, but it breaks segment effects.</p>
<h2>What works</h2>
<p>On/Off, brightness, and all configuration datapoints (46, 47, 53, 101, 102) can be read and written reliably through ZHA. A <code>light</code> entity with brightness works, and exposing colour as XY makes Home Assistant show a colour wheel — note that ZHA has no HS colour mode at all, it only checks <code>_color_xy_supported</code> and <code>_color_temp_supported</code>, so a Tuya device that speaks hue/saturation has to convert internally.</p>
<h2>What does not work — writing colour</h2>
<p>Writing DP 61 does not change the strip. The frame is sent correctly and the device acknowledges it:</p>
<pre><code>Sending request: set_data(data=TuyaCommand(status=0, tsn=151,
    datapoints=[TuyaDatapointData(dp=61, ...)]))
Received command 0x02: set_data_response(data=TuyaCommand(status=0, ...
    datapoints=[TuyaDatapointData(dp=61, ...)]))
</code></pre>
<p><code>set_data_response</code> comes back with no error status, and the payload content is byte-identical to a frame the RF remote produces for the same colour. The strip keeps its previous colour.</p>
<p>Things I ruled out:</p>
<ul>
<li><strong>Work mode</strong>: tested with DP 2 set to each of the four values (0–3) before sending a colour. No effect. Since the device never reports DP 2, I cannot confirm which value the controller actually adopted.</li>
<li><strong>Power cycle</strong>: 30 s without power, then a colour command from Home Assistant before touching the remote. No effect.</li>
<li><strong>Encoding</strong>: verified against real RF frames, byte-identical.</li>
<li><strong>Reconfigure / restart</strong>: done repeatedly, entities stay available.</li>
</ul>
<p>So the controller accepts and acknowledges colour commands over Zigbee but does not apply them, while the same values applied via the RF 2.4G remote work fine.</p>
<h2>Open question</h2>
<p>Does anyone with this controller know whether an additional condition has to be met before DP 61 is applied — a specific work mode value, a required command order, or a lock between the RF and Zigbee interfaces? A datapoint list from the manufacturer would settle it; I have asked Skydance and will post here if I get an answer.</p>
<h2>Enum discrepancy worth checking</h2>
<p>Issue #3972 lists the work mode enum for this same manufacturer ID as:</p>
<pre><code class="language-python">colour = 0x00
white  = 0x01
scene  = 0x02
</code></pre>
<p>I had assumed <code>white = 0x00, colour = 0x01</code>. Since the device never reports DP 2 back, neither mapping can be confirmed from the device side — worth flagging for anyone building on this.</p></body></html>