zigpy/zha-device-handlers #3444: [BUG] GLEDOPTO GL-C-002P recognized as GLEDOPTO GL-C-003P

Rainerlan, 2024-10-19:
### Bug description

I noticed that I cannot set the color of my RBG strip on my GLEDOPTO GL-C-002P. ZHA integration shows only color temperature & brightness controls.
After seaching for a while, I noticed that the type of the controller is displayed incorrectly in ZHA. Its being recognized as GLEDOPTO GL-C-003P instead of an GLEDOPTO GL-C-002P. Not sure if this is the source problem - but its a start

### Steps to reproduce

- resets of the controller & deleted the device from ZHA
- pair GLEDOPTO GL-C-002P in RGB mode

### Expected behavior

color wheel diaplayed in controls to change the color

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

jrfigura, 2026-01-01:
I have a similar issue caused by this bug.

I've bought (GL-C-301P) Ultra Mini ZigBee Pro+ 5 in 1 Smart LED Controller
https://gledopto.com/h-pd-93.html

That can be configured in 5 different options:

[image]

according to: https://www.zigbee2mqtt.io/devices/GL-C-003P.html
RGB mode is should be recognized as GL-C-003P.

Still - after adding the device it's correctly recognized but the colour wheel is not available.

[image]

[image]

If I change the mode of the device to RGB+CCT or RGBW - device will be recognized as GL-C-008P or GL-C-007P with the colour wheel available. But it won't be able to operate RGB LED strip correctly

jrfigura, 2026-02-04:
I managed to create somehow working zha cutom quirk. It still requires some tinkering but now we have access to RGB wheel and brightness:
```
"""GLEDOPTO GL-C-003P device quirk - RGB Mode."""

from zigpy.profiles import zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
)
from zigpy.zcl.clusters.lighting import Color
from zigpy.zcl.clusters.lightlink import LightLink
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.gledopto import GLEDOPTO

class GledoptoRGBCluster(CustomCluster, Color):
    """Custom Color cluster for GL-C-003P RGB mode."""
    cluster_id = Color.cluster_id
    
    # Force RGB-only support using XY and HS color modes
    _CONSTANT_ATTRIBUTES = {
        # ColorCapabilities: 0x0019 = 25 decimal
        # Bit 0 (Hue/Sat) + Bit 3 (XY) + Bit 4 (Enhanced Hue) = 0b11001 = 25
        0x400A: 25,
        # ColorMode: 1 = Current hue and saturation
        0x0008: 1,
    }

class GLC003P(CustomDevice):
    """GLEDOPTO GL-C-003P RGB mode device."""

    signature = {
        MODELS_INFO: [(GLEDOPTO, "GL-C-003P")],
        ENDPOINTS: {
            11: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: 0x0102,  # Color Dimmable Light
                INPUT_CLUSTERS: [
                    Basic.cluster_id,       # 0x0000
                    Identify.cluster_id,    # 0x0003
                    Groups.cluster_id,      # 0x0004
                    Scenes.cluster_id,      # 0x0005
                    OnOff.cluster_id,       # 0x0006
                    LevelControl.cluster_id,# 0x0008
                    Color.cluster_id,       # 0x0300
                    0x1000,                 # ZLL Commissioning
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,         # 0x0019
                ],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            11: {
                PROFILE_ID: zha.PROFILE_ID,
                # Use Extended Color Light to support both XY and HS
                DEVICE_TYPE: zha.DeviceType.EXTENDED_COLOR_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    GledoptoRGBCluster,  # Custom cluster for proper RGB support
                ],
                OUTPUT_CLUSTERS: [
                    Ota.cluster_id,
                ],
            },
        },
    }
```

What I did not figure out is how to make colour temperature adjustment (its there but does not work)

OsaPL, 2026-09-17:
Still an issue for me. The quirk file from @jrfigura seems to load just fine, but is not used for the device for some reason, I cant see anything in the logs.
Im open to go through device attributes/signature with someone, to figure it out. I have no idea how a quirk file works unfortunately, its all still new to me.
All HS/XY values seem to be there. `device_type` is `0x102`, should it be `0x103` for RGB?