zigpy/zha-device-handlers #3445: [Device Support Request]  _TZE200_9xfjixap Tuya Smart TRV Radiator actuator

Superbrain8, 2024-10-19:
### Problem description

Tuya TRV not Recognized by zha
same design as  _TZE200_yw7cahqs 

### Solution description

adding a quirk for this device

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_9xfjixap",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

ive tried to use the quirk from here https://github.com/zigpy/zha-device-handlers/issues/1818#issuecomment-1279997935 but it has some error with creating a command and refuses to work for me
it matches the inputs

Superbrain8, 2024-10-20:
https://github.com/home-assistant/core/issues/86347
Added quirk from this thread.
Modified it to contain the ID from my Device
Works also for this Device

Superbrain8, 2024-10-22:
there is a section that looks like this
`  MODELS_INFO: [
            ("_TZE200_bvu2wnxz", "TS0601"),
            ("_TZE200_6rdj8dzm", "TS0601"),
        ],`
       you add another ID in the same Schema

Schmandre, 2024-11-05:
Hello, 

I just got the same issue and edited the quirks like mentioned above. The error still exists. Device will be paired, but only showed as "firmware".

```
 MODELS_INFO: [
            ("_TZE200_bvu2wnxz", "TS0601"),
            ("_TZE200_6rdj8dzm", "TS0601"),
            ("_TZE200_9xfjixap", "TS0601"),
        ],
```

[image]

Any other ideas how to fix? 
Used the """ME167 TRV devices support.""" quirk as origin.

muchabo297, 2024-11-06:
got the same device put the recommend quirk in , device is paired it shows some entities but the thermostate is still unknown . 

[image]

grintroy, 2024-11-07:
Hi all, as per https://github.com/home-assistant/core/issues/86347#issuecomment-1713337957, make sure you have all the following:

1. Added `custom_quirks_path` to _`configuration.yaml`_:
```yaml
zha:
  custom_quirks_path: /config/custom_zha_quirks/
```

2. Added the _`ts0601_trv_me167.py`_ from [the link above](https://github.com/home-assistant/core/issues/86347#issuecomment-1713337957) to the _`custom_zha_quirks`_ folder.

3. Changed the line 185 to
```python
ME167_TEMP_CALIBRATION_ATTR: ("temperature_calibration", t.int32s, True),
```

4. Added the line with your model name after line 445:
```python
("_TZE200_9xfjixap", "TS0601"),
```

Personally, I missed steps 1 and 3 but once I did them, deleted the existing device, reloaded HA, and paired it again, it all worked.

There is also [an adapted version](https://gist.github.com/grintroy/26c0b3fd4411579473bf53c11b6a87ca) of _`ts0601_trv_me167.py`_. You can use this one instead in step 2, and you won't have to do steps 3 and 4.

mountbatt, 2024-11-07:
@grintroy Thanks for sharing! Now I can see the current temperature and mode like "heating" or "idle" in HA, but for example no battery status. So it looks like it is not fully supported right now. Can you see all the information?
[image]

grintroy, 2024-11-12:
@mountbatt mine's looking almost the same, but I can also see the battery charge. Is it still unavailable for you? Have you tried the basic routine of rebooting HA or re-adding/reconfiguring the device to the Zigbee network?

For all these extra features, it surely would be nice to have all of them working, but personally, I'm happy to see my thermostat and battery entities — that's more than enough smart radiator control for me. And then I just use Better Thermostat with external sensors, and I'm happy.

jacekk015, 2024-11-13:
You all can try to use the file from
https://github.com/jacekk015/zha_quirks/issues/63#issuecomment-2473156159

ME167 device doesn't report battery level. Only Full/Empty states are reported.
```
ME167_BATTERY_STATE_ATTR = 0x0523  # [0] OK [1] Empty
```

closed 2025-01-16 by TheJulianJES