zigpy/zha-device-handlers #3446: [Device Support Request] Thermostat Thaleos TRV06-AT TS0601 _TZE200_ow09xlxm

bsfaxi, 2024-10-20:
### Problem description

Could you please add this model to the quirk available [here](https://github.com/home-assistant/core/issues/86347) that worked for me also ?
The acquired Thermostat is the following : https://thaleos.com/wp-content/uploads/2024/05/Tete-thermostatique-4.pdf.
It's just a rebranded Tuya Thermostat TRV.

### Solution description

I added the quirk and updated the list of model with mine. It worked !

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_ow09xlxm",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```
[... 487 lines cut ...]
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_SWITCH,
                INPUT_CLUSTERS: [
                    ME167ChildLock,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.CONSUMPTION_AWARENESS_DEVICE,
                INPUT_CLUSTERS: [ME167TempCalibration],
                OUTPUT_CLUSTERS: [],
            },
        }
    }
```

</details>

### Additional information

NA

bsfaxi, 2024-10-20:
Still have to tweak the "Child Lock" Feature.
The button doesn't work!

[image]

bsfaxi, 2024-10-20:
I can see that the button state changes when call the lock function in the configuration, as per the screenshot below.
The better is to change the Button to a Boolean state...
[image]

Layochi, 2025-02-22:
Hi, have you new informations about the _TZE200_ow09xlxm ?

Pegazus57, 2025-03-01:
Hi, for my part I did the same (adding the  _TZE200_ow09xlxm model) but with the file "ts0601_thermostat_avatto.py" from this link : https://github.com/jacekk015/zha_quirks

Thank you !

closed 2025-09-06 by github-actions[bot]