zigpy/zha-device-handlers #3453: [Help] Converting Zlinky_TIC quirk to a quirk v2

blauret, 2024-10-24:
This is issue is request for help. I hope we can have a conversation about how to convert an existing quirk into a quirk v2. I could not find existing issue related to that.  I also hope documenting my steps here can help other developper during their conversion. 

Following the PR #3448, @TheJulianJES suggested to convert the quirk for Zlinky_TIC into a quirk v2. 

The proposed modification is to replace the signature for Fw 14.0 with

```
(
    QuirkBuilder(LIXEE, "ZLinky_TIC")
    .replaces(ZLinkyTICManufacturerCluster)
    .add_to_registry()
)
```
Looking at the log it does not look like it's working

```
2024-10-24 11:30:49.307 DEBUG (MainThread) [zha.application.discovery] Discovering entities for endpoint: 00:15:8d:00:05:be:83:47-1
2024-10-24 11:30:49.307 DEBUG (MainThread) [zha.application.discovery] 'button' platform -> 'IdentifyButton' using ['identify']
2024-10-24 11:30:49.307 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'Tier1SmartEnergySummation' using ['smartenergy_metering']
2024-10-24 11:30:49.307 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'Tier2SmartEnergySummation' using ['smartenergy_metering']
2024-10-24 11:30:49.307 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'Tier3SmartEnergySummation' using ['smartenergy_metering']
2024-10-24 11:30:49.307 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'Tier4SmartEnergySummation' using ['smartenergy_metering']
2024-10-24 11:30:49.307 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'Tier5SmartEnergySummation' using ['smartenergy_metering']
2024-10-24 11:30:49.307 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'Tier6SmartEnergySummation' using ['smartenergy_metering']
2024-10-24 11:30:49.307 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'Battery' using ['power']
2024-10-24 11:30:49.307 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'ElectricalMeasurementApparentPower' using ['electrical_measurement']
2024-10-24 11:30:49.307 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'ElectricalMeasurementRMSCurrent' using ['electrical_measurement']
2024-10-24 11:30:49.308 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'ElectricalMeasurementRMSVoltage' using ['electrical_measurement']
2024-10-24 11:30:49.308 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'ElectricalMeasurementFrequency' using ['electrical_measurement']
2024-10-24 11:30:49.308 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'ElectricalMeasurementPowerFactor' using ['electrical_measurement']
2024-10-24 11:30:49.308 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'SmartEnergySummationReceived' using ['smartenergy_metering']
2024-10-24 11:30:49.308 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'RSSISensor' using ['basic']
2024-10-24 11:30:49.308 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'LQISensor' using ['basic']
2024-10-24 11:30:49.308 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'PolledElectricalMeasurement' using ['electrical_measurement']
2024-10-24 11:30:49.308 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'PolledSmartEnergySummation' using ['smartenergy_metering']
2024-10-24 11:30:49.308 DEBUG (MainThread) [zha.application.discovery] 'update' platform -> 'FirmwareUpdateEntity' using ['ota']
2024-10-24 11:30:49.308 DEBUG (MainThread) [zha.zigbee.endpoint] Creating cluster handler for cluster id: 33 class: <class 'zha.zigbee.cluster_handlers.general.GreenPowerProxyClusterHandler'>
2024-10-24 11:30:49.308 DEBUG (MainThread) [zha.application.discovery] Discovering entities for endpoint: 00:15:8d:00:05:be:83:47-242
2024-10-24 11:30:49.308 DEBUG (MainThread) [zha.application.discovery] Discovering entities for device: 00:15:8d:00:05:be:83:47-LiXee ZLinky_TIC
2024-10-24 11:30:49.308 DEBUG (MainThread) [zha.application.discovery] Attempting to discover quirks v2 entities for device: 00:15:8d:00:05:be:83:47-LiXee ZLinky_TIC
2024-10-24 11:30:49.308 DEBUG (MainThread) [zha.application.discovery] Device: 00:15:8d:00:05:be:83:47-LiXee ZLinky_TIC does not expose any quirks v2 entities
2024-10-24 11:30:49.309 DEBUG (MainThread) [zha.application.gateway] [0xAC5D](LiXee ZLinky_TIC) restored as 'available', last seen: 0:03:54 ago, consider_unavailable_time: 7200 seconds
2024-10-24 11:30:49.309 DEBUG (MainThread) [zha.zigbee.endpoint] Creating cluster handler for cluster id: 0 class: <class 'zha.zigbee.cluster_handlers.general.BasicClusterHandler'>
2024-10-24 11:30:49.309 DEBUG (MainThread) [zha.zigbee.endpoint] Creating cluster handler for cluster id: 3 class: <class 'zha.zigbee.cluster_handlers.general.IdentifyClusterHandler'>
2024-10-24 11:30:49.309 DEBUG (MainThread) [zha.zigbee.endpoint] Creating cluster handler for cluster id: 32 class: <class 'zha.zigbee.cluster_handlers.general.PollControlClusterHandler'>
2024-10-24 11:30:49.309 DEBUG (MainThread) [zha.zigbee.endpoint] Creating cluster handler for cluster id: 4096 class: <class 'zha.zigbee.cluster_handlers.lightlink.LightLinkClusterHandler'>
2024-10-24 11:30:49.309 DEBUG (MainThread) [zha.zigbee.endpoint] Creating cluster handler for cluster id: 64599 class: <class 'zha.zigbee.cluster_handlers.ClusterHandler'>
2024-10-24 11:30:49.309 DEBUG (MainThread) [zha.zigbee.endpoint] Creating cluster handler for cluster id: 1 class: <class 'zha.zigbee.cluster_handlers.general.PowerConfigurationClusterHandler'>
2024-10-24 11:30:49.309 DEBUG (MainThread) [zha.application.discovery] Discovering entities for endpoint: 60:b6:47:ff:fe:fa:bd:6f-1
2024-10-24 11:30:49.309 DEBUG (MainThread) [zha.application.discovery] 'button' platform -> 'IdentifyButton' using ['identify']
2024-10-24 11:30:49.309 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'Battery' using ['power']
2024-10-24 11:30:49.309 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'RSSISensor' using ['basic']
2024-10-24 11:30:49.309 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'LQISensor' using ['basic']
2024-10-24 11:30:49.310 DEBUG (MainThread) [zha.application.discovery] 'update' platform -> 'FirmwareUpdateEntity' using ['ota']

```

Although in HA it looks like the device is associated with my quirks v2:

[image]

So I'm a little confused by that line in the logs:

```
2024-10-24 11:30:49.308 DEBUG (MainThread) [zha.application.discovery] Device: 00:15:8d:00:05:be:83:47-LiXee ZLinky_TIC does not expose any quirks v2 entities
```

blauret, 2024-10-24:
Ok, I figured out that if you don't define any entities in the quirkbuilder you get the message
```
2024-10-24 11:30:49.308 DEBUG (MainThread) [zha.application.discovery] Device: 00:15:8d:00:05:be:83:47-LiXee ZLinky_TIC does not expose any quirks v2 entities
```

After adding the sensor sensor the message is replaced

```
24-10-24 13:23:39.670 DEBUG (MainThread) [zha.application.discovery] Attempting to discover quirks v2 entities for device: 00:15:8d:00:05:be:83:47-LiXee ZLinky_TIC
2024-10-24 13:23:39.670 DEBUG (MainThread) [zha.application.discovery] 'sensor' platform -> 'Sensor' using ['smartenergy_metering']
2024-10-24 13:23:39.670 DEBUG (MainThread) [zha.application.gateway] [0xAC5D](LiXee ZLinky_TIC) restored as 'available', last seen: 0:00:50 ago, consider_unavailable_time: 7200 seconds
```

blauret, 2024-10-24:
Now I can add a sensor and get value, but I did not manage to get the information properly displayed. I was expecting the translation_key parameter to be the sensor name string, but it does not work that way. 

```
    .sensor(ElectricalMeasurement.AttributeDefs.active_power_max.name, ElectricalMeasurement.cluster_id,translation_key="Power Max")
```

[image]

blauret, 2024-10-25:
The assoicated PR is #3456

TheJulianJES, 2024-10-26:
You don't need to add any custom entities for a v2 quirk. It's just a debug message stating that none were added.
For just replacing a cluster, like the current v1, this should already be enough to replace all v1 quirks:
```python
(
    QuirkBuilder(LIXEE, "ZLinky_TIC")
    .replaces(ZLinkyTICMetering)
    .replaces(ZLinkyTICManufacturerCluster)
    .add_to_registry()
)
```
But I see you already have that in your PR.

blauret, 2024-10-28:
@TheJulianJES, thanks. I'm actually trying to extend the quirk as I convert it. The basic just works. Maybe you can help with the following. 

Right now, I can't get around string I fetch from the zlinky_tic device being processed by the stringify mechanism. I'm not quite sure, because the code base is new to me, but it feels like an issue in the ZHA sensor handling.

I'm adding a sensor like that:

```python
    .sensor(
        Basic.AttributeDefs.model.name,
        Basic.cluster_id,
        fallback_name="Model",
    )
```

The atrribute type is **type=t.LimitedCharString(32)** but the type seems to be ignore in ZHA sensor handling and eventully I get this error. 

```console
2024-10-28 10:38:43.407 ERROR (MainThread) [homeassistant.components.sensor] Error adding entity sensor.lixee_zlinky_tic_model for domain sensor with platform zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 595, in _async_add_entities
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 909, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1366, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1005, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1130, in _async_write_ha_state
    self.__async_calculate_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1067, in __async_calculate_state
    state = self._stringify_state(available)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1011, in _stringify_state
    if (state := self.state) is None:
                 ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 542, in state
    value = self.native_value
            ^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/sensor.py", line 136, in native_value
    return self.entity_data.entity.native_value
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zha/application/platforms/sensor/__init__.py", line 249, in native_value
    return self.formatter(raw_state)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zha/application/platforms/sensor/__init__.py", line 271, in formatter
    float(value * self._multiplier) / self._divisor, self._decimals
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: could not convert string to float: 'ZLinky_TIC'
```

prairiesnpr, 2024-10-28:
String sensors aren't currently supported in ZHA, you can specify the sensor class via [device_class](https://github.com/zigpy/zigpy/blob/76fa13ae14dcb2ad08174dfd05cd9eb4c30d6fde/zigpy/quirks/v2/__init__.py#L648), but only numeric and timestamps are currently supported. I don't even see a [SensorDeviceClass ](https://github.com/zigpy/zigpy/blob/76fa13ae14dcb2ad08174dfd05cd9eb4c30d6fde/zigpy/quirks/v2/homeassistant/sensor.py#L6) that would correspond to a string. That said, why are you trying to add a sensor for the model name?

blauret, 2024-10-28:
> String sensors aren't currently supported, you can specify the sensor class via [device_class](https://github.com/zigpy/zigpy/blob/76fa13ae14dcb2ad08174dfd05cd9eb4c30d6fde/zigpy/quirks/v2/__init__.py#L648), but only numeric and timestamps are currently supported. I don't even see a [SensorDeviceClass ](https://github.com/zigpy/zigpy/blob/76fa13ae14dcb2ad08174dfd05cd9eb4c30d6fde/zigpy/quirks/v2/homeassistant/sensor.py#L6) that would correspond to a string. That said, why are you trying to add a sensor for the model name?

Thankls, that makes more sense with what I see. 

I'm not trying to make a sensor from the model name. That was just an example. There are other ZCL attributes which report a string. For example, I get the current tarif applied by electricity supplier as a string from ZLIinky. I also have it as an integer and I can apply an enum on that one.

steenlarsen, 2024-10-30:
Thanks for all your work here! I am looking forward to using this for extracting the French EDF Tempo tariff information for today and tomorrow. That would be a good use case for strings, btw, since these are present in the TIC information as strings. An enum could work around this but then all user interfaces have to translate the enum to a string.

blauret, 2024-10-31:
> An enum could work around this but then all user interfaces have to translate the enum to a string.

Look in the PR, The enum is there for the current Tarrif. 

Is this the one you would like to see in the sensor?

[image]

closed 2024-11-25 by blauret