zigpy/zha-device-handlers #3472: [Device Support Request] TS0601 __TZE204_laokfqwu occupancy and light sensor

Igorgro, 2024-10-29:
### Problem description

Brought this device https://aliexpress.com/item/1005004705672138.html?sku_id=12000030169395611, it has model _TZE204_laokfqwu TS0601 and it is not supported in HA without quirks. I tried the following two quirks:
- https://community.home-assistant.io/t/how-i-got-it-working-wz-m100-tuya-presence-sensor-ts0601-tze204-7gclukjs/737497 (with replacing device id)
- https://community.home-assistant.io/t/zha-custom-quirk-to-work-with-tuya-ts0601-tze204-laokfqwu-occupancy-sensor/743184
In the first quirk all parameters except detection_delay was working and in the second all except detection_delay and fading_time. Anyway I tried to play with these two quirks and got very contradictory results - it hardly detect when I enter the room and almost never detected when I exit. I cannot sure, if this a problem of quirk, sensors itself, or my environment. 

### Solution description

Sensor works as expected in Home Assistant with built-in quirk/

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

Personally I didn't write quirksb ut found two attempts of doing that:
- https://community.home-assistant.io/t/how-i-got-it-working-wz-m100-tuya-presence-sensor-ts0601-tze204-7gclukjs/737497
- https://community.home-assistant.io/t/zha-custom-quirk-to-work-with-tuya-ts0601-tze204-laokfqwu-occupancy-sensor/743184

which may be helpful

### Additional information

If someone will pick this issue up, I can help by running quirks on my device.

tamngo3, 2024-11-04:
Same issue here so I don't think it's the device, probably something wrong with the quirk/handler. Subscribing.

MyUnknownSelf, 2025-01-02:
i see more some Zigbee2MQTT applications for the sensor. I tried this but its is a bit harse to setup and did not succeed because i want it to work with a 'normal' custom quirk. Also contacted Wenzhi but they dont reply on the ZHA integration. 

https://github.com/wzwenzhi/Wenzhi-ZigBee2mqtt/blob/main/wenzhi_tuya_M100_240704.js
https://github.com/Koenkk/zigbee-herdsman-converters

Have you got any solution on the QustomQuirk, otherwhise i have to try the the MQTT option again.

danpeig, 2025-01-09:
[ts0601_motion.zip](https://github.com/user-attachments/files/18367883/ts0601_motion.zip)

I got a modification from the standard Tuya Quirk to get it working.

Please see if it works for you as well so I can commit to the main repository.

MyUnknownSelf, 2025-01-09:
Seems to work quite nice!!!!!
- distance is working and so does the illumination. 
I will do some testing further tomorrow!

[image]

Igorgro, 2025-01-10:
Looks like it is working as expected. At least it correctly detects why I enter and exit the room. Will test more precisely in a couple of days.

danpeig, 2025-01-10:
Updated to adjust the limits and steps of the steps. This is the version I will submit to the repository.

[ts0601_motion_v2.zip](https://github.com/user-attachments/files/18376655/ts0601_motion_v2.zip)

egfku, 2025-01-10:
Hi there,

Thanks for your efforts on this!  I recently purchased the ceiling version of this sensor (WZ-M100-C) and came across your Quirk after some searching.  I was able to add your v2 update to my ZHA, but the values are coming up as Unknown:

[image]

Can you suggest any tips for debugging?  This is my first time using a Quirk.

danpeig, 2025-01-10:
You should not have a message in your log saying that the firmware was disabled. 

> Hi there,
>
> Thanks for your efforts on this!  I recently purchased the ceiling version of this sensor (WZ-M100-C) and came across your Quirk after some searching.  I was able to add your v2 update to [...]

egfku, 2025-01-10:
turns out I needed to delete the device and re-add, thank you!

[image]

f18m, 2025-01-13:
Just wanted to add that I also loaded this (https://github.com/zigpy/zha-device-handlers/issues/3472#issuecomment-2582579494) version of the quirk and it works just fine with my __TZE204_laokfqwu
Thanks!

closed 2025-01-26 by TheJulianJES