zigpy/zha-device-handlers #3476: [Device Support Request] Tuya Zigbee 3.0 Indoor Temperature and Humidity Sensor with Backlit Display (_TZE200_vvmbj46n)

AndersRipa, 2024-10-31:
### Problem description

The Tuya Zigbee 3.0 Indoor Temperature and Humidity Sensor with Backlit Display [(link)](https://www.banggood.com/Tuya-WiFi-or-Zigbee-Smart-Home-Temperature-Humidity-Sensor-with-Backlight-Hygrometer-Thermometer-Monitoring-Sensor-Support-Alexa-Google-Home-APP-Control-p-2009653.html?rmmds=myorder&cur_warehouse=CN&ID=6294241)  which shows up as 

> TS0601
> av _TZE200_vvmbj46n
> Zigbee info
> IEEE: a4:c1:38:cc:1a:7e:61:82
> Nwk: 0x68bf
> Device Type: EndDevice
> LQI: 255
> RSSI: -81
> Senast sedd: 2024-10-31T17:23:13
> Strömkälla: Battery or Unknown

in HomeAsssistant Core 2024.10.4

Don't have any sensors defined

### Solution description

I would like to access its Hygrometer and Thermometer sensors and battery values

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_vvmbj46n",
  "model": "TS0601",
[... 620 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

Please let me know if additional information is needed!

madkatz01, 2024-11-16:
Would love to see this fixed as well, having the same issue as above.

Bolkarr, 2025-01-18:
Did you find a way to integrate this sensor?

AndersRipa, 2025-01-18:
No unfortunely!