zigpy/zha-device-handlers #3477: [Device Support Request] Nous E6  Temperature and Humidity Sensor with Backlit Display (TS0601 by _TZE284_wtikaxzs)

duvaljm, 2024-10-31:
### Problem description

The Nous E6 is not supported. 

I tried the quirk from https://github.com/zigpy/zha-device-handlers/issues/1702 which was apparently for the same Nous E6, but with a different supplier name : _TZE200_nnrfa68v as opposed to _TZE284_wtikaxzs for mine. Unfortunately it did not work, including using the last quirk from 2023-03 and even when changing the "model info".

There is some DEBUG daata in the log such as "
 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'Nous_E6.TuyaNousE6TempHumiditySensor'>
 DEBUG (MainThread) [zigpy.quirks] Fail because input cluster mismatch on at least one endpoint

Unfortunately, I'm new to the quirks world and could not propose any corrections.

### Solution description

Would be great to have a quirk for that device.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_wtikaxzs",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

[... 584 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

ludgerbeckmann, 2024-12-01:
I have the same device _TZE284_wtikaxzs and the same problem. But it works with ZigBee2MQTT.

SytzeAndr, 2024-12-09:
Hi,

I got the same problem and managed to get something working. I took the quirk from https://github.com/zigpy/zha-device-handlers/issues/1702#issuecomment-1451493635 and had to change two things:

### 1. change manufacturer to from `_TZE200_nnrfa68v` to `_TZE284_wtikaxzs`

Pretty straightforward, replace 

```python
        MODELS_INFO: [("_TZE200_nnrfa68v", "TS0601")]
```

with

```python
        MODELS_INFO: [("_TZE284_wtikaxzs", "TS0601")],
 ```

### 2. add the missing input cluster

I noticed the signature did not match with this new model as it has an extra input cluster with id `ed00`, i.e. 60672. When I added this id to the input cluster, it was recognized correctly.

This is probably a bit hacky way to get it working as this will probably discard any information within that 60672 cluster. So maybe someone with a bit more knowledge of quirks can add this input cluster in a "nice" way.

Anyway, I replaced

```python
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TemperatureHumidityManufCluster.cluster_id,
                ],
```

with

```python
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    60672,
                    TemperatureHumidityManufCluster.cluster_id,
                ],
```

After doing that, it seems to be picked up correctly.

duvaljm, 2024-12-10:
Great @SytzeAndr, thanks for the hack!

I did it and - not sure this was necessary - added the additional line (",60672") also in the "replacement" part, between "TuyaRelativeHumidity" and "TuyaPowerConfigurationCluster2AAA".

This works anyway!

Maetzi87, 2024-12-11:
Wow @SytzeAndr, your solution works for me too, thank you very much! 😘

stematon, 2025-01-03:
Works for me, too. I added copy of the existing class `TuyaNousE6TempHumiditySensor` so that I can use old _TZE200_nnrfa68v along with the new _TZE284_wtikaxzs sensors. 
The new class: 
```python
class TuyaNousE6TempHumiditySensorNew(CustomDevice):
    """Custom device representing the new version of the tuya temp and humidity sensor with a screen (NOUS E6 with manufacturer ID _TZE284_wtikaxzs). 
    Newer device from manufacturer _TZE284_wtikaxzs"""
    signature = {
        # <SimpleDescriptor endpoint=1, profile=260, device_type=81
        # device_version=1
        # input_clusters=[4, 5, 61184, 0]
        # output_clusters=[25, 10]>
        MODELS_INFO: [("_TZE284_wtikaxzs", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG, # this is how the device reports itself
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    60672,
                    TemperatureHumidityManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TemperatureHumidityManufCluster,  # Single bus for temp, humidity, and battery
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                    60672,
                    TuyaPowerConfigurationCluster2AAA,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }
```

I still don't get the time in the new devices, any idea?

Maetzi87, 2025-01-03:
I simply made a second file in the custom quirks folder to use both types. Works fine including time.
[image]

Nous_E6_2.py:

`"""Tuya temp and humidity sensor with screen."""

from typing import Dict

################## clean this up
import zigpy.types as t
from zigpy.zcl import foundation
from zhaquirks.tuya import TuyaTimePayload, TuyaCommand, TuyaDPType
import datetime
from typing import Tuple, Optional, Union
##################

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, Groups, Ota, Scenes, Time, AnalogOutput
from zigpy.zcl.clusters.measurement import RelativeHumidity, TemperatureMeasurement

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
    SKIP_CONFIGURATION,
)
from zhaquirks.tuya import TuyaLocalCluster, TuyaPowerConfigurationCluster2AAA
from zhaquirks.tuya.mcu import DPToAttributeMapping, TuyaMCUCluster

TUYA_SET_TIME = 0x24

# NOTES:
# The data comes in as a string on cluster, if there is nothing set up you may see these lines in the logs:
# Unknown message (b'19830100a40102000400000118') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (a4:c1:38:d0:18:8b:64:aa, 1)'
#                                          28.0 degrees
# Unknown message (b'19840100a5020200040000022c') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (a4:c1:38:d0:18:8b:64:aa, 1)'
#                                          55.6% humid
# Unknown message (b'19850100a60402000400000064') on cluster 61184: unknown endpoint or cluster id: 'No cluster ID 0xef00 on (a4:c1:38:d0:18:8b:64:aa, 1)'
#                                          100% battery

class TemperatureUnitConvert(t.enum8):
    """Tuya Temp unit convert enum."""

    Celsius = 0x00
    Fahrenheit = 0x01

class TuyaTemperatureMeasurement(TemperatureMeasurement, TuyaLocalCluster):
    """Tuya local TemperatureMeasurement cluster."""

    attributes = TemperatureMeasurement.attributes.copy()
    attributes.update(
        {
            0x8001: ("temp_unit_convert", t.enum8),
            0x8002: ("alarm_max_temperature", t.Single),
            0x8003: ("alarm_min_temperature", t.Single),
            0x8004: ("temperature_sensitivity", t.Single),
        }
    )

class TuyaRelativeHumidity(RelativeHumidity, TuyaLocalCluster):
    """Tuya local RelativeHumidity cluster."""

class TemperatureHumidityManufCluster(TuyaMCUCluster):
    """Tuya Manufacturer Cluster with Temperature and Humidity data points."""
    
[... 92 lines cut ...]
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }

    replacement = {
        SKIP_CONFIGURATION: True,
        ENDPOINTS: {
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.TEMPERATURE_SENSOR,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    TemperatureHumidityManufCluster,  # Single bus for temp, humidity, and battery
                    TuyaTemperatureMeasurement,
                    TuyaRelativeHumidity,
                    60672,
                    TuyaPowerConfigurationCluster2AAA,
                ],
                OUTPUT_CLUSTERS: [Ota.cluster_id, Time.cluster_id],
            }
        },
    }`

Maybe try readding the device? Or sometimes a system-restart helps since its more 'intense' than a normal restart...

RobinR1, 2025-06-17:
I wanted to add that the Nous LCD temperature and humidity sensor E6 that I recently bought initially did not work with this quirk and it turns out I have a _TZE**200**_wtikaxzs TS0601 instead of a _TZE200_nnrfa68v or a _TZE284_wtikaxzs.
Also in this version an extra input cluster is present but this time it is 0xe**f**00 instead of 0xed00
Adapting the quirk :
```python
MODELS_INFO: [("_TZE200_wtikaxzs", "TS0601")],
```
and
```python
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    61184,
                    TemperatureHumidityManufCluster.cluster_id,
                ],
```
makes it also work on this device.