zigpy/zha-device-handlers #3479: [BUG] custom quirk doesn't load

nicktrigger, 2024-11-03:
### Bug description

I created custom quirks for 2 tuya valves from the esmxisting tuya valve quirk but they won't load. I screenshot 2 examples below.

Can anyone help to tell me why the custom quirk doesn't load?

### Steps to reproduce

Edit quirk
Save as new .py file in custom quirk folder
Reboot system
Re-add device 5+ times
No associated quirk loads

### Expected behavior

Should load quirk from product name

### Screenshots/Video

[image]
[image]
[image]
[image]

### Device signature

1
2
3
4
5
6
7
8
9
10
11
12
13
14
15
16
17
18
19
20
21
22
23
24
25
26
27
28
29
30
31
32
33
34
35
36
⌄
⌄
⌄
⌄
⌄
⌄
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_qtnjuoae",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>
[](url)

nicktrigger, 2024-11-03:
[image]
[image]
[image]
[image]

nicktrigger, 2024-11-03:
[zha-01J9XY83W61D8PMQM9FG20JYW5-_TZE204_qtnjuoae TS0601-489f92b4037a7c303964e85b8d87ffb4.json](https://github.com/user-attachments/files/17610191/zha-01J9XY83W61D8PMQM9FG20JYW5-_TZE204_qtnjuoae.TS0601-489f92b4037a7c303964e85b8d87ffb4.json)
[config_entry-zha-01J9XY83W61D8PMQM9FG20JYW5.json](https://github.com/user-attachments/files/17610192/config_entry-zha-01J9XY83W61D8PMQM9FG20JYW5.json)

nicktrigger, 2024-11-03:
Error from the log

Logger: zhaquirks
Source: /usr/local/lib/python3.12/site-packages/zhaquirks/__init__.py:464
First occurred: 20:16:14 (1 occurrences)
Last logged: 20:16:14

Unexpected exception importing custom quirk 'ts0601_TZE204_qtnjuoae_valve'
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zhaquirks/__init__.py", line 462, in setup
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/config/custom_zha_quirks/ts0601_TZE204_qtnjuoae_valve.py", line 25, in <module>
    from zhaquirks.tuya.mcu import (
ImportError: cannot import name 'TuyaDPType' from 'zhaquirks.tuya.mcu' (/usr/local/lib/python3.12/site-packages/zhaquirks/tuya/mcu/__init__.py)

nicktrigger, 2024-11-03:
I can't upload the full log here as it won't accept .bin files

prairiesnpr, 2024-11-04:
> Error from the log
> 
> Logger: zhaquirks Source: /usr/local/lib/python3.12/site-packages/zhaquirks/**init**.py:464 First occurred: 20:16:14 (1 occurrences) Last logged: 20:16:14
> 
> Unexpected exception importing custom quirk 'ts0601_TZE204_qtnjuoae_valve' Traceback (most recent call last): File "/usr/local/lib/python3.12/site-packages/zhaquirks/**init**.py", line 462, in setup spec.loader.exec_module(module) File "", line 995, in exec_module File "", line 488, in _call_with_frames_removed File "/config/config/custom_zha_quirks/ts0601_TZE204_qtnjuoae_valve.py", line 25, in from zhaquirks.tuya.mcu import ( ImportError: cannot import name 'TuyaDPType' from 'zhaquirks.tuya.mcu' (/usr/local/lib/python3.12/site-packages/zhaquirks/tuya/mcu/**init**.py)

The logs are telling you that your import is wrong.

Should be: `from zhaquirks.tuya import TuyaDPType`

nicktrigger, 2024-11-04:
Annoyingly this just leads to another issue.

Logger: zhaquirks
Source: /usr/local/lib/python3.12/site-packages/zhaquirks/__init__.py:464
First occurred: 11:41:50 (1 occurrences)
Last logged: 11:41:50

Unexpected exception importing custom quirk 'ts0601_TZE204_qtnjuoae_valve'
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zhaquirks/__init__.py", line 462, in setup
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/config/custom_zha_quirks/ts0601_TZE204_qtnjuoae_valve.py", line 63, in <module>
    class TuyaValveManufCluster(TuyaMCUCluster):
  File "/config/config/custom_zha_quirks/ts0601_TZE204_qtnjuoae_valve.py", line 96, in TuyaValveManufCluster
    1: DPToAttributeMapping(
       ^^^^^^^^^^^^^^^^^^^^^
TypeError: DPToAttributeMapping.__init__() got an unexpected keyword argument 'dp_type'

I tired another approach,  using the code from ZHA quirk for tuya valves to create a custom quirk, adding my products but it leads to a different error:

Logger: zhaquirks
Source: /usr/local/lib/python3.12/site-packages/zhaquirks/__init__.py:464
First occurred: 11:41:50 (1 occurrences)
Last logged: 11:41:50

Unexpected exception importing custom quirk 'Valves'
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zhaquirks/__init__.py", line 462, in setup
    spec.loader.exec_module(module)
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/config/custom_zha_quirks/Valves.py", line 27, in <module>
    from zhaquirks.tuya.builder import TuyaQuirkBuilder
ModuleNotFoundError: No module named 'zhaquirks.tuya.builder'

Is there a way to simply copy the existing tuya valve quirk, add my products and then use as a custom quirk until they get added officially?

My devices are ts0601_TZE200_akjefhj5 and ts0601_TZE204_qtnjuoae

nicktrigger, 2024-11-04:
This is what i tried but it gives the error above

"""Collection of Tuya Valve devices e.g. water valves, gas valve etc."""

from datetime import datetime, timedelta, timezone

from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.quirks.v2.homeassistant import UnitOfTime
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
import zigpy.types as t
from zigpy.zcl.clusters.general import Basic, Groups, Identify, OnOff, Ota, Scenes, Time
from zigpy.zcl.clusters.smartenergy import Metering

from zhaquirks import DoublingPowerConfigurationCluster
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import (
    EnchantedDevice,
    TuyaLocalCluster,
    TuyaPowerConfigurationCluster,
)
from zhaquirks.tuya.builder import TuyaQuirkBuilder
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaMCUCluster,
    TuyaOnOff,
    TuyaPowerConfigurationCluster4aa,
)

class TuyaValveWaterConsumed(Metering, TuyaLocalCluster):
    """Tuya Valve Water consumed cluster."""

    VOLUME_LITERS = 0x0007
    WATER_METERING = 0x02

    """Setting unit of measurement."""
    _CONSTANT_ATTRIBUTES = {
        0x0300: VOLUME_LITERS,
        0x0306: WATER_METERING,
    }

class TuyaValveManufCluster(TuyaMCUCluster):
    """On/Off Tuya cluster with extra device attributes."""

    attributes = TuyaMCUCluster.attributes.copy()
    attributes.update(
        {
            0xEF01: ("time_left", t.uint32_t, True),
            0xEF02: ("state", t.enum8, True),
            0xEF03: ("last_valve_open_duration", t.uint32_t, True),
            0xEF04: ("dp_6", t.uint32_t, True),
            0xEF05: ("valve_position", t.uint32_t, True),
        }
    )

    dp_to_attribute: dict[int, DPToAttributeMapping] = {
        1: DPToAttributeMapping(
            TuyaOnOff.ep_attribute,
            "on_off",
        ),
        5: DPToAttributeMapping(
            TuyaValveWaterConsumed.ep_attribute,
            "current_summ_delivered",
[... 350 lines cut ...]
    .also_applies_to("_TZE204_qtnjuoae", "TS0601")

    .tuya_number(
        dp_id=104,
        attribute_name="irrigation_target",
        type=t.uint32_t,
        min_value=0,
        max_value=GIEX_12HRS_AS_SEC,
        step=1,
        translation_key="irrigation_target",
        fallback_name="Irrigation Target",
    )
    .tuya_number(
        dp_id=105,
        attribute_name="irrigation_interval",
        type=t.uint32_t,
        min_value=0,
        max_value=GIEX_12HRS_AS_SEC,
        step=1,
        unit=UnitOfTime.SECONDS,
        translation_key="irrigation_interval",
        fallback_name="Irrigation Interval",
    )
    .add_to_registry()
)

prairiesnpr, 2024-11-04:
> Annoyingly this just leads to another issue.
> 
> Logger: zhaquirks Source: /usr/local/lib/python3.12/site-packages/zhaquirks/**init**.py:464 First occurred: 11:41:50 (1 occurrences) Last logged: 11:41:50
> 
> Unexpected exception importing custom quirk 'ts0601_TZE204_qtnjuoae_valve' Traceback (most recent call last): File "/usr/local/lib/python3.12/site-packages/zhaquirks/**init**.py", line 462, in setup spec.loader.exec_module(module) File "", line 995, in exec_module File "", line 488, in _call_with_frames_removed File "/config/config/custom_zha_quirks/ts0601_TZE204_qtnjuoae_valve.py", line 63, in class TuyaValveManufCluster(TuyaMCUCluster): File "/config/config/custom_zha_quirks/ts0601_TZE204_qtnjuoae_valve.py", line 96, in TuyaValveManufCluster 1: DPToAttributeMapping( ^^^^^^^^^^^^^^^^^^^^^ TypeError: DPToAttributeMapping.**init**() got an unexpected keyword argument 'dp_type'
> 
> I tired another approach, using the code from ZHA quirk for tuya valves to create a custom quirk, adding my products but it leads to a different error:
> 
> Logger: zhaquirks Source: /usr/local/lib/python3.12/site-packages/zhaquirks/**init**.py:464 First occurred: 11:41:50 (1 occurrences) Last logged: 11:41:50
> 
> Unexpected exception importing custom quirk 'Valves' Traceback (most recent call last): File "/usr/local/lib/python3.12/site-packages/zhaquirks/**init**.py", line 462, in setup spec.loader.exec_module(module) File "", line 995, in exec_module File "", line 488, in _call_with_frames_removed File "/config/config/custom_zha_quirks/Valves.py", line 27, in from zhaquirks.tuya.builder import TuyaQuirkBuilder ModuleNotFoundError: No module named 'zhaquirks.tuya.builder'
> 
> Is there a way to simply copy the existing tuya valve quirk, add my products and then use as a custom quirk until they get added officially?
> 
> My devices are ts0601_TZE200_akjefhj5 and ts0601_TZE204_qtnjuoae

TuyaQuirkBuilder was just added, it's only available on the beta. Will be in 2024.11.

prairiesnpr, 2024-11-04:
Yes: https://github.com/zigpy/zha-device-handlers/commits/dev/zhaquirks/tuya/ts0601_valve.py

Better to wait till the 6th and just upgrade to 2024.11

nicktrigger, 2024-11-04:
Ok thank you! I will wait.
With the new release I'd still need to make a custom quirk for
_TZE200_akjefhj5 and _TZE204_qtnjuoae to work, am I right?

closed 2024-11-08 by nicktrigger