zigpy/zha-device-handlers #3488: Docs: How to remove local quirk once it's released

storm1er, 2024-11-06:
Hello all,

Explicit title, not a bug report, not a device request.

I used the custom quirk for [RDM004](https://github.com/zigpy/zha-device-handlers/pull/3075) but now that it's released, I was wondering how to use the provided one instead of the custom.

I could guess and try, but having an explicit how-to in the readme/docs would be nice :)

Thanks!

TheJulianJES, 2024-11-06:
You should be able to just remove the custom quirk file. The inbuilt one will take over then.
Also, that quirk is only included in Home Assistant 2024.10.4 and later (currently still in beta, will release in a few hours or so).

Hedda, 2024-11-07:
@TheJulianJES maybe steps for that could be a good fit under this related docs PR suggestion?:

- https://github.com/zigpy/zha-device-handlers/pull/2419

storm1er, 2024-11-07:
@Hedda Almost ahah, this PR include how to test, not how to *untest*

You have written this point well in [your latest comment](https://github.com/zigpy/zha-device-handlers/pull/2419#issuecomment-2461789547)

Closing this, since a PR is already open

closed 2024-11-07 by storm1er