zigpy/zha-device-handlers #3502: [Device Support Request] TZE200_9xfjxap

ahgartner, 2024-11-10:
### Problem description

ZHA adds a device when pairing but no sensors or controls are loaded. Unpairing/pairing makes no visible difference. Adding it to custom quirk doesn't work

### Solution description

I would like to pair my device and have all entities and controls shown

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_9xfjixap",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.11.1",
[... 571 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

DarkNeroXIII, 2024-11-12:
I have the exact same issue. So far i tried this custom quirk:
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_trv_sas.py

I added the device ID at the bottom under Model_Info in the list. So far i can tell some functions do show up but they do not work or report properly.

The Hardware looks the same as those mentioned in the Model_Info list, however it looks like its again something special and different than the rest.

Looking still for a solution, sorry that i cant do more as i am not good in coding or changing the code provided by the quirks

ahgartner, 2024-11-12:
@DarkNeroXIII found this here that worked for me. Just dont understand Why it is Not implemented yet

DarkNeroXIII, 2024-11-12:
Thank you so much, this one worked fine for me. I was searching the half day for a solution in mulitple websites

grintroy, 2024-11-12:
@DarkNeroXIII

> "the action switch/turn_on could not be executed. 'ZCLAttributeDef' object is not subscriptable"

The same was for me, and I'm not even sure what this switch is supposed to do. I just disabled it, I think.

Elektronix-LTD, 2024-11-13:
The same problem for me with _TZE200_9xfjixap TS0601 (Radiator thermostat). Worst part is that it is the first device I paired with Home Assistant ever and I have problems since beginning... :-/

DarkNeroXIII, 2024-11-13:
@Elektronix-LTD do you still have problems? If yes, whats exactly the issue? I am by far not a master for coding or the HomeAssistant itself but i am running it since half a year, encountered some issues (mostly with Tuya Zigbee devices) but was alsways able to solve them by (timeconsuming) research :D

jacekk015, 2024-11-13:
You all can try to use the file from
https://github.com/jacekk015/zha_quirks/issues/63#issuecomment-2473156159

99% usage of switch is Child Lock

closed 2025-01-16 by TheJulianJES