zigpy/zha-device-handlers #3507: [Device Support Request] Smoke Sensor Nous E8 - _TZE284_0zaf1cr8

chcoust, 2024-11-11:
### Problem description

I recently bought this smoke detector from Amazon, not being recognized in HA.

### Solution description

I managed to get it recognized with a small tweak to ts0601_smoke.py.
There is a custom Tuya id in the signature that needed to be added.
I dont know if we can have multiple signature in ts0601_smoke.py or need to create a new one ?

The value is here  : https://github.com/zigpy/zha-device-handlers/blob/728ee427263d89a9c9186eacefc90e4994d81b82/zhaquirks/tuya/__init__.py#L36

TUYA_CLUSTER_ED00_ID = 0xED00

### Screenshots/Video

<details><summary>Screenshots of working device trigger test</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 17132,
    "manufacturer": "_TZE284_0zaf1cr8",
    "model": "TS0601",
    "name": "_TZE284_0zaf1cr8 TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "manufacturer_code": 4098,
    "power_source": "Battery or Unknown",
    "lqi": 248,
    "rssi": -38,
    "last_seen": "2024-11-11T12:43:55",
    "available": true,
    "device_type": "EndDevice",
    "signature": {
      "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4098, maximum_buffer_size=82, maximum_incoming_transfer_size=82, server_mask=11264, maximum_outgoing_transfer_size=82, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
      "endpoints": {
        "1": {
          "device_type": "0x0051",
          "profile_id": "0x0104",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xed00",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        }
      },
      "manufacturer": "_TZE284_0zaf1cr8",
      "model": "TS0601"
    },
```

</details>

### Logs

[... 315 lines cut ...]
    signature = {
        MODELS_INFO: [
            ("_TZE284_0zaf1cr8", "TS0601"),
        ],
        ENDPOINTS: {
            1: {
                PROFILE_ID: 0x0104,
                DEVICE_TYPE: 0x0051,
                INPUT_CLUSTERS: [
                    Basic.cluster_id, # 0x0000,
                    Groups.cluster_id, # 0x0004,
                    Scenes.cluster_id, # 0x0005,
                    0xed00, # 0xed00,
                    TuyaManufCluster.cluster_id, # 0xef00
                ],
                OUTPUT_CLUSTERS: [
                    Time.cluster_id,
                    Ota.cluster_id,
                ],
            },
        },
    }
```

</details>

closed 2024-12-09 by TheJulianJES

kovacmichal, 2025-06-02:
Hi @chcoust I bought same sensor few weeks ago, but it looks like device identification is different comparing to your debug info. Please will you find some to check mine https://github.com/zigpy/zha-device-handlers/issues/4098? Thanks Michal

bwip, 2025-09-24:
I have three of those sensors  _TZE284_0zaf1cr8. Quite new.

All of them report batter_low to be True with a value of 1:

[image]

When I look at the zigbee2mqtt definition

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/devices/tuya.ts
```json
    {
        fingerprint: tuya.fingerprint("TS0601", ["_TZE200_0zaf1cr8", "_TZE204_ntcy3xu1", "_TZE284_0zaf1cr8"]),
        model: "TS0601_smoke_1",
        vendor: "Tuya",
        description: "Smoke sensor",
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        configure: tuya.configureMagicPacket,
        exposes: [e.smoke(), e.tamper(), e.battery_low()],
        meta: {
            tuyaDatapoints: [
                [1, "smoke", tuya.valueConverter.trueFalse0],
                [4, "tamper", tuya.valueConverter.raw],
                [14, "battery_low", tuya.valueConverter.trueFalse0],
            ],
        },
        whiteLabel: [tuya.whitelabel("Nous", "E8", "Smoke sensor", ["_TZE200_0zaf1cr8", "_TZE284_0zaf1cr8"])],
    },

```

They consider a value of 0 to be true.

Needs this value to be inverted?

BR,
Bernhard