zigpy/zha-device-handlers #3515: [Device Support Request] _TZE204_rtrmfadk TRV

felixwerner, 2024-11-14:
### Problem description

TS0601  _TZE204_rtrmfadk 
TRV is not showing any entities.

tested this quirk: [https://github.com/jacekk015/zha_quirks/issues/62](https://github.com/jacekk015/zha_quirks/issues/62) and added a device: ("_TZE204_rtrmfadk", "TS0601"),
still nothing to use.

### Solution description

Be able to controle the device as a climate TRV

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Unknown",
    "version": "2024.11.1",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.12.7",
    "docker": false,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.21-v8+",
    "run_as_root": false
  },
  "custom_components": {
    "hacs": {
      "documentation": "https://hacs.xyz/docs/configuration/start",
      "version": "2.0.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "better_thermostat": {
      "documentation": "https://github.com/KartoffelToby/better_thermostat",
      "version": "1.6.1",
      "requirements": []
    }
  },
  "integration_manifest": {
    "domain": "zha",
    "name": "Zigbee Home Automation",
    "after_dependencies": [
      "hassio",
      "onboarding",
[... 545 lines cut ...]
</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

[aliexpress](https://de.aliexpress.com/item/1005007023705249.html)

jacekk015, 2024-11-14:
Apparently you did something wrong, since in post mentioned user reports working state.

felixwerner, 2024-11-17:
correct.

closed 2024-11-17 by felixwerner