zigpy/zha-device-handlers #3516: [Device Support Request] Avatto TRV06 TS0601 _TZE284_c6wv4xyo

LSz79, 2024-11-16:
### Problem description

I bought a TRV from [Aliexpress](https://de.aliexpress.com/item/1005004696707285.html): it comes with a new, yet not supported manufacturer ID. 
It also **implements input cluster 0xED00**, so it's signature doesn't match with any available TRV quirk. 

### Solution description

I tried to implement a quirk myself:  it has been recognized, the TRV is paring, but not displaying any controls/entities. 
I suppose the available implementation of 0xED00 cluster (ZosungIRTransmit) is not compatible with this TRV.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]
[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_c6wv4xyo",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

[... 676 lines cut ...]
[36m2024-11-16 16:20:19.279 DEBUG (bellows.thread_0) [bellows.ash] Received data 72eaa1a92b2a144ced7e[0m
[36m2024-11-16 16:20:19.280 DEBUG (bellows.thread_0) [bellows.ash] Received frame DataFrame(frm_num=7, re_tx=0, ack_num=2, ezsp_frame=b'\xa8\x80\x01\x7f\x00\x01')[0m
[36m2024-11-16 16:20:19.280 DEBUG (bellows.thread_0) [bellows.ash] Sending frame AckFrame(res=0, ncp_ready=0, ack_num=0) + FLAG[0m
[36m2024-11-16 16:20:19.280 DEBUG (bellows.thread_0) [bellows.ash] Sending data  8070787e[0m
[36m2024-11-16 16:20:19.281 DEBUG (MainThread) [bellows.ezsp.protocol] Received command getExtendedTimeout: {'extendedTimeout': <Bool.true: 1>}[0m
[36m2024-11-16 16:20:19.282 DEBUG (MainThread) [bellows.ezsp.protocol] Sending command  sendUnicast: () {'type': <EmberOutgoingMessageType.OUTGOING_DIRECT: 0>, 'indexOrDestination': 0x53E7, 'apsFrame': EmberApsFrame(profileId=0, clusterId=33, sourceEndpoint=0, destinationEndpoint=0, options=<EmberApsOption.APS_OPTION_RETRY|APS_OPTION_ENABLE_ROUTE_DISCOVERY: 320>, groupId=0, sequence=6), 'messageTag': 52, 'messageContents': b'\x06;\t\xf2\xb0\xb48\xc1\xa4\x01\x00\xef\x03@\xeb\xc9$\x00K\x12\x00\x01'}[0m
[36m2024-11-16 16:20:19.282 DEBUG (MainThread) [zigpy.appdb] Error handling '_save_attribute' event with (a4:c1:38:b4:b0:f2:09:3b, 1, <ClusterType.Server: 0>, 0, 5, 'TS0601', datetime.datetime(2024, 11, 16, 15, 20, 19, 243127, tzinfo=datetime.timezone.utc)) params: FOREIGN KEY constraint failed[0m
[36m2024-11-16 16:20:19.287 DEBUG (bellows.thread_0) [bellows.ash] Sending frame DataFrame(frm_num=2, re_tx=False, ack_num=0, ezsp_frame=b'\xa9\x00\x014\x00\x00\xe7S\x00\x00!\x00\x00\x00@\x01\x00\x00\x064\x16\x06;\t\xf2\xb0\xb48\xc1\xa4\x01\x00\xef\x03@\xeb\xc9$\x00K\x12\x00\x01') + FLAG[0m
[36m2024-11-16 16:20:19.287 DEBUG (bellows.thread_0) [bellows.ash] Sending data  20eb21a9602a15550a944a04aa5592099d4e27add9d861b0f434d33dc4bf9ba6eb22dd2f6436e3db9ec0698d66f17e[0m
[36m2024-11-16 16:20:19.292 DEBUG (bellows.thread_0) [bellows.ash] Received data 02eab1a97d312a15b658914e24ab5593499c30e767df199874f5de5583fc7d5e1ec9f7fcbc7e[0m
[36m2024-11-16 16:20:19.293 DEBUG (bellows.thread_0) [bellows.ash] Received frame DataFrame(frm_num=0, re_tx=0, ack_num=2, ezsp_frame=b'\xa8\x90\x01E\x00\x00\x04\x01\x05\x04\x01\x01\x00\x01\x00\x00~\xc0\xcc2\xd7\xff\xff\x08\x186\n\x00\x00!n\x1c')[0m
[36m2024-11-16 16:20:19.294 DEBUG (bellows.thread_0) [bellows.ash] Sending frame AckFrame(res=0, ncp_ready=0, ack_num=1) + FLAG[0m
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Flobuu, 2024-11-24:
I got the same: 

TS0601;  _TZE284_o3x45p96

Did you get it to work ?

matteohoeren, 2024-11-24:
@Flobuu Nooo unfortunately not :( Also tried Zigbee2MQTT, seems the also don't have the correct device handler. I might try to code a custom quirk for it, but that might not be done until end oft this year. If i do, i'll make sure to update you guys. Hope anyone finds a solution before me tho 😆

Flobuu, 2024-11-24:
I'm quite new here. Tried out some quirks and added the Informations to the quirks, but it didn't work out. 
Hope the same! I think the Thermostat version is quite new. 
Do we need to post more Information to the devices here ? That someone else can help us ?

LordMGR, 2024-11-25:
It looks like, Zigbee2MQTT has a solution to it, but I was not able, to port it to zha.
They seem to handle it like some other valves: https://github.com/Koenkk/zigbee-herdsman-converters/pull/8339/commits/9bfc71aab352c1a9a2aa74b823be9878c8ae2e2b
Maybe someone sees something I don't.

matteohoeren, 2024-11-26:
@LordMGR Soo i've got good and bad news. I did make it work with Zigbee2MQTT. However i have little idea how to port it into ZHA. What step did you fail when trying to port it to ZHA? I might give it another shot then :)

matteohoeren, 2024-11-26:
I tried the above as i said, and i got this once again: 

```
2024-11-26 15:20:24.056 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for _TZE284_o3x45p96 TS0601 (a4:c1:38:98:49:40:9e:19)
2024-11-26 15:20:24.057 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'ts0601_trv_me167.ME167'>
2024-11-26 15:20:24.057 DEBUG (MainThread) [zigpy.quirks] Fail because input cluster mismatch on at least one endpoint
```

lopikamper, 2025-01-05:
Hi, coming back to this topic after Christmas. 
This is my progress so far - I have made some over today but the entities are for some reason unavailable. 

Here is what I did:
Created a custom quirk - originally from post #3445 - here is the [actual config](https://github.com/lopikamper/zha/blob/main/ts0601_trv_me167.py)
 - I have defined the missing cluster 0xed00 - according to diag it has no attributes so dummy cluster
`class ZosungIRTransmitCluster(Cluster)`
 - added mine device _TZE284_o3x45p96
 - uploaded to zha custom quirks in HA
 - removed the device and paired again

Result:
Entities are now shown in HA but all are unavailable even with device connected to HA - screenshot
Device is connected and on any action (changing the temperature on valve) it communicates with the gateway
There are no errors in logs - here is an event from ZHA from this device

Conclusion:
To be honest I am not really sure if I made some progress or not. I believe I fixed the missing cluster but I have no clue why the entities are unavailable. I am not sure if the event below can indicate next step or not.

`2025-01-05 20:36:57.405 DEBUG (MainThread) [zha] Emitting event zha_channel_configure_reporting with data ClusterConfigureReportingEvent(cluster_name='ME167Thermostat', cluster_id=513, attributes={'local_temperature': {'min': 30, 'max': 900, 'id': 'local_temperature', 'name': 'local_temperature', 'change': 25, 'status': 'SUCCESS'}, 'occupied_cooling_setpoint': {'min': 30, 'max': 900, 'id': 'occupied_cooling_setpoint', 'name': 'occupied_cooling_setpoint', 'change': 25, 'status': 'SUCCESS'}, 'occupied_heating_setpoint': {'min': 30, 'max': 900, 'id': 'occupied_heating_setpoint', 'name': 'occupied_heating_setpoint', 'change': 25, 'status': 'SUCCESS'}, 'unoccupied_cooling_setpoint': {'min': 30, 'max': 900, 'id': 'unoccupied_cooling_setpoint', 'name': 'unoccupied_cooling_setpoint', 'change': 25, 'status': 'SUCCESS'}, 'unoccupied_heating_setpoint': {'min': 30, 'max': 900, 'id': 'unoccupied_heating_setpoint', 'name': 'unoccupied_heating_setpoint', 'change': 25, 'status': 'SUCCESS'}, 'running_mode': {'min': 30, 'max': 900, 'id': 'running_mode', 'name': 'running_mode', 'change': 25, 'status': 'SUCCESS'}, 'running_state': {'min': 30, 'max': 900, 'id': 'running_state', 'name': 'running_state', 'change': 5, 'status': 'SUCCESS'}, 'system_mode': {'min': 30, 'max': 900, 'id': 'system_mode', 'name': 'system_mode', 'change': 25, 'status': 'SUCCESS'}, 'occupancy': {'min': 30, 'max': 900, 'id': 'occupancy', 'name': 'occupancy', 'change': 1, 'status': 'SUCCESS'}, 'pi_cooling_demand': {'min': 30, 'max': 900, 'id': 'pi_cooling_demand', 'name': 'pi_cooling_demand', 'change': 5, 'status': 'SUCCESS'}, 'pi_heating_demand': {'min': 30, 'max': 900, 'id': 'pi_heating_demand', 'name': 'pi_heating_demand', 'change': 5, 'status': 'SUCCESS'}}, `

[image]

prairiesnpr, 2025-01-05:
Do we have the data points for this one or a working z2m converter?

matteohoeren, 2025-01-06:
@lopikamper Thank you so much for further looking into this. I looked at your quirk and i realized that the signature (line 431 following) were commented out. Once i uncomment those, the quirk started working!! As of now, only basic functionality like the heating setpoint work, but that's a great start!
What doesn't work: Battery state, min/max heating point, the switch (i suspect thats used for turning the thermostat on/off)
When i was working with Z2M i saw that this TRV supports schedules which in theory should be accessible via Zigbee. I won't need them though since i'll use HA to control that. 

This is how it looks for me right now:

[image]

matteohoeren, 2025-01-06:
@prairiesnpr Could you explain to me what you mean by data points? Sorry, Zigbee noob here. And yes, there is a working converter for Z2M, i've used that successfully.

prairiesnpr, 2025-01-06:
> @prairiesnpr Could you explain to me what you mean by data points? Sorry, Zigbee noob here. And yes, there is a working converter for Z2M, i've used that successfully.

This might help. https://github.com/zigpy/zha-device-handlers/wiki/Tuya-%E2%80%90-v2-Quirk-with-TuyaQuirkBuilder

We have another trv in work, that should be a good baseline for this one once it's done.

lopikamper, 2025-01-06:
@matteohoeren Thanks for finding the mistake ! I confirm, those entities are not working for me as well. 

Can we somehow test some others ZHA quirks if they have better compatibility already ? Or is it better to go through the link @prairiesnpr just sent ? I am complete newbie to zigbee so sorry if this question is a bit stupid.

I just took the quirk from #3445 not really paying attention to any details to be honest.

prairiesnpr, 2025-01-06:
This is the [quirk](Https://github.com/prairiesnpr/zha-device-handlers/blob/add-Tuya-_TZE204_ogx8u5z6/zhaquirks/tuya/ts0601_trv_v2.py) we are testing in https://github.com/zigpy/zha-device-handlers/issues/3628 If you have the z2m convertor, it should be a good baseline for this one.

rootjumper, 2025-01-10:
Hi, I can confirm PR from @prairiesnpr https://github.com/zigpy/zha-device-handlers/pull/3684 is working for "**_TZE284_o3x45p96**".
I was able to read/write few settings/values for the two additional tuya cluster(s) using ZHA.

Cluter _0x0201_:
- `local_temperature ` 
(read) value "2100" for 21°C, matched value shown on lcd

- `occupied_heating_setpoint `
(read/write) value "2400" for 24°C matched value shown on lcd checked by up/down the value using the hardware rotation, started everytime by the value i set

- `running_state `
(read) off/heat was able to read the values
- `system_mode `
(read/write) off/heat/auto 
setting to "off" shows "OF" on lcd and valve is closing, changing back to "heat" will open the valve, setting to "auto" will show the "clock" icon on lcd but donno what it does.

Cluter _0xef00_:
- `error `
(read/write) forced low battery read error "16", changing battery did not reset value! I had to reset error by writing value 0
- `child_lock `
(read/write) lcd showed LC on lock)
- `frost_protection `
only tested read/write no actual frost_protection was checked
- `scale_protection `
only tested read/write no actual scale_protection was checked

I did not check the "schedule_*" settings due this is my first thermostat i installed 4h ago.
As I am totally new to this HA / Zigbee stuff (12 devices since 2 days) I lack knowledge how to test / verify it as i am still in learing process.

The value for "firmware" is currently missing (dont know why i saw a value without using any quirk for my device)

fyi: Had to digg into the pull request and some other sourcecode to understand whats going on took me around 4h (includes learing) to get my thermostat running in HA using the https://github.com/zigpy/zha-device-handlers/pull/3684 

Here is my stripped [quirk](https://github.com/rootjumper/zha_quirks/blob/main/config/custom_zha_quirks/TS0601_TZE284_o3x45p96.py) i used for testing and finding out (copy/paste from PR and stripped down)

[image]

prairiesnpr, 2025-01-10:
> I did not check the "schedule_*" settings due this is my first thermostat i installed 4h ago.

Thanks for taking the time to test this. I didn't implement the scheduling functionality, z2m has the details, but it would take some effort to bring it over, not certain it's really worth it. Of everything is good, I'll make this PR as ready.

prairiesnpr, 2025-01-10:
> manipulate the actual temperature

You should have a climate entity. It's the top one in your screenshot, click that and you should be able to set temperature, mode, etc.

lopikamper, 2025-01-10:
With my comment I meant the actual temperature that smart valve reports to climate entity - **replace** it with an external temperature sensor of my choice or **apply an offset** to this reporting temperature with some method. Unfortunately I do not see any way to do it directly in climate entity documentation.

prairiesnpr, 2025-01-10:
> apply an offset

Isn't that what `local_temperature_calibration` does? As far as using an external temperature sensor, that's likely best done through the automation system, this device doesn't appear to support it as far as I see.

closed 2025-01-16 by TheJulianJES