zigpy/zha-device-handlers #3524: [Device Support Request] Tuya _TZ3000_zl1kmjqx Temp + Humidity Sensor

m00m007, 2024-11-19:
### Problem description

Device pairs but doesn´t show the temperature and humidity values. Just the values for battery-level and the button for identification is shown. Two entities are disabled
[image]

 

### Solution description

I think the device should maybe added to the device-list to work properly. I did not try to do a custom quirk but this should be also distributed to all people using this devices.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0302",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0402",
        "0x0405"
      ],
      "output_clusters": [
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZ3000_zl1kmjqx",
  "model": "",
  "class": "zhaquirks.tuya.ty0201.TuyaTempHumiditySensorNoModel"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
[... 810 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

m00m007, 2024-12-03:
Hey there, I just activated the 2 missing entities RSSI and RQI. Now the temperature and the humidity is showing up. I´m not getting it :/

ptoews, 2024-12-18:
In my case I had to first read the measurement values for each of temperature/humidity using the "Manage Zigbee Device" settings, and then (re-)enabling RSSI (which apparently does some kind of restart?). 
Now both values are showing for both of the sensors I bought.

closed 2025-08-03 by github-actions[bot]