zigpy/zha-device-handlers #3527: [Device Support Request] TZE200_hvaxb2tc TS0601 Valve

cloudchaserberlin, 2024-11-19:
### Problem description

I can add the device but there are no Sensors. I have allready try to use existent quirks. Sone of them add sensor but there are all in Status „Unknown“

I'm happy to provide further information or data. Unfortunately, I haven't been here long and therefore don't really know much about it.

### Solution description

[image]

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

Mattivdp, 2024-12-02:
Did u get this working ?? I have 6 off the same ones cant do anything with them ?

jabouchleih, 2025-01-16:
I have no problems getting it work with the custom quirks here: 
https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_trv.py#L1803C1-L1804C1 

As you can see, the exact model ID is supported as well.

[image]

AaronLayton, 2025-01-19:
@jabouchleih I am in the same boat and have just added a few of these to my network but no entities exposed.

I did see https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_trv.py#L1803C1-L1804C1 the other day but I am pretty new to Quirks. I believe I followed a setup guide correctly but when I restart the ZHA service it just errors out.

If you have time, would you mind posting a quick guide I could follow?

prairiesnpr, 2025-01-19:
@AaronLayton See https://github.com/zigpy/zigpy/issues/1508 There is a bug in zigpy, if you are trying to replace a built-in v2 quirk with a custom one, you need to delete the built-in quirk from the registry first. That said, these were just merged 4 days ago, so there shouldn't be a built-in quirk. Make sure you didn't define it twice or have multiple files with the same model and manufacturer. This will be in the HA release next month.

jabouchleih, 2025-01-19:
Hi @AaronLayton , 

set up was quite straight forward for me (side note: I'm running HA as a Docker container so this might not work with your setup but should be adaptable). 

1. Edit the Home Assistant configuration.yml found inside the root of your HA config directory by SSHing into your HA host:  
**Docker Compose YML:** 
```
volumes:
      - /srv/mydisk/homeassistant/config:/config
```
Edit the file:
```
nano /srv/mydisk/homeassistant/config/configuration.yml
```

Add the following text to your configuration.yml 

```
zha:
  custom_quirks_path: /config/custom_zha_quirks/
```

And create a directory called `custom_zha_quirks` inside the aforementioned configuration directory
```
mkdir /srv/mydisk/homeassistant/config/custom_zha_quirks/
```
2. Copy the file whole content of the Custom quirk into a file called `ts0601_trv.py`

Download the file from here: https://github.com/zigpy/zha-device-handlers/blob/dev/zhaquirks/tuya/ts0601_trv.py  
Copy all text into your clipboard (Ctrl + C)

```
nano /srv/mydisk/homeassistant/config/custom_zha_quirks/ts0601_trv.py
```

Press right click to paste the content and then Ctrl + O and Ctrl + N to save and exit the file

3. Make sure the correct access rights are set to the directory and file, if you're using specific user IDs for the Home Assistant process of container

4. Restart Home Assistant 

That should be it in all detail :-)

ahimsafantasyname, 2025-01-19:
Thank you for the hint, I will give it a try.StefanVon meinem/meiner Galaxy gesendet

  TZE200_hvaxb2tc TS0601 Valve (Issue #3527)

AaronLayton, 2025-01-19:
Hiya both, thanks for the quick replies

Looking at the link @prairiesnpr posted, I think it is this that I am getting. Hopefully it will all get resolved soon as its merged in, for now I will see if I can delete the built-in quirk from the registry.

I have added the below to the custom quirk I am trying to load up. This needs to go underneath the bottom of the other imports.
```python
from zigpy.quirks import DEVICE_REGISTRY
del DEVICE_REGISTRY._registry_v2[("_TZE200_hvaxb2tc", "TS0601")]
```

Adding the above has now make the ZHA integration load again. However on initial testing there are still no entities exposed. I will continue to look into it and report back.

-----------------

For anyone else looking at this in the meantime  
Here is the relevant entry in my `configuration.yml` file - just adding in the custom quirks folder. I already had the Ikea section for their firmware updates.  
[image]

I only have the 1 quirk installed, copied straight from the links mentioned above.
[image]

jabouchleih, 2025-01-19:
> Adding the above has now make the ZHA integration load again. However on initial testing there are still no entities exposed. I will continue to look into it and report back.

Did you readd the device? I had to do delete the device from HA, do the pairing again and only then the entities were visible.

AaronLayton, 2025-01-19:
> Did you readd the device? I had to do delete the device from HA, do the pairing again and only then the entities were visible.

I was hopeful that this may be the case for me so just tried it but still no luck. Looks like the quirk isn't applying 

[image]

closed 2025-08-03 by github-actions[bot]