zigpy/zha-device-handlers #3534: [Device Support Request] Add support for Tuya TS0601

robert4y, 2024-11-21:
### Problem description

The device connects to ZHA but there are no configuration entities. The only entity is "Firmware"

### Solution description

Compatibility with zha

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
[image]
[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_dapwryy7",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[  },
  "data": {
    "ieee": "**REDACTED**",
    "nwk": 17784,
    "manufacturer": "_TZE204_dapwryy7",
    "model": "TS0601",
    "name": "_TZE204_dapwryy7 TS0601",
    "quirk_applied": false,
    "quirk_class": "zigpy.device.Device",
    "quirk_id": null,
    "manufacturer_code": 4417,
    "power_source": "Mains",
    "lqi": null,
    "rssi": null,
    "last_seen": "2024-11-21T12:44:50",
    "available": false,
    "device_type": "Router",
    "signature": {
      "node_descriptor": {
        "logical_type": 1,
        "complex_descriptor_available": 0,
        "user_descriptor_available": 0,
        "reserved": 0,
        "aps_flags": 0,
        "frequency_band": 8,
        "mac_capability_flags": 142,
        "manufacturer_code": 4417,
        "maximum_buffer_size": 66,
        "maximum_incoming_transfer_size": 66,
        "server_mask": 10752,
        "maximum_outgoing_transfer_size": 66,
        "descriptor_capability_field": 0
      },
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "242": {
          "profile_id": "0xa1e0",
          "device_type": "0x0061",
          "input_clusters": [],
          "output_clusters": [
            "0x0021"
          ]
        }
      },]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

Product link:
https://pl.aliexpress.com/item/1005007050686755.html?spm=a2g0o.order_detail.order_detail_item.3.1dec43cepNCoAb&gatewayAdapt=glo2pol

antani04, 2024-12-15:
Apologies it's my first post and I hope I'm not breaking any rule posting a url.

Anyway I got this **_TZE204_dapwryy7** device working on ZHA with a custom quirk. 
Check this out here:
https://community.home-assistant.io/t/zha-custom-quirk-to-work-with-tuya-ts0601-tze204-dapwryy7-occupancy-sensor/794492/15

bangert, 2024-12-24:
this bug seems to be a duplicate of  #3626 (or vice versa)

closed 2025-01-20 by TheJulianJES