zigpy/zha-device-handlers #3538: [Device Support Request] _TZE284_ltwbm23f TS0601

felixwerner, 2024-11-23:
### Problem description

Add support for this TRV device: 
TS0601 _TZE284_ltwbm23f

bought here: [aliexpress](https://de.aliexpress.com/item/1005007519958883.html?spm=a2g0o.order_list.order_list_main.11.cc515c5flvFrRT&gatewayAdapt=glo2deu)

### Solution description

found this so far: https://github.com/Koenkk/zigbee2mqtt/issues/24247

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Unknown",
    "version": "2024.11.3",
    "dev": false,
    "hassio": false,
    "virtualenv": false,
    "python_version": "3.12.7",
    "docker": false,
    "arch": "aarch64",
    "timezone": "Europe/Berlin",
    "os_name": "Linux",
    "os_version": "6.1.21-v8+",
    "run_as_root": false
  },
  "custom_components": {
    "auto_backup": {
      "documentation": "https://github.com/jcwillox/hass-auto-backup",
      "version": "1.5.3",
      "requirements": []
    },
    "hacs": {
      "documentation": "https://hacs.xyz/docs/configuration/start",
      "version": "2.0.1",
      "requirements": [
        "aiogithubapi>=22.10.1"
      ]
    },
    "better_thermostat": {
      "documentation": "https://github.com/KartoffelToby/better_thermostat",
      "version": "1.6.1",
      "requirements": []
    },
    "rgbw_lamp": {
      "documentation": "",
      "version": "1.0.0",
[... 561 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

psubbotin, 2025-02-20:
please support deivce

prairiesnpr, 2025-02-20:
Try adding your device to this PR. https://github.com/zigpy/zha-device-handlers/pull/3769

closed 2025-08-28 by github-actions[bot]