zigpy/zha-device-handlers #3539: [BUG] Quirk not applied to Hue wall switch RDM004

abred, 2024-11-23:
### Bug description

Hi, I got a new Hue wall switch and tried to integrate it into HA. However the new quirk (#3075) does not get applied and the triggers are not available.

If I change the signature in https://github.com/zigpy/zha-device-handlers/blob/263e68c4877852d489688d4d23c1fe9e49795cd8/zhaquirks/philips/wall_switch.py by changing the DEVICE_TYPE
```
-DEVICE_TYPE: zha.DeviceType.NON_COLOR_CONTROLLER,
+DEVICE_TYPE: zha.DeviceType.NON_COLOR_SCENE_CONTROLLER,
```

and by adding `Basic.cluster_id,` to `OUTPUT_CLUSTERS`
```
                OUTPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    Groups.cluster_id,
                    OnOff.cluster_id,
                    LevelControl.cluster_id,
                    Ota.cluster_id,
                ],
```
(as it is done in #2764) things work as expected (quirk is listed and triggers are available)

### Steps to reproduce

1. Install Hue wall switch (RDM004)
2. Add Zigbee device in HA

### Expected behavior

1. PhilipsWallSwitch should be listed under "Quirk" under Device Info -> Zigbee Info
2. When adding an automation, a number of predefined triggers should be available (e.g. "Turn on" pressed, "Right" pressed)

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4107,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 128,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 128,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0830",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0xfc00"
      ],
      "output_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0006",
        "0x0008",
        "0x0019"
      ]
    }
  },
  "manufacturer": "Signify Netherlands B.V.",
  "model": "RDM004",
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
2024-11-23 16:26:47.671 DEBUG (MainThread) [zha] (ZHAGatewayProxy) handling event protocol for event: RawDeviceInitializedEvent(device_info=RawDeviceInitializedDeviceInfo(ieee=00:17:88:01:0e:26:3e:ef, nwk=0x189F, pairing_status=<DevicePairingStatus.INTERVIEW_COMPLETE: 2>, model='RDM004', manufacturer='Signify Netherlands B.V.', signature={'manufacturer': 'Signify Netherlands B.V.', 'model': 'RDM004', 'node_desc': {'logical_type': <LogicalType.EndDevice: 2>, 'complex_descriptor_available': 0, 'user_descriptor_available': 0, 'reserved': 0, 'aps_flags': 0, 'frequency_band': <FrequencyBand.Freq2400MHz: 8>, 'mac_capability_flags': <MACCapabilityFlags.AllocateAddress: 128>, 'manufacturer_code': 4107, 'maximum_buffer_size': 82, 'maximum_incoming_transfer_size': 128, 'server_mask': 11264, 'maximum_outgoing_transfer_size': 128, 'descriptor_capability_field': <DescriptorCapability.NONE: 0>}, 'endpoints': {1: {'profile_id': 260, 'device_type': <DeviceType.NON_COLOR_SCENE_CONTROLLER: 2096>, 'input_clusters': [0, 1, 3, 64512], 'output_clusters': [25, 0, 3, 4, 6, 8]}}}), event_type='zha_gateway_message', event='raw_device_initialized')
2024-11-23 16:26:47.673 DEBUG (MainThread) [zigpy.quirks.registry] Checking quirks for Signify Netherlands B.V. RDM004 (00:17:88:01:0e:26:3e:ef)
2024-11-23 16:38:36.729 DEBUG (MainThread) [zigpy.quirks.registry] Considering <class 'zhaquirks.philips.wall_switch.PhilipsWallSwitch'>
2024-11-23 16:38:36.733 DEBUG (MainThread) [zigpy.quirks] Fail because device_type mismatch on at least one endpoint
```

</details>

Harry-1976, 2024-11-23:
As posted in [https://community.home-assistant.io/t/how-to-configure-the-philips-hue-wall-module-to-use-push-button-momentary-type-wall-switches-zha/451125/49?u=harry-1976](url) me and others are having the same issue.

closed 2024-11-25 by TheJulianJES

RalfCGM, 2025-04-20:
I have exactly the same device signature, but the only difference is that the device is reported as rdm001 instead of rdm004.

I think im really unique in this, I think these where one of the first wall switch modules.

Problem is I cant get the to work properly with the two switches since the right setting isn't available.

Can somebody provide me with the right direction to alter the quirk for my situation?