zigpy/zha-device-handlers #3548: [Device Support Request] Ecobuy Bosch copy thermostat

cecinitmortem, 2024-11-25:
### Problem description

The device is connected, but nothing is available beside the firmware version.

### Solution description

I would like to have full support for sensors and temp settings

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_rtrmfadk",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.11.3",
[... 594 lines cut ...]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

That is the device I bought.
https://de.aliexpress.com/item/1005007023469613.html?spm=a2g0o.order_list.order_list_main.10.1ca05c5fJF4Whf&gatewayAdapt=glo2deu
It seems to be a copy by this MOES device:
https://de.aliexpress.com/item/1005006849417402.html?spm=a2g0o.productlist.main.21.27bf2943soFCUE&algo_pvid=247b7533-df4d-4722-994c-eda9c241fe0a&algo_exp_id=247b7533-df4d-4722-994c-eda9c241fe0a-10&pdp_npi=4%40dis%21EUR%2159.59%2130.39%21%21%2160.99%2131.11%21%40210384b917325653536293941e483a%2112000038508546950%21sea%21DE%214373395917%21X&curPageLogUid=LFxLfRwfvZqM&utparam-url=scene%3Asearch%7Cquery_from%3A
All of these devices seem to be a copy of the Bosch thermostat or at least an older version:
https://www.bosch-smarthome.com/de/de/produkte/geraete/heizkoerper-thermostat?srsltid=AfmBOorFXw1U5erO31kAKMfWlqgYDrfo7Ex3zjuKTAVUHSEnHGk_sysB

I guess there is a discussion over at Z2M over support for this:
https://github.com/Koenkk/zigbee2mqtt/issues/20189

pro-jul, 2024-11-26:
In addition to the above query: the named device can also have the vendor name “_TZE204_qyr2m29i”, the manufacturer is MOES.

https://de.aliexpress.com/item/1005006194184981.html?spm=a2g0o.productlist.main.1.444bSlj0Slj02c&algo_pvid=a87a6439-55d5-41e1-b4da-408362f584d9&algo_exp_id=a87a6439-55d5-41e1-b4da-408362f584d9-0&pdp_npi=4%40dis%21EUR%2159.78%2126.99%21%21%2161.04%2127.56%21%40211b80c217326307031986758e897b%2112000036215544418%21sea%21DE%210%21ABX&curPageLogUid=e3E4PXd8LyDe&utparam-url=scene%3Asearch%7Cquery_from%3A

tco99ttocs, 2024-12-03:
i got a similar product with the name "_TZE204_ltwbm23f" its also from Ecobuy
[https://de.aliexpress.com/item/1005007023469613.html?spm=a2g0o.order_list.order_list_main.15.52475c5fspQMHs&gatewayAdapt=glo2deu](https://de.aliexpress.com/item/1005007023469613.html?spm=a2g0o.order_list.order_list_main.15.52475c5fspQMHs&gatewayAdapt=glo2deu)

cecinitmortem, 2025-01-07:
I found out that after editing this quirk [https://github.com/jacekk015/zha_quirks/blob/main/ts0601_trv_rtitek2.py](url) just by copying the last signature and adapting to this device I could no control the device and set temperature. It does not include everything I need and I do not know currently how reliable it is, but it is a startpoint.

pro-jul, 2025-01-11:
hello everyone, since the new zigbee2mqtt update the device (at least the one from MOES) is directly supported from zigbee2mqtt! It works perfectly! just try it!

closed 2025-01-20 by TheJulianJES

cecinitmortem, 2025-02-11:
I have two "weird" endpoints with this device:

[image]
[image]
After setting up the device freshly I wanted to know what they do. After configuring the second one, the thermostat had a minimum temperature of 3°C. I could not reconfigure it to the value it had before. Even on the device itself I could not set the temperature higher than 3°C. Removing and setting up again did not fix that. The only thing that helped was resetting the device to factory defaults helped.
There are two other endpoints to configure the same settings. I have not used them. 

[image]

I have updated to HA 2025.2.2 .