zigpy/zha-device-handlers #3549: [Device Support Request] TS0505B TZ3210_bfwvfyx1 smart bulb model C3009 RGB

c4baceira, 2024-11-25:
### Problem description

TS0505B by _TZ3210_bfwvfyx1 is recognized and some functions work:

- on/off
- color temperature (white)
- brightness
- start-up 

Missing:
- RGB contorl

### Solution description

Possible custom quirk based on this https://github.com/zigpy/zha-device-handlers/issues/1429 but user deleted his github repo.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]
[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010d",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xe000",
        "0xe002",
        "0xee00",
        "0xef00"
      ],
      "output_clusters": [
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x000a",
        "0x0300",
        "0x1000"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_bfwvfyx1",
  "model": "TS0505B",
  "class": "zigpy.device.Device"
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

wenzelt, 2024-12-08:
Same issue where I can use the colour cycle in z2m but little else

pippus88, 2025-09-18:
Please reopen a issue