zigpy/zha-device-handlers #3551: [Device Support Request] TS0601 _TZE284_eaet5qt5

niiconn, 2024-11-26:
### Problem description

Tuya Smart Water Timer

Device can be added via ZHA to HA, but there are no sensors and switches. Purchased from [here](https://de.aliexpress.com/item/1005007549776847.html?spm=a2g0o.order_list.order_list_main.35.cb265c5fomLdiC&gatewayAdapt=glo2deu).

### Solution description

 
[image]
[image]

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_eaet5qt5",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

prairiesnpr, 2024-11-26:
That looks identical to the GIEX GX04, you might be able to modify the existing [quirk](https://github.com/zigpy/zha-device-handlers/blob/213ce1012210c443ef64c5088dedf9ae55fd1886/zhaquirks/tuya/ts0601_valve.py#L488), try adding your model, otherwise we would need to have the data points it's using to confirm.

niiconn, 2024-11-26:
Thanks @prairiesnpr for that extremely fast answer. Much appreciated. I will modify the mentioned Quirk and test it tonight.

closed 2024-11-28 by TheJulianJES