zigpy/zha-device-handlers #3553: Expose heating mode for NodOn SIN-4-FP-21 and Legrand cable outlet

TheJulianJES, 2024-11-26:
We should expose the heating/wire mode for the NodOn SIN-4-FP-21 and Legrand cable outlet to Home Assistant.
Related PRs:
- #3364
- #3031

Concerns were raised that we would not conform to Home Assistant standards by exposing the heating/wire mode of the devices as a `select` entity. Rather, we should consider having a climate entity. The heating mode would replace the automatically discovered on/off `switch` entity and the enums would be shown as heating presets.

Zigbee2MQTT just exposes an enum, similar to what's proposed in #3364.

The Legrand device is marketed as a smart power outlet with power consumption monitoring. The NodOn device is a shelly-like module with a full focus on being connected to an electric heater. Both devices seem to be similar software-wise.
For the Legrand device, it would be weird removing the existing switch entity for users only utilizing the smart on/off functionality for a non-heating device.

There's been no decision on what we can/want to do yet

vinc, 2024-11-27:
I've been following the PR for the NodOn SIN-4-FP-21 that I installed on some heaters in my house. I successfully validated the current switch to turn my heaters on or off but a simple solution to switch between the different modes (comfort, eco, off, etc) would be really useful for my usage!

TheJulianJES, 2025-01-27:
@ikruglov Not promising anything yet, but can you just PR the same change with the enum/select entity again for this device?

ikruglov, 2025-01-28:
> [@ikruglov](https://github.com/ikruglov) Not promising anything yet, but can you just PR the same change with the enum/select entity again for this device?

@TheJulianJES I see that https://github.com/zigpy/zha-device-handlers/pull/3798 has already implemented it

TheJulianJES, 2025-01-28:
Yep!

closed 2025-08-04 by github-actions[bot]