zigpy/zha-device-handlers #3557: [Device Support Request] TS0601_TZE200_drs6j6m5

DanielSobczak, 2024-11-27:
### Problem description

I recently have bought a led driver Lifud LF-GAZ150A6250-24.
Device can be added via ZHA to HA, but there are no switches available. 
Purchased from [here](https://msled.pl/pl/p/Zasilacz-LED-ZigBee-40W-750-1050mA-20-42VDC-stalopradowy-/1274).

### Solution description

I would like to control the brightness and be able to turn off/on light.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE200_drs6j6m5",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
"""Tuya dimmable led controller."""
from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomCluster, CustomDevice
from zigpy.zcl.clusters.general import (
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    LevelControl,
    OnOff,
    Ota,
    Scenes,
    Time,
)

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import NoManufacturerCluster, TuyaDimmerSwitch, TuyaSwitch
from zhaquirks.tuya.mcu import (
    TuyaInWallLevelControl,
    TuyaLevelControlManufCluster,
    TuyaOnOff,
    TuyaOnOffNM,
    MoesSwitchManufCluster
)

class DimmableLedController(TuyaSwitch):
    """Tuya dimmable led controller."""

    signature = {
        MODELS_INFO: [("_TZE200_drs6j6m5", "TS0601")],
        ENDPOINTS: {
            1: {
                PROFILE_ID: 260,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    4, 5, 61184, 0
                ],
                OUTPUT_CLUSTERS: [25, 10],
            },
            242: {
                PROFILE_ID: 41440,
                DEVICE_TYPE: 97,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [33],
            },
        },
    }
    replacement = {
        ENDPOINTS: {
             1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    Basic.cluster_id,
                    MoesSwitchManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [25, 10],
            }
        }
    }
```

</details>

### Additional information

With the above quirk I'm able to control the on/off state, to try brightness control I've tried to use `TuyaLevelControl`, `TuyaLevelControlManufCluster` and `TuyaInWallLevelControl` but they didn't work for me. 

As I have seen, this device is supported by [Zigbee2MQTT](https://github.com/Koenkk/zigbee-herdsman-converters/blob/a7517c4186be8f68eeef76db1499bd8d23f6c662/src/devices/tuya.ts#L10430) where they set

> [3, 'brightness', tuya.valueConverter.scale0_254to0_1000], 

as tuya datapoint 
After trying to figure out what should be send I noticed that the values within payload might need to be changed from `transition_time` to `transtime` - I was trying to customize `F000LevelControlCluster` but due to not fully understanding what exactly needs to be send I was not able to control the brightness.

jtosic, 2025-01-13:
I have LTech LM-150-24-G1Z2 ("_TZE200_qyss8gjy") which, according to Zigbee2MQTT, should be pretty much the same as your device but if I modify "MODELS_INFO" variable to match my manufacturer I cannot get your quirk to load. If I make a copy of "ts0601_dimmer.py" and add my model there I can control on/off but not the levels.
I'm a complete novice when it comes to these quirks so I would be very grateful if you could give me a few pointers.

closed 2025-08-02 by github-actions[bot]

muratozbayraktar, 2025-08-22:
Hi, i have ("_TZE200_p0gzbqct", "TS0601") and as all said i can only control on/off. cannot use the dimmer. But I can see on the screen.

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE200_p0gzbqct",
  "model": "TS0601",
  "class": "ts0601_dimmer.TuyaSingleSwitchDimmer"
}

[image]

[image]