zigpy/zha-device-handlers #3559: [Device Support Request] TS0601 _TZE284_o3x45p96

zantanex, 2024-11-27:
### Problem description

[AVATTO Tuya TRV Smart Radiator Valve](https://www.aliexpress.com/item/1005003952150617.html)

Device is detected and added easily, however the entities/controls are missing. Although the model looks identical to most supported TRV valves, I haven't been able to find any quirks that work for this particular manufacturer code.

### Solution description

Would like to have support for the entities/controls.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_o3x45p96",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
[... 626 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

smithjames0, 2024-11-29:
Hi, 
I have the same issue with this TRV, I have tried to modify other quirks but they all fail to work.

Regards

LordMGR, 2024-12-02:
It has one more input cluster than all the other TS0601.

smithjames0, 2024-12-03:
> It has one more input cluster than all the other TS0601.

I realised that as well, I am not sure what the extra cluster is to add it to an existing quirk.

LordMGR, 2024-12-19:
I switched to zigbee2mqtt and saw the clusters used there: 
Input clusters:
genGroups
genScenes
manuSpecificTuya
genBasic
zosungIRTransmit

Maybe this info helps somebody to do the next step.

jamiesonp, 2024-12-25:
I have added an identical request for "manufacturer": "_TZE204_ltwbm23f"
https://github.com/zigpy/zha-device-handlers/issues/3639

rootjumper, 2025-01-10:
> I have added an identical request for "manufacturer": "_TZE204_ltwbm23f" #3639

Tested https://github.com/zigpy/zha-device-handlers/pull/3684 for "_TZE284_o3x45p96" and basic functionallity is working.
See my comment in https://github.com/zigpy/zha-device-handlers/issues/3516#issuecomment-2573113310

closed 2025-01-16 by TheJulianJES

falkgeist, 2025-03-31:
I'm not sure if I'm the only one, please let me know if yes, but I am missing the local_temperature value readout with zha for my thermostat. It's it possible this was overlooked? Also, automatic mode is not detected/shown in HA...

falkgeist, 2025-04-02:
Okay, been skimming (and occasionally deep-diving) now through several issues (this one, #1818, #3516, [jacekk015/zha-quirks#65](https://github.com/jacekk015/zha_quirks/issues/65)), repos ([jacekk015/zha-quirks](/jacekk015/zha-quirks)) and PRs (#3684) and have a better understanding of the state of things.

Although my thermostat is mostly working, without even adding any custom quirk (only slowly learning what that is) myself (I guess it was automatically found and set for my device, nice), I am not fully satisfied yet.

There seems to be no issue left open except for the one mentioned above (jacekk015/zha-quirks#65), but frankly, the thermostat is not functioning as well as it should be (I am also guessing, since it's a cheap Chinese product, it might never technically work as well as others, but that's what you pay for, I guess).

__So, I propose to reopen this issue and/or the PR #3684__

In particular, I feel like at least a few things are missing until I (and hopefully others) would consider this issue resolved:

1. a working thermostat entity with:
    - [x] desired room temperature
    - [ ] measured room temperature -> at least for me I only got that after manually reading the local_temperature value in the TuyaThermostatV2 cluster of the "Manage Zigbee Device" window (still not sure how reliable the readout of the thermostat sensor is, waiting to test it with some other thermometers, but at least the HA value matches the LCD readout)
    - modes:
      - [x] on (= automatically turns on if desired temp is below local temp, off if desired temp is at or above local temp)
      - [x] off (= no heating at all)
      - [ ] auto (= respect the devices weekly schedule)
1. a schedule entity (configuration) with
    - [ ] a) each weekday as a text field to add the schedule string (e.g. "06:00/18.0 12:00/20.0 18:00/18.0")
    - [ ] b) better: an extendable multi-field control where the values for time and temperature can be set like some sort of key-value pair (hope you get what I am aiming at...), which translates the user input to the string expected by the device
    - [ ] c) even better: a schedule control e.g. like in the scheduler card entity, which translates the user input to the string expected by the device
1. - [ ] possibly: the correct firmware status (even though I guess it will never get updated)

As you can see, at least (imho) the local_temperature should be added immediately. And for convenience, I'd also like to use the on-device schedule, but user-friendly controlled from a dashboard entity. (And yes, I added some already implemented functionalities, but those - bar the schedules - are the bare minimum I would expect.)

I'd implement it myself, but I feel there are many people already versed with quirks and HA/ZHA and even this specific thermostat. So maybe someone is still up for adding the mentioned functionality? I propose to reopen #3684 and add at at least proper support for the local_temperature and the firmware, and optionally for the schedules.

By the way, and iirc its not only my experience: the Z2M implementation is a bit superior in its completeness, but at least in my case it lacked in one critical point: connection quality. It took quite some time until my devices updated and even so, some of them never did (or at least not in the time I checked them), since the connection failed. I thought Zigbee was supposed to cover a large area or was I misinformed? Or was it because of mqtt? No matter, I like ZHA, and even though the difference is small for mostly unsupported devices, I think I'll stick with it and either try to figure out in time how to improve the functionality or wait until someone did.

Sorry for the wall of text, I just needed to find someplace to sidle my way into the discussion and start working towards a solution for myself (and possibly others).

TheJulianJES, 2025-04-02:
Please open a new issue, so we can track that properly. General support is already implemented.

The `local_temperature` attribute/datapoint not reporting might be due to Tuya detecting it's not their hub. Try adding `.tuya_enchantment()` in the `_TZE284_o3x45p96` quirk.
Just download the entire file and "import" it as a custom quirk, then add the `.tuya_enchantment()` line there to test.
Afterwards, reconfigure or re-pair the device and see if it then behaves differently after you did that once.

Unfortunately, there's no visible way in the UI to tell if you're using an in-built v2 quirk or a custom v2 quirk, yet.

falkgeist, 2025-04-02:
I'm sorry, I moved it and I'll try your solution, but it does show up as custom, if I see it correctly:
[image]

TheJulianJES, 2025-04-02:
A built-in v2 quirk will also show up like that, as that's the class name in zigpy for all quirked devices.

We'll add the file path of v2 quirks to the UI in the future.