zigpy/zha-device-handlers #3560: [Device Support Request] TS0505B w/ _TZ3210_rcggc0ys

VCTGomes, 2024-11-28:
### Problem description

Current Tuya bulbs in ZHA do not offer the Do Not Disturb, the Tuya power-on behavior version.

### Solution description

Please add Do Not Disturb feature which seems to process Tuya custom cluster 0xef00 with DP 21.

Z2M support is here:
https://github.com/Koenkk/zigbee2mqtt/issues/19718

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010d",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_rcggc0ys",
  "model": "TS0505B",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

VCTGomes, 2024-12-02:
It looks to use the Color Cluster:
https://github.com/Koenkk/zigbee-herdsman/blob/0408bcc7196a520f6314c9bb5e18e377d54a8667/src/zcl/definition/cluster.ts#L2198-L2458

Many Tuya Lamps has this same cluster. I hope it'd be implemented.

VCTGomes, 2024-12-05:
New updates: It indeed uses Color Cluster:
I must send a command to cluster 0x0300 (768) with attribute is 0xFA (250). Data is 0-disable DND, 1 - enable DND.
So, it's not something hard to implement.

andrei-lazarov, 2025-01-27:
> New updates: It indeed uses Color Cluster: I must send a command to cluster 0x0300 (768) with attribute is 0xFA (250). Data is 0-disable DND, 1 - enable DND. So, it's not something hard to implement.

Hey, I'm new to ZigBee devices, can you detail this a bit? It would help me a lot

VCTGomes, 2025-01-28:
> > New updates: It indeed uses Color Cluster: I must send a command to cluster 0x0300 (768) with attribute is 0xFA (250). Data is 0-disable DND, 1 - enable DND. So, it's not something hard to implement.
> 
> [@VCTGomes](https://github.com/VCTGomes) Hey, I'm new to ZigBee devices, can you detail this a bit? It would help me a lot

Unfortunately I also don't know how to do it... 😕

andrei-lazarov, 2025-01-28:
Oh sorry, I thought you found a workaround to manually set the value. I tried this but it does not find the attribute:

```
action: zha.set_zigbee_cluster_attribute
data:
  value: 1
  ieee: <address>
  cluster_type: in
  attribute: 250
  cluster_id: 768
  endpoint_id: 1
```

rodolfomachado, 2025-02-22:
> Oh sorry, I thought you found a workaround to manually set the value. I tried this but it does not find the attribute:
> 
> ```
> action: zha.set_zigbee_cluster_attribute
> data:
>   value: 1
>   ieee: <address>
>   cluster_type: in
>   attribute: 250
>   cluster_id: 768
>   endpoint_id: 1
> ```

> New updates: It indeed uses Color Cluster: I must send a command to cluster 0x0300 (768) with attribute is 0xFA (250). Data is 0-disable DND, 1 - enable DND. So, it's not something hard to implement.

Hi, I have the device "TS0505B by _TZ3210_ijczzg9h" and I think it is almost the same. 
but It does not have the attribute 0xFA in cluster 0x0300. What is the name of the attribute?

andrei-lazarov, 2025-02-23:
@rodolfomachado we don't have the DND attribute either because it's not supported by ZHA. Currently only Zigbee2MQTT supports it

rodolfomachado, 2025-02-23:
> [@rodolfomachado](https://github.com/rodolfomachado) we don't have the DND attribute either because it's not supported by ZHA. Currently only Zigbee2MQTT supports it

 Ive read again what you said and I guess I misread last time.

I thought you changed DND manually. But I was wrong.