zigpy/zha-device-handlers #3562: [BUG] TS0505B Light turning on issues and incorrect mireds

flybrys, 2024-11-29:
### Bug description

Purchased once of these ceiling lights and it paired with ZHA straight away, however I have the following issues:

* I can only turn it on via the switch in the UI. Automations or scenes that get triggered do not turn it on
* I get an error in the logs about min mireds is 0:
>[0x96E6:1:0x0300]: [Min mireds is 0, setting to 153] Please open an issue on the quirks repo to have this device corrected
* Start-up temperature and identify options don't seem to do anything (don't really care tho
* I've also noticed, despite being a mains powered device, that there is no connection to anything on the visualisation map. Still able to turn it on and off via the UI.

### Steps to reproduce

1. Purchase the zigbee version of this: https://www.aliexpress.com/item/1005005754093751.html?spm=a2g0o.order_list.order_list_main.17.b50518021ZinDX
2. Install it and turn it on
3. Pair it to ZHA
4. Notice log about mireds and wonder why your scenes aren't activating when it works in the UI

### Expected behavior

Expected to turn on with scenes or automations

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json

```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x010d",
      "input_clusters": [
        "0x0000",
        "0x0003",
        "0x0004",
        "0x0005",
        "0x0006",
        "0x0008",
        "0x0300",
        "0x1000",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZ3210_bfwvfyx1",
  "model": "TS0505B",
  "class": "zigpy.device.Device"
}```

</details>

### Additional information

Had trouble pasting the logs. If required will try and host the file

flybrys, 2024-11-29:
Have just done some more testing and found its only failing to turn on when setting a scene or specific colour from an off state.
If I open the light entity and click one of the colours once, nothing happens despite HA saying its on. If I click it a second time, it turns on and applies the colour.

However no matter how many times I try and apply a scene from an off state, nothing happens

flybrys, 2024-11-29:
Also just tried something in the developer tools.
I can turn the light on and off using the light.turn_on action, however if I specify any attributes such as color, brightness etc, it fails.

flybrys, 2024-11-29:
Another issue is that the temperature range on the device is 1000k to 6535k
It should be 2700k to 6000k

flybrys, 2024-11-29:
Sorry for the flood of messages whilst I keep testing. Will delete previous messages and consolidate into my findings below:

1. Light does turn on and off fine if there are no attributes specified, so light does not turn on with a scene despite Home Assistant showing the light as on.
2. Scene does seem to apply in the background without turning the light on. So if I call a basic light.turn_on after applying the scene it does work correctly however with a short delay.
3. Another issue is the temperature range indicates 1000k to 6535k where it should be 2700k to 6000k (related to mireds error?)
4. The same behaviour happens with Zigbee2MQTT.
5. No firmware updates exist in the Tuya Smart Life app. I didn't test functionality via Tuya in HA because I don't want to sign up for an account

I guess I can work around the scene issue by turning the light on after applying the scene, but its clunky and would prefer it to work as intended. However apart from the mireds error in the log I'm starting to think this could be a firmware issue.

foxtom22, 2025-03-07:
Similar issue with this product https://fr.aliexpress.com/item/1005006807040028.html
In zha it shows as TS0505B by _TZ3210_3lbtuxgp