zigpy/zha-device-handlers #3567: [Device Support Request]

toomaskordemets, 2024-11-29:
### Problem description

There is AVATTO Tuya dimmer device [in aliexpress](url) that identifies as "_TZE204_nqqylykc" in TS0601 device family. It pairs flawlessly but does not expose any meaningful controls other than LQI and RSSI. Because of its price and dimensions it seems growing popularity.

### Solution description

Would be nice to have it's support added to ZHA. I assume many will follow me with similar wishes.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    },
    "242": {
      "profile_id": "0xa1e0",
      "device_type": "0x0061",
      "input_clusters": [],
      "output_clusters": [
        "0x0021"
      ]
    }
  },
  "manufacturer": "_TZE204_nqqylykc",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information
[... 641 lines cut ...]
  }
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

sanderpuh, 2024-12-01:
Seconded, installed mine yesterday and looking for a solution!

gawsneakerz, 2024-12-20:
I have the same! Did you guys find a custom quirk for it already?

sanderpuh, 2024-12-20:
Yes, I have! I've copied the following quirk from this github: https://github.com/zigpy/zha-device-handlers/blob/979551d67135cf4ad68c4dbc447df76d5426b78c/zhaquirks/tuya/ts0601_dimmer.py
Added the `ts0601_dimmer.py` file in my `custom_zha_quirks` folder. 

Then, under line 53, in `class TuyaSingleSwitchDimmer(TuyaDimmerSwitch):`

I've added the following device: `("_TZE204_nqqylykc", "TS0601"),  # Avattto ZDMS16-1`

It did show up as a double switch dimmer, but I simply made one hidden, and it is working for me for now.
I'm not sure if this is 100% the proper quirk to implement the dimmer, which is why I haven't opened a PR.
Feel free to give it a try!

gawsneakerz, 2024-12-20:
Yeah, I managed to fix it myself after this comment!

gawsneakerz, 2024-12-20:
Also made a pull request.
https://github.com/zigpy/zha-device-handlers/pull/3624

closed 2024-12-22 by TheJulianJES