zigpy/zha-device-handlers #3569: TS0601 _TZE200_laqjm8qd

Buckwallen, 2024-12-01:
### Problem description

Brand name 'Melery' 5 gang ZigBee light switch device that identifies as "_TZE200_laqjm8qd" in TS0601 device family. It pairs flawlessly but does not expose any meaningful controls other than LQI and RSSI.

### Solution description

Added ZHA support would be amazing as these switches grow in popularity.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

      "node_descriptor": {
        "logical_type": 1,
        "complex_descriptor_available": 0,
        "user_descriptor_available": 0,
        "reserved": 0,
        "aps_flags": 0,
        "frequency_band": 8,
        "mac_capability_flags": 142,
        "manufacturer_code": 4417,
        "maximum_buffer_size": 66,
        "maximum_incoming_transfer_size": 66,
        "server_mask": 10752,
        "maximum_outgoing_transfer_size": 66,
        "descriptor_capability_field": 0
      },
      "endpoints": {
        "1": {
          "profile_id": "0x0104",
          "device_type": "0x0051",
          "input_clusters": [
            "0x0000",
            "0x0004",
            "0x0005",
            "0xef00"
          ],
          "output_clusters": [
            "0x000a",
            "0x0019"
          ]
        },
        "242": {
          "profile_id": "0xa1e0",
          "device_type": "0x0061",
          "input_clusters": [],
          "output_clusters": [
            "0x0021"
          ]
        }
      },
      "manufacturer": "_TZE200_laqjm8qd",
      "model": "TS0601"
    },
    "active_coordinator": false,
    "entities": [
      {
        "entity_id": "update.house_main_light_switches_firmware",
        "name": "_TZE200_laqjm8qd TS0601"
      }
    ],
    "neighbors": [],
    "routes": [],
    "endpoint_names": [
[... 949 lines cut ...]
}

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

Device also pairs with ZigBee hub through Tuya flawlessly and also shows all switches without any issues.  This problem only lies in HA. The purpose of this quirk is to try and get myself completely away from Tuya.

prairiesnpr, 2024-12-01:
Can you grab the Tuya data points, either from the Tuya dev console or local Tuya?

Buckwallen, 2024-12-01:
Hi Prairiesnpr,

Thank you so much for the speedy reply! I will get the Tuya data points info tomorrow as its late here and I'm about to jump into bed. 
I have copied your quirk and placed it in the config/custom_zha_quirks/ folder as a .py file (no other info in the file except quirk). Upon restarting home assistant there was no change to TS0601 _TZE200_laqjm8qd. I also deleted the device and re-added through ZHA. I am not sure if I am missing some thing though as I am new to this :)

I have attached a screenshot of what the device looks like in ZHA.

I appreciate your help.
Regards
Pete

[image]

Buckwallen, 2024-12-01:
> Hi Prairiesnpr,
>
> Thank you so much for the speedy reply! I will get the Tuya data points info tomorrow as its late here and I'm about to jump into bed. I have copied your quirk and placed it in the [...]

Here are the data points from Tuya Developer Consol

Buckwallen, 2024-12-01:
> Here are the data points from Tuya Developer Consol

{"1":"Switch 1","2":"Switch 2","3":"Switch 3","4":"Switch 4","5":"Switch 5","14":"Restart Status","15":"Indicator Status"}

prairiesnpr, 2024-12-01:
From your image, the quirk either isn't loading or isn't matching. DPs look good, check the logs for errors. You have used custom quirks in the past?

Buckwallen, 2024-12-01:
> From your image, the quirk either isn't loading or isn't matching. DPs look good, check the logs for errors. You have used custom quirks in the past?

Hey. I have never used custom quirks before. Funny thing I have noticed though is all my other zigbee devices (which never had any issues are showing as using quirks. Screenshot below:

[image]

Is there any other info needed in the .py file? I have only entered yours from above. No other info.

prairiesnpr, 2024-12-01:
That file should work as is, just ensure you have configured zha to load custom quirks. https://www.home-assistant.io/integrations/zha/

Buckwallen, 2024-12-04:
Hi Prairiesnpr,

Unfortunately I was not able to get my TS0601 _TZE200_laqjm8qd working with the code provided above, however I was able to use the below code to get all five switches showing in HA:

 """Tuya DP based switches."""

from zigpy.profiles import zgp, zha
from zigpy.zcl.clusters.general import Basic, GreenPowerProxy, Groups, Ota, Scenes, Time

from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.quirk_ids import TUYA_PLUG_MANUFACTURER
from zhaquirks.tuya import TuyaSwitch
from zhaquirks.tuya.mcu import (
    MoesSwitchManufCluster,
    TuyaOnOff,
    TuyaOnOffManufCluster,
    TuyaOnOffNM,
)

class TuyaQuintupleSwitchTO(TuyaSwitch):
    """Tuya quintuple channel switch time on out cluster device."""

    quirk_id = TUYA_PLUG_MANUFACTURER

    signature = {
        MODELS_INFO: [
            ("_TZE200_laqjm8qd", "TS0601"),
        ],
        ENDPOINTS: {
            # <SimpleDescriptor endpoint=1 profile=260 device_type=81
            # input_clusters=[0x0000,0x0004,0x0005,0x0006,0xEF00]
            # output_clusters=[0x000A,0x0019]>
            1: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.SMART_PLUG,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster.cluster_id,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            # <SimpleDescriptor endpoint=242 profile=41440 device_type=97
            # input_clusters=[]
            # output_clusters=[33]
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        },
    }

    replacement = {
        ENDPOINTS: {
            1: {
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    Basic.cluster_id,
                    Groups.cluster_id,
                    Scenes.cluster_id,
                    TuyaOnOffManufCluster,
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [Time.cluster_id, Ota.cluster_id],
            },
            2: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            3: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.ON_OFF_LIGHT,
                INPUT_CLUSTERS: [
                    TuyaOnOffNM,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }

The problem I have now is that the switches when turned on/off in ZHA dont react well. When turning a switch on there is no reaction until I either press it again (which results in turning the switch back off again) and vice versa. Or, if i turn a switch on it wont turn on until I press another switch on. The reaction time off my switches are all over the place. Sometimes it works ok but very inconsistent.
Do you have any tips that could point me in the right direction?

Regards
Pete

Buckwallen, 2024-12-04:
> Hi Prairiesnpr,
>
> Unfortunately I was not able to get my TS0601 _TZE200_laqjm8qd working with the code provided above, however I was able to use the below code to get all five switches showing in [...]

This is how the device is now showing in ZHA:

[image]

prairiesnpr, 2024-12-04:
Do you have logs from the quirk I posted?

Buckwallen, 2024-12-04:
Here is the log from startup:

[image]

[image]

Buckwallen, 2024-12-04:
> Here is the log from startup: [...]

Here is line 7 if that helps:

[image]

prairiesnpr, 2024-12-04:
Can you grab debug logs when trying to switch it? I'm curious if you are getting delivery errors. We could have a network issue, not necessarily a quirk issue.

Buckwallen, 2024-12-04:
Hey. I hope this is what you are looking for:

[zha-ebd54d31d0fd721638dd66ddfac2e08f-_TZE200_laqjm8qd TS0601-db1e910835fab6bc9e923d370e01cba8.json](https://github.com/user-attachments/files/18009665/zha-ebd54d31d0fd721638dd66ddfac2e08f-_TZE200_laqjm8qd.TS0601-db1e910835fab6bc9e923d370e01cba8.json)

Buckwallen, 2024-12-08:
Hi Prairiesnpr. I just wanted to say thank you for your help with getting my 5 gang switch entities into ZHA. I am going to close this issue now that my initial problem is solved. I will open a ne request regarding my switch timing issue. 

I really appreciate your help!

Cheers
Pete

closed 2024-12-08 by Buckwallen