zigpy/zha-device-handlers #3570: [Device Support Request] _TZE200_iba1ckek TS0601 (ZG-103Z) Tuya Vibration Tilt Door Window Sensor

weboide, 2024-12-01:
### Problem description

The default ZHA support for _TZE200_iba1ckek TS0601 (ZG-103Z) is missing the tilt, x, y, z, sensitivity values. Only the vibration value is exposed to Home Assistant.

A definition of the sensor is available at https://www.zigbee2mqtt.io/devices/ZG-103Z.html and purchased from https://www.aliexpress.us/item/3256807349737583.html

### Solution description

Add quirk supporting missing tilt, x, y, z, sensitivity values.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4742,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0402",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0500"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZE200_iba1ckek",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.11.3",
    "dev": false,
[... 727 lines cut ...]
}
```

</details>

### Logs

<details><summary>Logs</summary>

```python
2024-12-01 15:36:25.925 DEBUG (MainThread) [zha] (ZHAGatewayProxy) handling event protocol for event: DeviceFullInitEvent(device_info=ExtendedDeviceInfoWithPairingStatus(ieee=removed, nwk=0x9EA5, manufacturer='_TZE200_iba1ckek', model='TS0601', name='_TZE200_iba1ckek TS0601', quirk_applied=False, quirk_class='zigpy.device.Device', quirk_id=None, manufacturer_code=4742, power_source='Battery or Unknown', lqi=138, rssi=None, last_seen='2024-12-01T15:36:23', available=True, device_type='EndDevice', signature={'node_descriptor': NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=4742, maximum_buffer_size=66, maximum_incoming_transfer_size=66, server_mask=10752, maximum_outgoing_transfer_size=66, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False), 'endpoints': {1: {'profile_id': '0x0104', 'device_type': '0x0402', 'input_clusters': ['0x0000', '0x0001', '0x0003', '0x0500'], 'output_clusters': []}}, 'manufacturer': '_TZE200_iba1ckek', 'model': 'TS0601'}, active_coordinator=False, entities={}, neighbors=[], routes=[], endpoint_names=[EndpointNameInfo(name='IAS_ZONE')], pairing_status=<DevicePairingStatus.CONFIGURED: 3>), new_join=True, event_type='zha_gateway_message', event='device_fully_initialized')

```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

IntegersOfK, 2025-01-07:
I have this sensor and it is working properly in Home Assistant with ZHA but I wish I had some way to increase the sensitivity.

BlikeBread, 2025-02-11:
I follow.

infinitytec, 2025-02-22:
I just got one of these as well. It does report, but doesn't have XYZ tilt values.

andion, 2025-03-18:
No XYZ values here too. How do you managed to get them? @IntegersOfK

IntegersOfK, 2025-03-19:
I don't have tilt values either, I guess I should have specified that. So when I said it's "working properly" I mean it tells me when vibration is detected generally.

tangobones, 2025-04-09:
My device only shows vibration but the entity is unavailable

Haerteleric, 2025-04-17:
i added this device via deCONZ, and it advertises another Cluster (0xEF00) that is not shown in ZHA. I think configuration of sensitivity is done via that cluster

[image]

Haerteleric, 2025-04-17:
First Byte is always 0, second byte is a sequence number, third byte looks like a param identifier, last byte is value

identifyer 0x65, 0x66 and 0x67 seem to report tilt

Haerteleric, 2025-04-17:
if i flip the sensor (from face to back) value 67 has the biggest change, if i roll the sensor on its case 66 is affected the most

Haerteleric, 2025-04-17:
its actully an accelerometer.

[image]

~64 seems to be 1G (or 9.81 m/(s*s)).

jezzaaa, 2025-04-17:
Yes, it's an accelerometer. I've been trying to get various AIs to write me a quirk for the past week, but I've not been able to finish it off. I have a quirk that logs X, Y and Z values, but I can't get them into an attribute or entity.

The last byte is X/Y/Z accel level, ranging between around -70 to +70 under 1G, increasing up to -128 to +127 if you swing it around violently. I've noticed that if you push the Gs to +/- 127 for more than a couple of seconds, the X/Y/Z numbers stop for a while, maybe 15 seconds.

The third byte, 0x65, 0x66 and 0x67 seem to be the DP IDs 101, 102 and 103. These represent X, Y and Z. ID 105 appears to be battery level as a percentage (your battery appears to be at 0x64=100%). ID 1 seems to be the vibration state (1 or 0), but there's ID 7 that seems to give a similar result most of the time, but is less consistent.

jezzaaa, 2025-04-17:
The tilt angle can be calculated from X/Y/Z accelerometer values. However, they can't be calculated on the fly when a message comes in, because no single message has all three values. Instead, the quirk would need to wait for all three messages and then perform the calculation. And even this wouldn't work, because in practice only the axes that change are reported, so if you rotate rotate around the X axis, only messages for Y and Z will be sent.

I don't know if a quirk can keep state between messages. If so, then it would be possible to calculate tilt angles from the most recent message for each axis. Of not, then some sort of helper within HA would be required (perhaps an automation would be necessary).

Or perhaps when any X/Y/Z message comes in, the quirk could ask the device for all of the values and do the calculation. But I suspect that the device doesn't accept queries for the current values.

Haerteleric, 2025-04-18:
I have not fully wrapped my head around how the quircks work yet. 

But if read the Philips remote cluster code and the use a timeout to simulate multi button presses. This could be repurposed for that

jezzaaa, 2025-04-19:
In case it helps someone else, below is my quirk. All it's able to do is extract the accel values from the Zigbee messages and send them to the log.

The cmd_id value gets picked up as a two-byte value, but I believe it should be a single byte (and I don't know how to fix this because I'm not sufficiently skilled in Python). Consequently, I'm comparing this value against 0x0265 (for X axis) instead of 0x65 (decimal 101).

I don't know how to get the extracted values into Home Assistant.

Digging around the Zigbee device handler source code, I can see that there are some magic Tuya modules that would probably make this all a whole lot simpler, but I'm not able to understand how to use them. In particular there's a TuyaNewManufCluster class that is an attempt to consolidate other modules and classes into a single framework.

```
import logging
from zigpy.profiles import zha
from zigpy.quirks import CustomDevice
from zigpy.zcl.clusters.general import Basic, PowerConfiguration, Identify
from zigpy.zcl.clusters.security import IasZone
from zhaquirks.tuya import TuyaManufCluster
import zigpy.types as t

_LOGGER = logging.getLogger(__name__)

class TuyaAxisCluster(TuyaManufCluster):
    """Custom cluster to decode accelerometer axis data from TS0601."""

    cluster_id = 0xEF00

    attributes = {
        0x0101: ("accel_x", t.int32s, True),
        0x0102: ("accel_y", t.int32s, True),
        0x0103: ("accel_z", t.int32s, True),
    }

    def handle_cluster_request(self, hdr, args, dst_addressing=None):
        """Handle incoming Tuya cluster requests."""
        _LOGGER.debug("handle_cluster_request: hdr=%s args=%s", hdr, args)

        if not args or not hasattr(args[0], "command_id"):
            _LOGGER.warning("Unexpected args: %s", args)
            return

        cmd_id = args[0].command_id
        raw_val = int.from_bytes(args[0].data[-1:], byteorder="big", signed=True)

        if cmd_id == 0x0265:
            self._update_attribute(0x0101, raw_val)
            _LOGGER.debug("Updated accel_x: %d", raw_val)
        elif cmd_id == 0x0266:
            self._update_attribute(0x0102, raw_val)
            _LOGGER.debug("Updated accel_y: %d", raw_val)
        elif cmd_id == 0x0267:
            self._update_attribute(0x0103, raw_val)
            _LOGGER.debug("Updated accel_z: %d", raw_val)
        else:
            _LOGGER.debug("Unhandled command_id: 0x%04x", cmd_id)

class TuyaVibrationSensor(CustomDevice):
    """Custom device representing Tuya vibration sensor _TZE200_iba1ckek."""

    signature = {
        "model": "TS0601",
        "manufacturer": "_TZE200_iba1ckek",
        "endpoints": {
            1: {
                "profile_id": zha.PROFILE_ID,
                "device_type": 0x0402,
                "input_clusters": [
                    0x0000,  # Basic
                    0x0001,  # PowerConfiguration
                    0x0003,  # Identify
                    0x0500,  # IasZone
                ],
                "output_clusters": [],
            },
        },
    }

    replacement = {
        "endpoints": {
            1: {
                "profile_id": zha.PROFILE_ID,
                "device_type": 0x0000,
                "input_clusters": [
                    Basic.cluster_id,
                    Identify.cluster_id,
                    IasZone.cluster_id,
                    PowerConfiguration.cluster_id,
                    TuyaAxisCluster,
                ],
                "output_clusters": [],
            },
        },
    }

```

Haerteleric, 2025-04-22:
could you clarify how to see the log output (also new to quircks, but have a quite solid python understanding)

Haerteleric, 2025-04-22:
nvm got it

jezzaaa, 2025-05-18:
> This has been open for almost 6 months. Has anyone figured this out? I am specifically interested in the ability to adjust sensitivity...

I'm not aware of any solution to this.

I don't believe the internal sensitivity is adjustable, but if you can get the raw accelerometer data, you'd be able to lessen the sensitivity in your HA automations. I doubt you could increase sensitivity because I suspect data is only transmitted when the internal sensitivity threshold is reached (to save battery) so fine movements simply wouldn't send any data at all.

IntegersOfK, 2025-05-19:
I think somebody had mentioned that the sensitivity settings are exposed when using some other proprietary app though. Does anybody have more information on that?

derpuma, 2025-07-04:
Can someone explain how I get the device or entity into HA? I am able to pair with Zigbee2Mqtt but I do not get a device or entity. Help would be nice!

jezzaaa, 2025-08-08:
Created pull request: https://github.com/zigpy/zha-device-handlers/pull/4242

infinitytec, 2026-11-11:
Going to bump this.

I was able to take @thknidreeg 's work and get AI to format it (I made sure there were no breaking changes!) and it works again. 

Here's the quirk I'm now running:
```
# ZHA quirk v3 for Tuya TS0601 vibration sensors with accelerometer data.
# https://github.com/zigpy/zha-device-handlers/issues/3570

import zigpy.types as t

from zhaquirks.const import BatterySize
from zhaquirks.tuya.builder import TuyaQuirkBuilder
from zigpy.quirks.v2.homeassistant import EntityPlatform, EntityType
from zigpy.quirks.v2.homeassistant.sensor import SensorStateClass

def uint_to_sint(value: t.uint8_t) -> t.int8s:
    """Convert unsigned 8-bit value to signed 8-bit value."""
    if value > 127:
        value -= 256
    return value

class VibrationSensitivity(t.enum8):
    """Vibration sensitivity setting."""
    Low = 0
    Normal = 1
    High = 2

(
    TuyaQuirkBuilder("_TZE200_iba1ckek", "TS0601")
    .applies_to("HOBEIAN", "ZG-103Z")
    .applies_to("_TZE200_hggxgsjj", "TS0601")
    .applies_to("_TZE200_yjryxpot", "TS0601")
    .applies_to("_TZE200_afycb3cg", "TS0601")

    # Tilt
    .tuya_binary_sensor(
        dp_id=7,
        attribute_name="tilt",
        entity_type=EntityType.STANDARD,
        attribute_initialized_from_cache=False,
        translation_key="tilt",
        fallback_name="Tilt",
    )

    # X-axis accelerometer
    .tuya_sensor(
        dp_id=101,
        attribute_name="x_axis",
        type=t.uint8_t,
        converter=uint_to_sint,
        state_class=SensorStateClass.MEASUREMENT,
        entity_type=EntityType.STANDARD,
        attribute_initialized_from_cache=False,
        translation_key="x_axis",
        fallback_name="X-axis",
    )

    # Y-axis accelerometer
    .tuya_sensor(
        dp_id=102,
        attribute_name="y_axis",
        type=t.uint8_t,
        converter=uint_to_sint,
        state_class=SensorStateClass.MEASUREMENT,
        entity_type=EntityType.STANDARD,
        attribute_initialized_from_cache=False,
        translation_key="y_axis",
        fallback_name="Y-axis",
    )

    # Z-axis accelerometer
    .tuya_sensor(
        dp_id=103,
        attribute_name="z_axis",
        type=t.uint8_t,
        converter=uint_to_sint,
        state_class=SensorStateClass.MEASUREMENT,
        entity_type=EntityType.STANDARD,
        attribute_initialized_from_cache=False,
        translation_key="z_axis",
        fallback_name="Z-axis",
    )

    # Vibration sensitivity
    .tuya_enum(
        dp_id=104,
        attribute_name="sensitivity",
        enum_class=VibrationSensitivity,
        entity_type=EntityType.CONFIG,
        entity_platform=EntityPlatform.SELECT,
        attribute_initialized_from_cache=False,
        translation_key="vibration_sensitivity",
        fallback_name="Vibration sensitivity",
    )

    # Tuya initialization/query behavior
    .tuya_enchantment(
        read_attr_spell=True,
        data_query_spell=True,
    )

    # Battery
    .tuya_battery(
        dp_id=4,
        battery_type=BatterySize.CR2450,
        battery_qty=1,
    )

    .skip_configuration()
    .add_to_registry()
)
```

This may be good to ship. However I tried applying this to a ZG-102ZM by HOBEIAN but didn't get good entities so that will need more work.

It might also be nice to have some way to reset the sensor values if possible.