zigpy/zha-device-handlers #3572: [Device Support Request]

e4kenndr, 2024-12-02:
### Problem description

TS0601
by _TZE204_81yrt3lo

It is for this ZigBee energy monitor clamp device: https://www.amazon.co.uk/Zigbee-Energy-Monitor-Remote-Control/product-reviews/B0CH8D7WDD/ref=cm_cr_dp_d_show_all_btm?ie=UTF8&reviewerType=all_reviews

ZHA adds the device but with zero entities showing.

### Solution description

We'd like to see the current running through the clamps to monitor energy costs. 

Currently using a skynet ZigBee dongle.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>
[image]

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

juf00th, 2024-12-11:
I support the request. China actively sells these sensors. People will need their support.

panagiks, 2024-12-16:
@e4kenndr  You might want to look at #2549 (device support request for the same device signature). Specifically into this file: https://raw.githubusercontent.com/jeverley/zha-device-handlers/refs/heads/ts0601_energy_meter_devices/zhaquirks/tuya/ts0601_energy_meter.py

I'm not sure why this hasn't made it into this repo but it did work for me on a `TS0601` with manufacturer signature `_TZE204_81yrt3lo`, with some caveats (some values are reporting negative but that might be an installation error on my part).

Hope this helps :)

marshall805, 2025-01-28:
Thanks @panagiks 

I have the same TS0601 with manufacturer signature _TZE204_81yrt3lo I'm able to get usable data now, but I'm also getting the same issues with some of the values reporting in the negative.

e4kenndr, 2025-01-29:
> Thanks [@panagiks](https://github.com/panagiks)
> 
> I have the same TS0601 with manufacturer signature _TZE204_81yrt3lo I'm able to get usable data now, but I'm also getting the same issues with some of the values reporting in the negative.

I ended up buying another wifi version in the end as I could not get the zigbee one working, however, with the wifi version, I had to say it was 0.001 conversion or something in order to get it to report in kW or 0.1 in watts - maybe that could stop the negative figures?