zigpy/zha-device-handlers #3574: TS0601 _TZE284_fzo2pocs

svanlare, 2024-12-02:
### Problem description

I have another Zemismart roller blind motors that work, and this one shows up with no support.  

### Solution description

I haven't figured out how to manually add the following quirk, but I suspect it would work.  
Quirk: zhaquirks.tuya.ts0601_cover. TuyaZemismartSmartCover0601_3

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>

### Device signature

{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4098,
    "maximum_buffer_size": 82,
    "maximum_incoming_transfer_size": 82,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 82,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xed00",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE284_fzo2pocs",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
Quirk: zhaquirks.tuya.ts0601_cover. TuyaZemismartSmartCover0601_3```

</details>

Cod053, 2025-03-31:
I also have the exact same problem with this blind motor. Can get it to add to Home Assistant with ZHA, but it's showing no entities. I found a thread from January which advised there was a previous quirk but that doesn't work now, due to a change in the Tuya chip, and another newer thread from January this year which advised the newer version does work with Zigbee2MQTT, so I think that there should be a way to get a quirk that works for the newer version in ZHA.

Cod053, 2025-03-31:
First thread advising original quirk and alleged change of Tuya firmware which rendered existing quirk inoperable: 

https://community.home-assistant.io/t/zemismart-integration-ts0601-tze284/785234

Second thread advising newer motor works with ZigBee2MQTT:

https://community.home-assistant.io/t/switch-zigbee-tze200-ts0601-paired-but-without-entities/262762/58