zigpy/zha-device-handlers #3578: TS0601 _TZE204_ltwbm23f

tco99ttocs, 2024-12-03:
### Problem description

The device is connected, but nothing is available beside the firmware version.

### Solution description

I would like to have full support for sensors and temp settings

### Screenshots/Video

[image]

### Device signature

```
{
  "node_descriptor": {
    "logical_type": 2,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 128,
    "manufacturer_code": 4417,
    "maximum_buffer_size": 66,
    "maximum_incoming_transfer_size": 66,
    "server_mask": 10752,
    "maximum_outgoing_transfer_size": 66,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0051",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": [
        "0x000a",
        "0x0019"
      ]
    }
  },
  "manufacturer": "_TZE204_ltwbm23f",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

### Diagnostic information

[zha-01J7KD0AEAPKQVM54GE1ARFDY9-_TZE204_ltwbm23f TS0601-54bfe0398cc70ad6c2e22b7e65549b9d.json](https://github.com/user-attachments/files/17995749/zha-01J7KD0AEAPKQVM54GE1ARFDY9-_TZE204_ltwbm23f.TS0601-54bfe0398cc70ad6c2e22b7e65549b9d.json)

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

i bought this one [https://de.aliexpress.com/item/1005007023469613.html?spm=a2g0o.order_list.order_list_main.15.52475c5fspQMHs&gatewayAdapt=glo2deu](url)

It looks like its still supported in Z2M
https://github.com/Koenkk/zigbee-herdsman-converters/blob/7ca2798cc12c40676bcde89907ab848cbf568039/src/devices/tuya.ts#L5235

https://github.com/Koenkk/zigbee2mqtt/discussions/24984

Could you please add it to ZHA!

pimvanelsen, 2024-12-06:
would love to see this added. guess this one is the same so i only need to add device signature
https://fr.aliexpress.com/item/1005006068601747.html?spm=a2g0o.order_detail.order_detail_item.3.74877d56DZl1kJ&gatewayAdapt=glo2fra

felixwerner, 2024-12-14:
@https://github.com/Koenkk/zigbee2mqtt/discussions/24984

PR #3769 (2025-02-26): Add Moes (Tuya) TRV602Z and TRV801Z TRVs
## Proposed change

Adds Moes _TZE204_qyr2m29i and _TZE204_ltwbm23f

Based on: https://github.com/Koenkk/zigbee-herdsman-converters/blob/79e4c8560f96e6ba52a94370137d420e2a9641b8/src/devices/tuya.ts#L5169

## Additional information

DP 2 is used for system mode and preset mode. Currently this quirk ignores preset mode and maps those values to a system mode value. #3643 may address, but I haven't had time to see how that would work when setting attributes.

Closes: #3603
Closes: #3620
Closes: #3639
Closes: #3609
Closes: #3578

## Checklist

- [ ] The chan [... cut]
closed 2025-02-26 by TheJulianJES