zigpy/zha-device-handlers #3579: [BUG] Incorrect behaviour for Detached Relay mode in Sonoff ZBMINIR2

marcohamersma, 2024-12-03:
### Bug description

I’ve been having issues using the detached relay mode on the Sonoff ZBMINIR2. I'm currently using the quirk as it was merged in #3428.

It functions fine without the detached relay feature, but with that feature turned on the behaviour is erratic and unusable. I’m using a pulse switch (as opposed to a rocker switch). Here’s some symptoms:

- Often after pressing the switch, its LED blinks in a pattern that [the manual](https://sonoff.tech/wp-content/uploads/2024/09/User-Manual-ZBMINIR2EN-V1.1.pdf) explains as a “Abnormal connection with the gateway”
- Instead of firing a ‘toggle’ event or action, presses on the physical switch turn on the switch inside Home Assistant’s device page on, then turns if off a second or two later. Sometimes repeating this action an extra time. Often followed by _the blinking_
	- This makes testing for things like double tapping useless. 
- The “sensor” it exposes in the HA UI never changes state
- It seems to merge the state of the relay and the state of the button into one ‘switch’ in the UI, even though these should be considered separately (i.e. you should be able to toggle the physical relay from HA)

It’s worth noting that today I tried to see how the switch worked with Zigbee2MQTT, and it worked without a hitch:
- The state of the physical relay turns into a switch in the HA UI
- It fires actions for each toggle, and doesn’t touch the state of the relay 
- The switch does not blink after sending an event
- Events seem to arrive instantly

This was also brought up in #3428, but not in this detail.

### Steps to reproduce

1. Turn on the pulse switch mode and detached relay mode
2. press the switch
3. Notice the events received by HA, see that the switch toggles off and on by itself after pressing the buttons

### Expected behavior

- In detached relay mode, the physical switch should trigger events inside HA without changing HA's perception of the state of the physical relay
- Pulse mode switches should either send a simple 'press' event or something like a 'press/release"event

### Screenshots/Video

<details><summary>Video</summary>

https://github.com/user-attachments/assets/09f45562-e03a-4f86-984e-ec9a0985e519

</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

lcheng33775823, 2024-12-09:
Hello TheJulianJES@TheJulianJES ,

I have investigated this issue and identified two problems:

1. ZBMINIR2 supports both the OnOff Server Cluster and the OnOff Client Cluster. I have found that when ZBMINIR2 reports the OnOff Client Toggle command, the OnOff UI on the ZHA page flips accordingly.

2. When ZBMINIR2 reports the OnOff Client Toggle command and expects a ZCL Default Response, ZHA does not provide the required ZCL Default Response.

Is there anything I need to modify?

liubotongs, 2024-12-11:
I had a similar problem:
After ZBMINIR2 enables the relay separation function and triggers the external switch, the UI interface on HA will flip the "switch" status of the control bar, and the log on the right side of it will also be recorded. Then the device network indicator blinks quickly. Through the Ubiqua Sniffer packet capture, it is observed that the device sends the On/Off Toggle command, which is a Client to sever data, but the coordinator does not reply to it. I had no such problem when I added this device to Z2M.

Fougues, 2024-12-17:
same issue here

pqhf5kd, 2024-12-19:
Same Issue here

tigger07, 2024-12-27:
same issue here with ZHA

Fougues, 2024-12-30:
Do you think this can be resolved soon? THANKS

TheJulianJES, 2025-01-04:
This might be related to how ZHA implements the cluster handler for the OnOff client cluster. It might conflict when there’s also a server cluster on the same endpoint, but I‘m not sure about this yet. It could also be something trivial.

Can anyone enable debug logs for ZHA (can be done on the HA integrations page), then reproduce the issue, disable debug logs (to download the whole HA log file) and then provide the (roughly) relevant section here?

pqhf5kd, 2025-01-04:
[home-assistant_zha_2025-01-04T10-55-08.085Z.log](https://github.com/user-attachments/files/18306539/home-assistant_zha_2025-01-04T10-55-08.085Z.log)

I've removed a small number of irrelevant lines, but I'm not sure about the rest so I've left it in.
I enable logging
Then enabled detached mode
Then toggled the lights twice
Then disabled logging

Pheerink, 2025-01-11:
I have the same problem with ZHA !

niklasstoffers, 2025-01-20:
Same issue here. I was wondering if this could be solved by just not using the detached mode at all and wiring the module differently. Shouldn't it work when you simply skip the L out wire and the neutral wire going to the bulb and wire the bulb to power directly? In that case the smart light would always have power and the switch could still be used as a remote switch for controlling the light bulb via Home Assistant. Has anyone tried that?

Tom61680, 2025-01-21:
I have the same issue with ZHA.
Additionally when S1 is quickly switched several times, the module disconnects and "Add device" function must be used to connect it again.

pqhf5kd, 2025-01-21:
That’s the device’s built in factory reset Sent from Jake Edmonds' iPhone

liubotongs, 2025-01-23:
> I have the same issue with ZHA. Additionally when S1 is quickly switched several times, the module disconnects and "Add device" function must be used to connect it again.

This is a function of the device. Quick switch 10 times in a row will restore factory Settings and search the net. It is very convenient for devices that have been installed inside the cassette.

tigger07, 2025-01-23:
> Same issue here. I was wondering if this could be solved by just not using the detached mode at all and wiring the module differently. Shouldn't it work when you simply skip the L out wire and the [...]

Of course you can connect the bulb directly to the power if it is a smart bulb. But what would be the point of using a relay switch then? In this case a simple switch without relay would do the job.
Personally I like the possibility to be able to switch off power to the bulb completely - even though it is usually turned on for smart bulbs. There are probably other scenarios switching other devices where the detached mode even makes more sense.

marafc, 2025-01-28:
Same Issue here, using ZHA

jejones90, 2025-01-29:
also having the exact same issue.  Have to leave it off now, as the bug switched the power of and on quickly causing the smart bulbs to reset and loose connectivity to hub

lukz, 2025-02-04:
I have the same issue.

dlveach, 2025-02-04:
Another voice to add.  I am also seeing the issue.  On latest firmware (1002).  Home Assistant with ZHA integration.  Using a simple single pole/single throw physical switch connected to S1/S2 for testing.   

Getting erratic behavior when using the S1/S2 switch in detached relay mode.   Nothing at all like one would expect.   Sometimes the relay is shut off/on by changing the switch, sometimes not (it should not ever change state in detached relay mode as a result of S1/S2 state change.   The "virtual" switch in HA seems to often briefly toggle back off then on again when going from the off state to the on state via the S1/S2 switch.   Doesn't seem to do this when being turned off.

Also in this integration I don't see a way in detached relay to separately trigger on the S1/S2 state change.  The device switch entity controls the relay state and what one would want to separately control via automation vs detecting S1/S2 state changes.

I don't know how much of this is a device firmware issue, ZHA issue, or both.

inthevidual, 2025-02-05:
Can confirm that this is an ZHA issue. I jumped ship to Z2M, and there it works flawlessly.

Fougues, 2025-02-05:
ZHA is the poor man's solution that only works with the bare minimum

0n3-70uch, 2025-02-09:
Hi Same problem here. Everything up-to-date.

ggu-ami, 2025-02-14:
With every HomeAssistant update, I check if anything has changed in this matter, but it's still the same. Detach still doesn't work on ZHA.

rmires, 2025-03-06:
Hi, same problem here. I had to switch to Z2MQTT to use my ZBMINIR2s which was not my preference :(. I hope this is fixed in ZHA eventually do I can switch back

narfotic, 2025-03-07:
Same issue with ZHA

richardbecht, 2025-03-08:
Same issue here

priva7eer, 2025-03-09:
Same issue here, ZBMINIR2s and ZHA

Wazbat, 2025-03-18:
I want to expand on the initial issue and say that I'm also experiencing this behaviour without using pulse mode, just using the default regular edge mode switch

Happy to provide logs if needed

Pheerink, 2025-03-20:
> Hi, same problem here. I had to switch to Z2MQTT to use my ZBMINIR2s which was not my preference :(. I hope this is fixed in ZHA eventually do I can switch back

I have 2 possibilties to use zigbee. ZHA has a very strange unpredictable reactions. MQTT is working but the detached relais mode is not. In that mode while pushing/toggle the switch no reaction in Home Assistant

Apfelstruhdl, 2025-03-30:
I just tried it again and It does not seem to report the switch state neither in normal nor in detached mode. Using the small button on the device, the relais is switched, but the status is not updated with HA. In detached mode the "Switch" control in HA seems to be both signaling the switch state as well as the switching the relais. Wouldn't we need a separate relais and switch state?

reeett, 2025-04-02:
Same issue with ZHA. It works with Z2M.

jontaylor, 2025-04-09:
In case it helps anyone else. My workaround for now is to use the zbmini in relay mode, but set up an automation to immediately turn the relay back on again, wait for the light to power back on, and then issue the relevant command. This works if you can set your lights to default to off when power is resumed.

Bappy1988, 2025-05-02:
I'm trying to set up a ZBMINIR2 so that a motion sensor controls the light when the switch is off, but will leave the light permanently on if the switch is on.

The ZBMINIR2 exposes a sensor called 'Opening' which I assumed to be state of the physical switch but its value never changes. It seems that there is no way to distinguish in ZHA the difference between the Switch's state and the Relay's state,

tobyroworth, 2025-05-11:
> https://community.hubitat.com/t/sonoff-zbminir2-driver/147085/47
> 
> Perhaps this can be helpful?

Looking through logs, and the hubitat notes @warfarincoumadin provided, it looks like a ZCL default response isn't sent - a more useful message in the attached discussion might be:

https://community.hubitat.com/t/sonoff-zbminir2-driver/147085/29

To summarise [what I think are] the important parts of the link above, the ZBMINIR2 sends the update when the switch is toggled, but if not acknowledged three times it goes to a fallback state. In the logs I'm seeing that message sent three times, so it would seem to match up.

puddly, 2025-05-11:
Is this issue still present in 2025.5.0? We changed OnOff cluster handling a bit so both cluster handlers should be set up now.

tobyroworth, 2025-05-11:
> Is this issue still present in 2025.5.0? We changed OnOff cluster handling a bit so both cluster handlers should be set up now.

@puddly Doesn't look fixed to me (on 2025.5.1).

_With "detach relay" on, the lights now come on, go off, and come on again - I don't recall this happening before, but I think it's mentioned above._

The "opening" sensor still never turns on.

I'll try and check logs in more detail when I get a chance.

range-of-motion, 2025-05-17:
I too am experiencing this issue (or at least, I think it's the same; in detach relay mode sometimes it double toggles upon clicking the physical switch).

AnatolyRugalev, 2025-05-21:
I hacked some things in zcl here: https://github.com/zigpy/zigpy/pull/1599

It's late here, but as far as I see, flipping a switch doesn't trigger fallback mode anymore. 

Also [here](https://github.com/zigpy/zha-device-handlers/pull/4084) I'm switching quirk to use newer facilities of custom clusters.

Thanks for all the inputs, it was super helpful. I learned a lot today :)

TM90, 2025-05-31:
Same issue here also on 2025.5.3

Flash777777, 2025-06-06:
> Also [here](https://github.com/zigpy/zha-device-handlers/pull/4084) I'm switching quirk to use newer facilities of custom clusters.

I implemented this quirk, but no change in the faulty behaviour of the detached mode...

uschindler, 2025-07-04:
Hi to explain better: Actually this is not an "obscure feature", it is a "security mechanism" by the ZBMINIR2: When detached mode is enabled, it is of course no longer possible to control the light when the coordinator is offline. Therefore the ZBMINIR2 implements some fallback mode: When the coordinator does not respond correctly to its messages it toggles into fallback mode and disables the detached mode temporarily. This is why you can suddenly control the lights with the rocker again. In that mode the Sonoff device has a blinking led showing communication error. After some time once it receives a new message from the coordinator it changes back to detached mode.
But unfortunately on every rocker change it detects the coordinator as "offline" and goes to fallback mode, because the message sent to coordinator gets not replied.
This cannot be fixed by a zigpy quirk/devhandler, as this is faulty behaviour of the coordinator in general. This is a fix, will try it once I figure out how to replace zigpy in HASS.

uschindler, 2025-07-04:
There is a revised fix: https://github.com/zigpy/zigpy/pull/1612

range-of-motion, 2025-07-16:
@uschindler apologies for my question–I'm merely a dumb end-user.

I'm using Home Assistant with the SkyConnect dongle, that's using Zigpy (probably, I think). I currently have various ZBMINIR2's that do not worked as intended.

Is there an ETA of when this will be fixed? And am I correct to assume this has to be shipped with a Home Assistant update (or can it be shipped through a separate update on the [ZHA integration](https://www.home-assistant.io/integrations/zha/)?

Thank you! 🙇

uschindler, 2025-07-16:
> And am I correct to assume this has to be shipped with a Home Assistant update (or can it be shipped through a separate update on the [ZHA integration](https://www.home-assistant.io/integrations/zha/)?

NMo needs an update of HASS. You can theoretically replace the zigpy version inside home assistant by messing with the docker containers, but this is nothing fo normal end-users.

> Is there an ETA of when this will be fixed?

No not yet. The solution is a bit more complex as this is a bigger problem in zigpy that needs a bigger cleanup. The ZBMINIR2's problems are just the top of iceberg. The attached patch above does not solve the problem completely there are different fixes to be applied before.

But when everything is solved (I tried the fix already and it works, but breaks other devices), the following works: The "Opening" sensor will show the state of the physical switch (also in normal, not detached mode). You can register an automation on it when its state changes to detect a change in physical switch.

range-of-motion, 2025-07-16:
@uschindler thank you for the clear answer.

As Zigpy is open-source, there is the possibility of this being "broken" (for my use-case) for another year, with contributors casually choosing to spend time on this.

Is that the expectation, or is the aim to have this resolved within a couple of weeks or months?

I'm asking because I need to figure out whether to how deal with broken light switches, and for how long.

Thanks again!

MindFreeze, 2025-07-26:
> Just to confirm that this bug is affecting also [Sonoff ZBM5](https://sonoff.tech/products/sonoff-switchman-zigbee-smart-wall-switch-zbm5)

ZBM5's detached relay works differently. I am working on a quirk for it in #4198

uschindler, 2025-07-26:
> > Just to confirm that this bug is affecting also [Sonoff ZBM5](https://sonoff.tech/products/sonoff-switchman-zigbee-smart-wall-switch-zbm5)
> 
> ZBM5's detached relay works differently. I am working on a quirk for it in [#4198](https://github.com/zigpy/zha-device-handlers/pull/4198)

That's correct. The PR fixes the missing settings for this device. But the problem with missing N is unrelated to the issues here. If the ZBM5 has an input and output On off cluster it will get affected by the same problems as described here. One easy way to figure this out: does the device also shows the "Opening" sensor and is this one broken?

I figured out that all PRs related to this bug were merged by @puddly  for this issue. I will give it a try later my replacing Zigpy in my HA and report back. I will try to test for affects of this change with other devices here. In the first fixes many other devices broke due to the change as the direction was misinterpreted. Hue RDM04 no longer reported button presses, and various onoff devices did not update on actual onoff state anymore.

Many many thanks for looking into this.

uschindler, 2025-07-26:
I tested it, with latest dev version it looks like it works, see report here: https://github.com/zigpy/zigpy/pull/1612#issuecomment-3122164643

david-romero, 2025-10-29:
Hi @uschindler , thanks a lot for your help! I'm trying to get this feature working on my ZBMINI Extreme, but in the configuration section I only see the “Initial Behavior” option. I’ve already reconfigured the device, but the situation remains the same.

[image]

thijseurlingsroutz, 2025-10-30:
Same issue here for the ZBM5-1C-80/86. I'm missing the detach function, that was the reason I've bought the device for:

[image]