zigpy/zha-device-handlers #3581: [BUG]If the unit is set, the operation information will not be displayed in the logbook after the control is set for the number component of Quirks v2

hwzolin, 2024-12-04:
### Bug description

I added two UI components based on Quirks v2 in the code, the type is Number. After repeated testing, I found that component B, which has a set unit, has no information recorded in the logbook after control. However, if you don't have a unit set for component A, you can see the record information you just operated in the logbook after waiting for a while.

Here's my full code, check it out:

```
"""Third Reality temperature and humidity sensor devices."""

from typing import Final

from zigpy.quirks import CustomCluster
from zigpy.quirks.v2 import QuirkBuilder
from zigpy.quirks.v2.homeassistant import UnitOfTemperature
import zigpy.types as t
from zigpy.zcl.foundation import BaseAttributeDefs, ZCLAttributeDef

class ThirdRealityTemperatureAndHumidityCluster(CustomCluster):
    """Third Reality's temperature and humidity sensor private cluster."""

    cluster_id = 0xFF01

    class AttributeDefs(BaseAttributeDefs):
        """Define the attributes of a private cluster."""

        temperature_correction_fahrenheit: Final = ZCLAttributeDef(
            id=0x0033,
            type=t.int16s,
            is_manufacturer_specific=True,
        )

        temperature_correction_celsius: Final = ZCLAttributeDef(
            id=0x0031,
            type=t.int16s,
            is_manufacturer_specific=True,
        )

        humidity_correction: Final = ZCLAttributeDef(
            id=0x0032,
            type=t.int16s,
            is_manufacturer_specific=True,
        )

(
    QuirkBuilder("Third Reality, Inc", "3RTHS24BZ")
    .also_applies_to("Third Reality, Inc", "3RTHS0224Z")
    .replaces(ThirdRealityTemperatureAndHumidityCluster)
    .number(
        attribute_name=ThirdRealityTemperatureAndHumidityCluster.AttributeDefs.temperature_correction_celsius.name,
        min_value=-1000,
        max_value=1000,
        step=1,
        # unit=UnitOfTemperature.CELSIUS,
        cluster_id=ThirdRealityTemperatureAndHumidityCluster.cluster_id,
        translation_key="temperature_correction_celsius",
        # fallback_name="Correction/100",
        fallback_name="A",
    )
    .number(
        attribute_name=ThirdRealityTemperatureAndHumidityCluster.AttributeDefs.temperature_correction_fahrenheit.name,
        min_value=-1800,
        max_value=1800,
        step=1,
        unit=UnitOfTemperature.FAHRENHEIT,
        cluster_id=ThirdRealityTemperatureAndHumidityCluster.cluster_id,
        translation_key="temperature_correction_fahrenheit",
        # fallback_name="Correction/100",
        fallback_name="B",
    )
    .number(
        attribute_name=ThirdRealityTemperatureAndHumidityCluster.AttributeDefs.humidity_correction.name,
        min_value=-5000,
        max_value=5000,
        step=1,
        cluster_id=ThirdRealityTemperatureAndHumidityCluster.cluster_id,
        translation_key="humidity_correction",
        fallback_name="Humidity Correction/100",
    )
    .add_to_registry()
)
```

### Steps to reproduce

1. enable_quirks set to True.
2. Set up the custom_quirks_path and place the file in a directory.
3. Add temperature and humidity sensors.
4. Control the UI components that are A and B separately fallback_name.
5. After the control is complete, check whether the control information exists in the logbook.

### Expected behavior

Regardless of whether the unit is set or not, the operation record must be recorded in the logbook after the operation is completed.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]

</details>
[image]

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

hwzolin, 2024-12-17:
Can anyone help answer this question?

TheJulianJES, 2024-12-17:
`unit` shouldn't affect the state, but I wonder if HA is doing this intentionally. This might be for not spamming the logbook for sensors and so on.

Just to confirm that the state is correctly updated:
1. Are both attributes writeable on the device? Can you use the "Manage Zigbee device" menu (three dots on the left device card) and then read the attributes you changed using the number entities?
2. If you read them, do you (a) see the updated value and (b) then see the logbook entries?
3. If you refresh the page after changing a number entity, does it show the same (new) value or a previous value?

The logbook entries reflect the actual state in HA. It'll be updated when we get a confirmation the attribute was written correctly or when you read it. Zigpy attribute cache will updated and also update HA state (adding a logbook entry).
If the state is updated correctly, it might be down to HA behavior to not show this. If that's the case, we can't change that behavior.

TheJulianJES, 2025-03-26:
I reproduced this. This is a really weird issue.. It's only a visual issue in Home Assistant, directly probably (so not a ZHA issue). Maybe even intentional? Not sure..
The entity still updates as expected and it shows on the logbook page for all entities. It's only missing on the device-specific logbook on the right. So, you should still add the correct units. Other than it missing in that logbook section on the right, everything seems to work fine still.

hwzolin, 2025-03-26:
> I reproduced this. This is a really weird issue.. It's only a visual issue in Home Assistant, directly probably (so not a ZHA issue). ~Maybe even intentional? Not sure..~ Clicking on the number entity, the logbook section of that doesn't load. There's an endless spinner, so this really seems like some sort of Home Assistant bug. Weird.. Same issue with the Matter ingration.
> 
> The entity still updates as expected and it shows on the logbook page for all entities. It's only missing on the device-specific logbook on the right. So, you should still add the correct units. Other than it missing in that logbook section on the right, everything seems to work fine still.

Well......
Can you please communicate with HA's official staff about this issue? Because it doesn't seem like a way to go on like this.

TheJulianJES, 2025-09-13:
The behavior should be more streamlined in 2025.10.x with this:
- https://github.com/home-assistant/core/pull/149349
- https://github.com/home-assistant/frontend/issues/18847

closed 2025-09-13 by TheJulianJES