zigpy/zha-device-handlers #3582: [BUG] Why can't I use the number component fallback_name "Humidity Correction/100" when I set unit to PERCENTAGE? And I'll replace the unit with something else.

hwzolin, 2024-12-04:
### Bug description

Here's my full code:

```
"""Third Reality temperature and humidity sensor devices."""

from typing import Final

from zigpy.quirks import CustomCluster
from zigpy.quirks.v2 import QuirkBuilder
from zigpy.quirks.v2.homeassistant import UnitOfTemperature,PERCENTAGE
import zigpy.types as t
from zigpy.zcl.foundation import BaseAttributeDefs, ZCLAttributeDef

class ThirdRealityTemperatureAndHumidityCluster(CustomCluster):
    """Third Reality's temperature and humidity sensor private cluster."""

    cluster_id = 0xFF01

    class AttributeDefs(BaseAttributeDefs):
        """Define the attributes of a private cluster."""

        temperature_correction_fahrenheit: Final = ZCLAttributeDef(
            id=0x0033,
            type=t.int16s,
            is_manufacturer_specific=True,
        )

        temperature_correction_celsius: Final = ZCLAttributeDef(
            id=0x0031,
            type=t.int16s,
            is_manufacturer_specific=True,
        )

        humidity_correction: Final = ZCLAttributeDef(
            id=0x0032,
            type=t.int16s,
            is_manufacturer_specific=True,
        )

(
    QuirkBuilder("Third Reality, Inc", "3RTHS24BZ")
    .also_applies_to("Third Reality, Inc", "3RTHS0224Z")
    .replaces(ThirdRealityTemperatureAndHumidityCluster)
    .number(
        attribute_name=ThirdRealityTemperatureAndHumidityCluster.AttributeDefs.temperature_correction_celsius.name,
        min_value=-1000,
        max_value=1000,
        step=1,
        # unit=UnitOfTemperature.CELSIUS,
        cluster_id=ThirdRealityTemperatureAndHumidityCluster.cluster_id,
        translation_key="temperature_correction_celsius",
        # fallback_name="Correction/100",
        fallback_name="A",
    )
    .number(
        attribute_name=ThirdRealityTemperatureAndHumidityCluster.AttributeDefs.temperature_correction_fahrenheit.name,
        min_value=-1800,
        max_value=1800,
        step=1,
        unit=UnitOfTemperature.FAHRENHEIT,
        cluster_id=ThirdRealityTemperatureAndHumidityCluster.cluster_id,
        translation_key="temperature_correction_fahrenheit",
        # fallback_name="Correction/100",
        fallback_name="B",
    )
    .number(
        attribute_name=ThirdRealityTemperatureAndHumidityCluster.AttributeDefs.humidity_correction.name,
        min_value=-5000,
        max_value=5000,
        step=1,
        unit=PERCENTAGE,
        cluster_id=ThirdRealityTemperatureAndHumidityCluster.cluster_id,
        translation_key="humidity_correction",
        fallback_name="Humidity Correction/100",
    )
    .add_to_registry()
)

```

In the following code block, I've set unit to PERCENTAGE：
```
 .number(
        attribute_name=ThirdRealityTemperatureAndHumidityCluster.AttributeDefs.humidity_correction.name,
        min_value=-5000,
        max_value=5000,
        step=1,
        unit=PERCENTAGE,
        cluster_id=ThirdRealityTemperatureAndHumidityCluster.cluster_id,
        translation_key="humidity_correction",
        fallback_name="Humidity Correction/100",
    )
```
Then when I add a device, the "Humidity Correction/100" component of the device in the detail page is not controlled.

### Steps to reproduce

1. Move the code file into the custom_quirks_path directory.
2. Add the device to ZHA.
3. Try to control the page component with the fallback_name "Humidity Correction/100".

### Expected behavior

When unit=PERCENTAGE, the UI components can still be used normally.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[Paste/upload your media here]
[image]
</details>

### Device signature

<details><summary>Device signature</summary>

```json
[Paste the device signature here]
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
[Paste the diagnostic information here]
```

</details>

### Logs

<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

hwzolin, 2024-12-17:
Can anyone help answer this question?

TheJulianJES, 2024-12-17:
Some units (like percentage) aren't accepted in quirks v2 yet. We're tracking that bug here: https://github.com/zigpy/zha/issues/327

closed 2025-08-02 by github-actions[bot]