zigpy/zha-device-handlers #3584: Please add support in ZHA for TZE200_ya4ft0w4 Tuya M100 Presence Sensor

Rabman416, 2024-12-05:
### Problem description

The Tuya Presence Sensor is not supported in the ts0601_motion.py Quirk. 

### Solution description

Would like the presence sensor device to show the controls and sensors.

### Screenshots/Video

<details><summary>Screenshots/Video</summary>

[image]

</details>

### Device signature

<details><summary>Device signature</summary>

```json
{
  "node_descriptor": {
    "logical_type": 1,
    "complex_descriptor_available": 0,
    "user_descriptor_available": 0,
    "reserved": 0,
    "aps_flags": 0,
    "frequency_band": 8,
    "mac_capability_flags": 142,
    "manufacturer_code": 4660,
    "maximum_buffer_size": 108,
    "maximum_incoming_transfer_size": 0,
    "server_mask": 11264,
    "maximum_outgoing_transfer_size": 0,
    "descriptor_capability_field": 0
  },
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0x0100",
      "input_clusters": [
        "0x0000",
        "0x0004",
        "0x0005",
        "0xef00"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "_TZE200_ya4ft0w4",
  "model": "TS0601",
  "class": "zigpy.device.Device"
}
```

</details>

### Diagnostic information

<details><summary>Diagnostic information</summary>

```json
{
  "home_assistant": {
    "installation_type": "Home Assistant OS",
    "version": "2024.12.0",
    "dev": false,
    "hassio": true,
    "virtualenv": false,
[... 537 lines cut ...]
<details><summary>Logs</summary>

```python
[Paste the logs here]
```

</details>

### Custom quirk

<details><summary>Custom quirk</summary>

```python
[Paste your custom quirk here]
```

</details>

### Additional information

I tried entering the device name in the different various sections of the current Quirk with no success. It only shows two Diagnostic entities. Does not show the controls or sensor information.

This is the link to the device 24G Presence sensor on AliExpress. I did order from this vendor before and the device name on that order came as _TZE204_7gclukjs and it works.

https://www.aliexpress.com/item/1005005249827156.html?spm=a2g0o.order_list.order_list_main.5.54441802Kg4T6C

pjunzip, 2024-12-09:
I also got this model, TZE200_ya4ft0w4, which seems to be a new version. There are no quirks available for it yet. Could you please add support for this device?

With this model, in the Smart Life app, the human detection status is displayed as three states: cleared, moving, and exits. However, when connected to ZHA, no entities are detected at all. If I connect it through the Tuya integration via the Tuya Zigbee hub in Home Assistant, it only displays two states: cleared and occupancy.

The issue is that when the status is moving in the Smart Life app, it is interpreted as cleared in Home Assistant, which makes it unusable.

I would prefer ZHA to support this device rather than relying on the Tuya integration. Thank you!

Itay1787, 2024-12-11:
Hi, I have the same model. Does he have no quirk at all?
Is it impossible to take from the other models that are similar to it?
I tried to take this code - https://gist.githubusercontent.com/fixtse/09316d48f23688938fce31add711ea07/raw/358353f9274fe0654cd784f4301cab011ddda942/TS0601_radar.py from here -https://fixtse.com/blog/zy-m100-full-zha-support
and added his model in the model sections, but it didn't work.
Do I need to add the model in another place for it to work?

Rabman416, 2024-12-11:
> Hi, I have the same model. Does he have no quirk at all? Is it impossible to take from the other models that are similar to it? I tried to take this code - [...]

I tried putting the model in various areas with no success. I think it involves more than one that.

Rabman416, 2024-12-12:
> This should get you started, the rest of the DPs can be found in the [z2m source](https://github.com/Koenkk/zigbee-herdsman-converters/blob/989f2d91a4c3328d6eb90d254593bb7a4b6b76ca/src/devices/tuya.ts#L10872)
> 
> This is mapping the presence and move states to an occupied state.
> 
> ```python
> from zhaquirks.tuya.builder import TuyaQuirkBuilder
> from zhaquirks.tuya import TuyaLocalCluster
> from zigpy.zcl.clusters.measurement import OccupancySensing
> 
> class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
>     """Tuya local OccupancySensing cluster."""
> 
> (
>     TuyaQuirkBuilder("_TZE200_ya4ft0w4", "TS0601")
>     .tuya_dp(
>         dp_id=1,
>         ep_attribute=TuyaOccupancySensing.ep_attribute,
>         attribute_name=OccupancySensing.AttributeDefs.occupancy.name,
>         converter=lambda x: True if x in (1, 2) else False,
>     )
>     .skip_configuration()
>     .add_to_registry()
> )
> ```

Thank you. Not sure how I would implement this. Would I add it to the existing quirk?

prairiesnpr, 2024-12-12:
> Thank you. Not sure how I would implement this. Would I add it to the existing quirk?

Assuming you have enabled custom quirks, if not the ZHA docs detail how to enable them, then just save it to a file in the custom quirk directory. It's the newer v2 format, see the following links for more details on extending it.

https://github.com/zigpy/zha-device-handlers/blob/dev/tuya.md
https://github.com/zigpy/zha-device-handlers/wiki/Tuya-%E2%80%90-v2-Quirk-with-TuyaQuirkBuilder

If you get a `MultipleQuirksMatchException` error, see https://github.com/zigpy/zigpy/issues/1508

Rabman416, 2024-12-12:
I will give it a try.

Rabman416, 2024-12-12:
Well I looked into things this morning and realized I am out of my depth. I don't know how I integrate this new code with my existing quirk which I have included below. I do have other presence sensors and don't want to mess them up.

`"""BlitzWolf IS-3/Tuya motion rechargeable occupancy sensor."""

import math
from typing import Dict, Optional, Tuple, Union

from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    AnalogOutput,
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing,
    RelativeHumidity,
    TemperatureMeasurement,
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    MOTION_EVENT,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster, TuyaManufCluster, TuyaNewManufCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

ZONE_TYPE = 0x0001

class TuyaMmwRadarSelfTest(t.enum8):
    """Mmw radar self test values."""

    TESTING = 0
    TEST_SUCCESS = 1
    TEST_FAILURE = 2
    OTHER = 3
    COMM_FAULT = 4
    RADAR_FAULT = 5

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaAnalogInput(AnalogInput, TuyaLocalCluster):
    """Tuya local AnalogInput cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class TuyaMmwRadarSensitivity(TuyaAttributesCluster, AnalogOutput):
    """AnalogOutput cluster for sensitivity."""
[... 539 lines cut ...]
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarDetectionDelay,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarFadingTime,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }`

Rabman416, 2024-12-12:
`"""BlitzWolf IS-3/Tuya motion rechargeable occupancy sensor."""

import math
from typing import Dict, Optional, Tuple, Union

from zigpy.profiles import zgp, zha
from zigpy.quirks import CustomDevice
import zigpy.types as t
from zigpy.zcl import foundation
from zigpy.zcl.clusters.general import (
    AnalogInput,
    AnalogOutput,
    Basic,
    GreenPowerProxy,
    Groups,
    Identify,
    Ota,
    Scenes,
    Time,
)
from zigpy.zcl.clusters.measurement import (
    IlluminanceMeasurement,
    OccupancySensing,
    RelativeHumidity,
    TemperatureMeasurement,
)
from zigpy.zcl.clusters.security import IasZone

from zhaquirks import Bus, LocalDataCluster, MotionOnEvent
from zhaquirks.const import (
    DEVICE_TYPE,
    ENDPOINTS,
    INPUT_CLUSTERS,
    MODELS_INFO,
    MOTION_EVENT,
    OUTPUT_CLUSTERS,
    PROFILE_ID,
)
from zhaquirks.tuya import TuyaLocalCluster, TuyaManufCluster, TuyaNewManufCluster
from zhaquirks.tuya.mcu import (
    DPToAttributeMapping,
    TuyaAttributesCluster,
    TuyaMCUCluster,
)

ZONE_TYPE = 0x0001

class TuyaMmwRadarSelfTest(t.enum8):
    """Mmw radar self test values."""

    TESTING = 0
    TEST_SUCCESS = 1
    TEST_FAILURE = 2
    OTHER = 3
    COMM_FAULT = 4
    RADAR_FAULT = 5

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

class TuyaAnalogInput(AnalogInput, TuyaLocalCluster):
    """Tuya local AnalogInput cluster."""

class TuyaIlluminanceMeasurement(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya local IlluminanceMeasurement cluster."""

class TuyaMmwRadarSensitivity(TuyaAttributesCluster, AnalogOutput):
    """AnalogOutput cluster for sensitivity."""

    def __init__(self, *args, **kwargs):
[... 537 lines cut ...]
            },
            4: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarDetectionDelay,
                ],
                OUTPUT_CLUSTERS: [],
            },
            5: {
                PROFILE_ID: zha.PROFILE_ID,
                DEVICE_TYPE: zha.DeviceType.COMBINED_INTERFACE,
                INPUT_CLUSTERS: [
                    TuyaMmwRadarFadingTime,
                ],
                OUTPUT_CLUSTERS: [],
            },
            242: {
                PROFILE_ID: zgp.PROFILE_ID,
                DEVICE_TYPE: zgp.DeviceType.PROXY_BASIC,
                INPUT_CLUSTERS: [],
                OUTPUT_CLUSTERS: [GreenPowerProxy.cluster_id],
            },
        }
    }`

prairiesnpr, 2024-12-12:
Just place the following in a file named `tuya__TZE200_ya4ft0w4.py` and then place that file in your custom quirks directory, usually something like `config/custom_zha_quirks`, the directory would be configured in `configuration.yaml`.

```python
from zhaquirks.tuya.builder import TuyaQuirkBuilder
from zhaquirks.tuya import TuyaLocalCluster
from zigpy.zcl.clusters.measurement import OccupancySensing

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

(
    TuyaQuirkBuilder("_TZE200_ya4ft0w4", "TS0601")
    .tuya_dp(
        dp_id=1,
        ep_attribute=TuyaOccupancySensing.ep_attribute,
        attribute_name=OccupancySensing.AttributeDefs.occupancy.name,
        converter=lambda x: True if x in (1, 2) else False,
    )
    .adds(TuyaOccupancySensing) 
    .skip_configuration()
    .add_to_registry()
)
```

Rabman416, 2024-12-12:
Awesome. I will try that.

Rabman416, 2024-12-12:
Thanks for the additional instructions. I now have Occupancy showing up in HA for the sensor which is the most important. I guess I will wait until other controls and sensors are added. I am most appreciative!

prairiesnpr, 2024-12-12:
> Thanks for the additional instructions. I now have Occupancy showing up in HA for the sensor which is the most important. I guess I will wait until other controls and sensors are added. I am most appreciative!

Is it working as expected? If do, the rest of them should be fairly easy to add, just a matter of working through them.

Rabman416, 2024-12-12:
Yes, it appears to be working as expected.

Unfortunately I do not have the skills to add the other items.

Itay1787, 2024-12-12:
> Just place the following in a file named `tuya_TZE200_ya4ft0w4.py` and then place that file in your custom quirks directory, usually something like `config/custom_zha_quirks`, the directory would be [...]

It works for me too!
Unfortunately, I don't know how to convert from z2m or the files of the other models that will work with it.
The other models are the same product, just a different model number.
I tried to look in the docs but I didn't understand anything.
I'm currently trying to do something with ChatGPT, but even that, at least at the moment, turns out not to help

Rabman416, 2024-12-12:
> It works for me too! Unfortunately, I don't know how to convert from z2m or the files of the other models that will work with it. The other models are the same product, just a different model [...]

Let me know if you have any success.

prairiesnpr, 2024-12-13:
```python
from zigpy.quirks.v2 import EntityType
from zigpy.quirks.v2.homeassistant import UnitOfLength, UnitOfTime
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
import zigpy.types as t
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement, OccupancySensing

from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.builder import TuyaQuirkBuilder

class TuyaIlluminanceCluster(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya Illuminance cluster."""

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

(
    TuyaQuirkBuilder("_TZE200_ya4ft0w4", "TS0601")
    .tuya_dp(
        dp_id=1,
        ep_attribute=TuyaOccupancySensing.ep_attribute,
        attribute_name=OccupancySensing.AttributeDefs.occupancy.name,
        converter=lambda x: True if x in (1, 2) else False,
    )
    .tuya_number(
        dp_id=2,
        attribute_name="move_sensitivity",
        type=t.uint16_t,
        min_value=0,
        max_value=10,
        step=1,
        translation_key="move_sensitivity",
        fallback_name="Motion sensitivity",
    )
    .tuya_number(
        dp_id=3,
        attribute_name="detection_distance_min",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.METERS,
        min_value=0,
        max_value=8.25,
        step=0.75,
        translation_key="detection_distance_min",
        fallback_name="Minimum range",
    )
    .tuya_number(
        dp_id=4,
        attribute_name="detection_distance_max",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.METERS,
        min_value=0.75,
        max_value=9.0,
        step=0.75,
        translation_key="detection_distance_max",
        fallback_name="Maximum range",
    )
    .tuya_sensor(
        dp_id=9,
        attribute_name="distance",
        type=t.uint16_t,
        divisor=10,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.METERS,
        entity_type=EntityType.STANDARD,
        translation_key="distance",
        fallback_name="Target distance",
    )
    .tuya_switch(
        dp_id=101,
        attribute_name="find_switch",
        entity_type=EntityType.STANDARD,
        translation_key="find_switch",
        fallback_name="Distance switch",
    )
    .tuya_number(
        dp_id=102,
        attribute_name="presence_sensitivity",
        type=t.uint16_t,
        min_value=0,
        max_value=10,
        step=1,
        translation_key="presence_sensitivity",
        fallback_name="Presence sensitivity",
    )
    .tuya_dp(
        dp_id=103,
        ep_attribute=TuyaIlluminanceCluster,
        attribute_name=TuyaIlluminanceCluster.AttributeDefs.measured_value.name,
    )
    .adds(TuyaIlluminanceCluster)
    .tuya_number(
        dp_id=105,
        attribute_name="presence_timeout",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DURATION,
        unit=UnitOfTime.SECONDS,
        min_value=1,
        max_value=15000,
        step=1,
        translation_key="presence_timeout",
        fallback_name="Fade time",
    )
    .skip_configuration()
    .add_to_registry()
)
```

Rabman416, 2024-12-13:
Unfortunately it is not working. ZHA fails to load.

[image]

This is an excerpt of some of the Log file if this helps.

2024-12-13 18:46:39.464 WARNING (SyncWorker_7) [zhaquirks] Loaded custom quirks. Please contribute them to https://github.com/zigpy/zha-device-handlers
2024-12-13 18:46:39.467 WARNING (MainThread) [zigpy.config.validators] The `ikea_provider` key is deprecated, migrate your configuration to the `extra_providers` list instead: `extra_providers: [{'type': 'ikea'}]`
2024-12-13 18:46:39.737 ERROR (MainThread) [homeassistant.config_entries] Error setting up entry Sonoff Zigbee 3.0 USB Dong - /dev/serial/by-id/usb-ITead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_e82ffa45aac9eb11b0058f4f1d69213e-if00-port0, s/n: e82ffa45aac9eb11b0058f4f1d69213e - ITead - 10C4:EA60 for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 132, in async_setup_entry
    async with radio_mgr.connect_zigpy_app() as app:
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/contextlib.py", line 214, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/radio_manager.py", line 182, in connect_zigpy_app
    app = await self.radio_type.controller.new(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        app_config, auto_form=False, start_radio=False
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zigpy/application.py", line 239, in new
    await app._load_db()
  File "/usr/local/lib/python3.13/site-packages/zigpy/application.py", line 111, in _load_db
    await self._dblistener.load()
  File "/usr/local/lib/python3.13/site-packages/zigpy/appdb.py", line 684, in load
    device = zigpy.quirks.get_device(device)
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/__init__.py", line 43, in get_device
    return _DEVICE_REGISTRY.get_device(device)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/registry.py", line 130, in get_device
    return quirk_entry.create_device(device)
           ~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/v2/__init__.py", line 429, in create_device
    return CustomDeviceV2(device.application, device.ieee, device.nwk, device, self)
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/v2/__init__.py", line 96, in __init__
    replace_meta(self)
    ~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/v2/__init__.py", line 212, in __call__
    self.add(device)
    ~~~~~~~~^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/v2/__init__.py", line 173, in __call__
    cluster = self.cluster(endpoint, is_server=is_server_cluster)
  File "/usr/local/lib/python3.13/site-packages/zhaquirks/tuya/mcu/__init__.py", line 209, in __init__
    super().__init__(*args, **kwargs)
    ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zhaquirks/tuya/__init__.py", line 1531, in __init__
    cluster = getattr(endpoint, dp_map.ep_attribute, None)
TypeError: attribute name must be string, not 'type'

Itay1787, 2024-12-14:
Hi, first of all, thanks for the help!
Unfortunately, it doesn't work for me either.
This is the logs:

Log 1:

`Logger: homeassistant.config_entries
Source: config_entries.py:635
First occurred: 01:53:24 (1 occurrences)
Last logged: 01:53:24

Error setting up entry Zigbee-Hub for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 635, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 132, in async_setup_entry
    async with radio_mgr.connect_zigpy_app() as app:
  File "/usr/local/lib/python3.12/contextlib.py", line 210, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/radio_manager.py", line 182, in connect_zigpy_app
    app = await self.radio_type.controller.new(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 238, in new
    await app._load_db()
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 110, in _load_db
    await self._dblistener.load()
  File "/usr/local/lib/python3.12/site-packages/zigpy/appdb.py", line 684, in load
    device = zigpy.quirks.get_device(device)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/quirks/__init__.py", line 43, in get_device
    return _DEVICE_REGISTRY.get_device(device)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/quirks/registry.py", line 130, in get_device
    return quirk_entry.create_device(device)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/quirks/v2/__init__.py", line 429, in create_device
    return CustomDeviceV2(device.application, device.ieee, device.nwk, device, self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/quirks/v2/__init__.py", line 96, in __init__
    replace_meta(self)
  File "/usr/local/lib/python3.12/site-packages/zigpy/quirks/v2/__init__.py", line 212, in __call__
    self.add(device)
  File "/usr/local/lib/python3.12/site-packages/zigpy/quirks/v2/__init__.py", line 173, in __call__
    cluster = self.cluster(endpoint, is_server=True)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zhaquirks/tuya/mcu/__init__.py", line 209, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/zhaquirks/tuya/__init__.py", line 1529, in __init__
    cluster = getattr(endpoint, dp_map.ep_attribute, None)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: attribute name must be string, not 'type'`

Log 2:

`Error setting up entry Zigbee-Hub for zha

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 635, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 132, in async_setup_entry
    async with radio_mgr.connect_zigpy_app() as app:
  File "/usr/local/lib/python3.12/contextlib.py", line 210, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/radio_manager.py", line 182, in connect_zigpy_app
    app = await self.radio_type.controller.new(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 238, in new
    await app._load_db()
  File "/usr/local/lib/python3.12/site-packages/zigpy/application.py", line 110, in _load_db
    await self._dblistener.load()
  File "/usr/local/lib/python3.12/site-packages/zigpy/appdb.py", line 684, in load
    device = zigpy.quirks.get_device(device)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/quirks/__init__.py", line 43, in get_device
    return _DEVICE_REGISTRY.get_device(device)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zigpy/quirks/registry.py", line 125, in get_device
    raise MultipleQuirksMatchException(
zigpy.exceptions.MultipleQuirksMatchException: Multiple matches found for device <Device model='TS0601' manuf='_TZE200_ya4ft0w4' nwk=0xD4CC ieee=e4:98:bb:43:00:8d:58:22 is_initialized=True>: [QuirksV2RegistryEntry(quirk_file=PosixPath('/usr/local/lib/python3.12/site-packages/zhaquirks/tuya/builder/__init__.py'), quirk_file_line=75, manufacturer_model_metadata=(ManufacturerModelMetadata(manufacturer='_TZE200_ya4ft0w4', model='TS0601'),), friendly_name=None, device_alerts=(), filters=(), custom_device_class=None, device_node_descriptor=None, skip_device_configuration=True, adds_metadata=(AddsMetadata(cluster=<class 'tuya_motion_sensor_ya4ft0w4.TuyaIlluminanceCluster'>, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, constant_attributes=frozendict.frozendict({})),), removes_metadata=(), replaces_metadata=(ReplacesMetadata(remove=RemovesMetadata(cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>), add=AddsMetadata(cluster=<class [... cut ...] type=<EntityType.CONFIG: 'config'>, cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='presence_timeout', fallback_name='Fade time', attribute_name='presence_timeout', reporting_config=None, min=1, max=15000, step=1, unit=<UnitOfTime.SECONDS: 's'>, mode=None, multiplier=None, device_class=<SensorDeviceClass.DURATION: 'duration'>)), device_automation_triggers_metadata=frozendict.frozendict({}))]`

prairiesnpr, 2024-12-14:
I missed `ep_attribute` on 103, edited above, see if that works. I don't have the device.

Rabman416, 2024-12-14:
Making progress. Occupancy is no longer working.

[image]

prairiesnpr, 2024-12-14:
Try the updated version above

Rabman416, 2024-12-14:
I added this and it now appears to work:

[image]

Itay1787, 2024-12-14:
Weirdly, I don't have the same sensors and controls in Home Assistance as you do

[image]

> Making progress. Occupancy is no longer working. [...]

Itay1787, 2024-12-14:
Hi, I found more log that may explain why I don't have all the controls

`Logger: zha.application.gateway
Source: /usr/local/lib/python3.12/site-packages/zha/application/gateway.py:339
First occurred: 02:28:48 (3 occurrences)
Last logged: 02:28:48

Failed to create platform entity: <class 'zha.application.platforms.number.NumberConfigurationEntity'> [args=('e4:98:bb:43:00:8d:58:22-1', [<zha.zigbee.cluster_handlers.manufacturerspecific.TuyaClusterHandler object at 0x7f31abd456d0>], <zha.zigbee.endpoint.Endpoint object at 0x7f31abd45490>, <CustomDeviceV2 model='TS0601' manuf='_TZE200_ya4ft0w4' nwk=0xD4CC ieee=e4:98:bb:43:00:8d:58:22 is_initialized=True> - quirk_applied: True - quirk_or_device_class: zigpy.quirks.v2.CustomDeviceV2 - quirk_id: None), kwargs={'entity_metadata': NumberMetadata(entity_platform=<EntityPlatform.NUMBER: 'number'>, entity_type=<EntityType.CONFIG: 'config'>, cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='detection_distance_min', fallback_name='Minimum range', attribute_name='detection_distance_min', reporting_config=None, min=0, max=8.25, step=0.75, unit=<UnitOfLength.METERS: 'm'>, mode=None, multiplier=None, device_class=<SensorDeviceClass.DISTANCE: 'distance'>)}]
Failed to create platform entity: <class 'zha.application.platforms.number.NumberConfigurationEntity'> [args=('e4:98:bb:43:00:8d:58:22-1', [<zha.zigbee.cluster_handlers.manufacturerspecific.TuyaClusterHandler object at 0x7f31abd456d0>], <zha.zigbee.endpoint.Endpoint object at 0x7f31abd45490>, <CustomDeviceV2 model='TS0601' manuf='_TZE200_ya4ft0w4' nwk=0xD4CC ieee=e4:98:bb:43:00:8d:58:22 is_initialized=True> - quirk_applied: True - quirk_or_device_class: zigpy.quirks.v2.CustomDeviceV2 - quirk_id: None), kwargs={'entity_metadata': NumberMetadata(entity_platform=<EntityPlatform.NUMBER: 'number'>, entity_type=<EntityType.CONFIG: 'config'>, cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='detection_distance_max', fallback_name='Maximum range', attribute_name='detection_distance_max', reporting_config=None, min=0.75, max=9.0, step=0.75, unit=<UnitOfLength.METERS: 'm'>, mode=None, multiplier=None, device_class=<SensorDeviceClass.DISTANCE: 'distance'>)}]
Failed to create platform entity: <class 'zha.application.platforms.sensor.Sensor'> [args=('e4:98:bb:43:00:8d:58:22-1', [<zha.zigbee.cluster_handlers.manufacturerspecific.TuyaClusterHandler object at 0x7f31abd456d0>], <zha.zigbee.endpoint.Endpoint object at 0x7f31abd45490>, <CustomDeviceV2 model='TS0601' manuf='_TZE200_ya4ft0w4' nwk=0xD4CC ieee=e4:98:bb:43:00:8d:58:22 is_initialized=True> - quirk_applied: True - quirk_or_device_class: zigpy.quirks.v2.CustomDeviceV2 - quirk_id: None), kwargs={'entity_metadata': ZCLSensorMetadata(entity_platform=<EntityPlatform.SENSOR: 'sensor'>, entity_type=<EntityType.STANDARD: 'standard'>, cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='distance', fallback_name='Target distance', attribute_name='distance', reporting_config=None, divisor=10, multiplier=1, unit=<UnitOfLength.METERS: 'm'>, device_class=<SensorDeviceClass.DISTANCE: 'distance'>, state_class=<SensorStateClass.MEASUREMENT: 'measurement'>)}]
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/zha/application/gateway.py", line 335, in create_platform_entities
    platform_entity = platform_entity_class.create_platform_entity(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zha/application/platforms/number/__init__.py", line 229, in create_platform_entity
    return cls(unique_id, cluster_handlers, endpoint, device, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zha/application/platforms/number/__init__.py", line 242, in __init__
    super().__init__(unique_id, cluster_handlers, endpoint, device, **kwargs)
  File "/usr/local/lib/python3.12/site-packages/zha/application/platforms/__init__.py", line 297, in __init__
    self._init_from_quirks_metadata(entity_metadata)
  File "/usr/local/lib/python3.12/site-packages/zha/application/platforms/number/__init__.py", line 269, in _init_from_quirks_metadata
    self._attr_native_unit_of_measurement = validate_unit(
                                            ^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/zha/units.py", line 167, in validate_unit
    return UNITS_OF_MEASURE[type(external_unit).__name__](external_unit.value)
           ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
KeyError: 'UnitOfLength'`

Rabman416, 2024-12-14:
Glad you understand why. I sure don’t. 

Thanks to you both for helping out!

Itay1787, 2024-12-14:
> Please make sure you are on the latest HA. I'm seeing python 3.12 in your logs and I would expect 3.13.

Yup😅 you are correct I will update HA and check again

Itay1787, 2024-12-14:
> Please make sure you are on the latest HA. I'm seeing python 3.12 in your logs and I would expect 3.13.

Ok, I updated HA and now I have all the controls. but Illuminance is not working as expected it only shows 1 lx. 
and control for the Maximum range doesn't work. I won't let me change it, after I tried to make a change to this setting it got back to the max of 150m (or setting 9) 
and Minimum range also seems to not working as expected if I make a change it tries to apply it but goes back to 0m but I don't know if it is set for 0m on the device or if it is HA reading problem 
I don't see anything in the log about this.

Rabman416, 2024-12-14:
Maybe try switching the stuff in the mfgr app

Itay1787, 2024-12-14:
> Maybe try switching the stuff in the mfgr app

Sorry, in what app?

prairiesnpr, 2024-12-14:
> Ok, I updated HA and now I have all the controls. but Illuminance is not working as expected it only shows 1 lx. and control for the Maximum range doesn't work. I won't let me change it, after I [...]

There are a few options here, if it's working for @Rabman416, then it's possible you have a defective device, does it work when paired to a Tuya hub? It's also possible that the devices have different firmware and don't use the same DPs. The quirk above is based on the z2m converter, and should work the same, I did see that there are widely reported issues with the illuminance measurement for this sensor though.

You can enable debug logging in ZHA, note the start and end timestamps and the device ieee and nwk, run it through all of its paces, then attach the logs and we can see if there is anything logged.

Rabman416, 2024-12-14:
Sorry forget about my comment about app. I am also trying to get the wifi version working. I am using the Smart Life app for it and the TUYA integration. 

Maximum range works but looks like it needs to be divide by 100.

Sliders do change and stay where I set them.

Itay1787, 2024-12-14:
> There are a few options here, if it's working for @Rabman416, then it's possible you have a defective device, does it work when paired to a Tuya hub? It's also possible that the devices have [...]

Yes, it is working as I expected in the Tuya app with a Tuya hub, the illuminance, and all the controls.
Yup, I also saw reports about the illuminance measurement but because in the Tuya app, it was working I thought that maybe they fixed it with this version of the sensor.

Itay1787, 2024-12-14:
> Sliders do change and stay where I set them.

all the sliders??
and illuminance with ZHA is working? 
can you share what version of the sensor you have?

prairiesnpr, 2024-12-14:
> Maximum range works but looks like it needs to be divide by 100.

Add `multiplier=0.1,` to that `.tuya_number`

Rabman416, 2024-12-14:
> multiplier=0.1,

My mistake on this. It was the Wifi version, not the Zigbee.

Itay1787, 2024-12-14:
> All the sliders work. The illumination sensor shows only 1 or 2. Some other posts indicated the hole in the sensor needed to be enlarged. Never tried it though.
> 
> My model is a Loginovo ZY-M100-24G

Ok, then the illumination sensor acts the same as mine. but in the Tuya app, it works like normal so it is a ZHA problem.

I don't get why then the Maximum range sliders don't work in mine.
also, the Minimum range acts up it stays sometimes and sometimes it goes to 0 
but Minimum range and Minimum range changes do log, in the logbook like the other control upon changing

we have the some model

prairiesnpr, 2024-12-14:
> Ok, then the illumination sensor acts the same as mine. but in the Tuya app, it works like normal so it is a ZHA problem.

I'm thinking that it's not scaled correctly, there isn't any scaling in z2m either, but looking at the other Tuya quirk, they use `10000 * math.log10(x) + 1 if x != 0 else 0`

You might try the following and see if it looks better, otherwise you would need to reverse engineer the scaling.

```python
import math
from zigpy.quirks.v2 import EntityType
from zigpy.quirks.v2.homeassistant import UnitOfLength, UnitOfTime
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
import zigpy.types as t
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement, OccupancySensing

from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.builder import TuyaQuirkBuilder

class TuyaIlluminanceCluster(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya Illuminance cluster."""

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

(
    TuyaQuirkBuilder("_TZE200_ya4ft0w4", "TS0601")
    .tuya_dp(
        dp_id=1,
        ep_attribute=TuyaOccupancySensing.ep_attribute,
        attribute_name=OccupancySensing.AttributeDefs.occupancy.name,
        converter=lambda x: True if x in (1, 2) else False,
    )
    .adds(TuyaOccupancySensing)
    .tuya_number(
        dp_id=2,
        attribute_name="move_sensitivity",
        type=t.uint16_t,
        min_value=0,
        max_value=10,
        step=1,
        translation_key="move_sensitivity",
        fallback_name="Motion sensitivity",
    )
    .tuya_number(
        dp_id=3,
        attribute_name="detection_distance_min",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.METERS,
        min_value=0,
        max_value=8.25,
        step=0.75,
        translation_key="detection_distance_min",
        fallback_name="Minimum range",
    )
    .tuya_number(
        dp_id=4,
        attribute_name="detection_distance_max",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.METERS,
        min_value=0.75,
        max_value=9.0,
        step=0.75,
        translation_key="detection_distance_max",
        fallback_name="Maximum range",
    )
    .tuya_sensor(
        dp_id=9,
        attribute_name="distance",
        type=t.uint16_t,
        divisor=10,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.METERS,
        entity_type=EntityType.STANDARD,
        translation_key="distance",
        fallback_name="Target distance",
    )
    .tuya_switch(
        dp_id=101,
        attribute_name="find_switch",
        entity_type=EntityType.STANDARD,
        translation_key="find_switch",
        fallback_name="Distance switch",
    )
    .tuya_number(
        dp_id=102,
        attribute_name="presence_sensitivity",
        type=t.uint16_t,
        min_value=0,
        max_value=10,
        step=1,
        translation_key="presence_sensitivity",
        fallback_name="Presence sensitivity",
    )
    .tuya_dp(
        dp_id=103,
        ep_attribute=TuyaIlluminanceCluster.ep_attribute,
        attribute_name=TuyaIlluminanceCluster.AttributeDefs.measured_value.name,
        converter=lambda x: 10000 * math.log10(x) + 1 if x != 0 else 0,
    )
    .adds(TuyaIlluminanceCluster)
    .tuya_number(
        dp_id=105,
        attribute_name="presence_timeout",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DURATION,
        unit=UnitOfTime.SECONDS,
        min_value=1,
        max_value=15000,
        step=1,
        translation_key="presence_timeout",
        fallback_name="Fade time",
    )
    .skip_configuration()
    .add_to_registry()
)
```

Rabman416, 2024-12-14:
OK, I’ll have to try it out this evening. Thanks for all your hard work.

Rabman416, 2024-12-14:
Well it is presenting a higher number now....no clue as to whether it is correct or not. Better than 1-2 lx though.

Itay1787, 2024-12-17:
> I'm thinking that it's not scaled correctly, there isn't any scaling in z2m either, but looking at the other Tuya quirk, they use `10000 * math.log10(x) + 1 if x != 0 else 0`
>
> You might try the [...]

Hi, sorry for the delay.
Everything works for me, and the light sensor works properly like in Tuya's app.
except for the min/max, it still doesn't work for me. I can't change them and have them stay in place, and I don't think they change anything at all because it doesn't seem like it was recorded in the logbook - that's why I don't think it was sent to the device.

In addition, is it possible to add the state of "occupancy" as there is in Tuya app

Tuya's app has: cleared, moving, and exits

Is it possible to add? It doesn't matter that much but it's always nice to have all the options.

When the device does not respond, it seems that things simply remain from the last reading from the device
There is an option to change it to "Unavailable" like when other devices in HA don't respond

Thanks for all your help!!

Itay1787, 2024-12-18:
Hi, I just got an error from ZHA and It can't get started.

I didn't change anything. it just happened 

`Logger: homeassistant.config_entries
Source: config_entries.py:640
First occurred: 18:00:11 (6 occurrences)
Last logged: 18:03:50

Error setting up entry Zigbee-Hub for zha
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 640, in __async_setup_with_context
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/__init__.py", line 132, in async_setup_entry
    async with radio_mgr.connect_zigpy_app() as app:
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/local/lib/python3.13/contextlib.py", line 214, in __aenter__
    return await anext(self.gen)
           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/zha/radio_manager.py", line 182, in connect_zigpy_app
    app = await self.radio_type.controller.new(
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        app_config, auto_form=False, start_radio=False
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/site-packages/zigpy/application.py", line 239, in new
    await app._load_db()
  File "/usr/local/lib/python3.13/site-packages/zigpy/application.py", line 111, in _load_db
    await self._dblistener.load()
  File "/usr/local/lib/python3.13/site-packages/zigpy/appdb.py", line 684, in load
    device = zigpy.quirks.get_device(device)
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/__init__.py", line 43, in get_device
    return _DEVICE_REGISTRY.get_device(device)
           ~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/zigpy/quirks/registry.py", line 125, in get_device
    raise MultipleQuirksMatchException(
        f"Multiple matches found for device {device}: {matches}"
    )
zigpy.exceptions.MultipleQuirksMatchException: Multiple matches found for device <Device model='TS0601' manuf='_TZE200_ya4ft0w4' nwk=0x7E88 ieee=e4:98:bb:43:00:8d:58:22 is_initialized=True>: [QuirksV2RegistryEntry(quirk_file=PosixPath('/usr/local/lib/python3.13/site-packages/zhaquirks/tuya/builder/__init__.py'), quirk_file_line=75, manufacturer_model_metadata=(ManufacturerModelMetadata(manufacturer='_TZE200_ya4ft0w4', model='TS0601'),), friendly_name=None, device_alerts=(), filters=(), custom_device_class=None, device_node_descriptor=None, skip_device_configuration=True, adds_metadata=(AddsMetadata(cluster=<class 'tuya_motion_sensor_ya4ft0w4.TuyaOccupancySensing'>, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, constant_attributes=frozendict.frozendict({})), AddsMetadata(cluster=<class 'tuya_motion_sensor_ya4ft0w4.TuyaIlluminanceCluster'>, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, constant_attributes=frozendict.frozendict({}))), removes_metadata=(), replaces_metadata=(ReplacesMetadata(remove=RemovesMetadata(cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>), add=AddsMetadata(cluster=<class 'zhaquirks.tuya.builder.TuyaQuirkBuilder.add_to_registry.<locals>.TuyaReplacementCluster'>, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, constant_attributes=frozendict.frozendict({}))),), replaces_cluster_occurrences_metadata=(), entity_metadata=(NumberMetadata(entity_platform=<EntityPlatform.NUMBER: 'number'>, entity_type=<EntityType.CONFIG: 'config'>, cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='move_sensitivity', fallback_name='Motion sensitivity', attribute_name='move_sensitivity', reporting_config=None, min=0, max=10, step=1, unit=None, mode=None, multiplier=None, device_class=None), NumberMetadata(entity_platform=<EntityPlatform.NUMBER: 'number'>, entity_type=<EntityType.CONFIG: 'config'>, cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='detection_distance_min', fallback_name='Minimum range', attribute_name='detection_distance_min', reporting_config=None, min=0, max=8.25, step=0.75, unit=<UnitOfLength.METERS: 'm'>, mode=None, multiplier=None, device_class=<SensorDeviceClass.DISTANCE: 'distance'>), NumberMetadata(entity_platform=<EntityPlatform.NUMBER: 'number'>, entity_type=<EntityType.CONFIG: 'config'>, cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='detection_distance_max', fallback_name='Maximum range', attribute_name='detection_distance_max', reporting_config=None, min=0.75, max=9.0, step=0.75, unit=<UnitOfLength.METERS: 'm'>, mode=None, multiplier=None, device_class=<SensorDeviceClass.DISTANCE: 'distance'>), ZCLSensorMetadata(entity_platform=<EntityPlatform.SENSOR: 'sensor'>, entity_type=<EntityType.STANDARD: 'standard'>, cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='distance', fallback_name='Target distance', attribute_name='distance', reporting_config=None, divisor=10, multiplier=1, unit=<UnitOfLength.METERS: 'm'>, device_class=<SensorDeviceClass.DISTANCE: 'distance'>, state_class=<SensorStateClass.MEASUREMENT: 'measurement'>), SwitchMetadata(entity_platform=<EntityPlatform.SWITCH: 'switch'>, entity_ [... cut ...] type=<EntityType.CONFIG: 'config'>, cluster_id=61184, endpoint_id=1, cluster_type=<ClusterType.Server: 0>, initially_disabled=False, attribute_initialized_from_cache=True, translation_key='presence_timeout', fallback_name='Fade time', attribute_name='presence_timeout', reporting_config=None, min=1, max=15000, step=1, unit=<UnitOfTime.SECONDS: 's'>, mode=None, multiplier=None, device_class=<SensorDeviceClass.DURATION: 'duration'>)), device_automation_triggers_metadata=frozendict.frozendict({}))]`

prairiesnpr, 2024-12-18:
`zigpy.exceptions.MultipleQuirksMatchException`

You have multiple quirks, make sure you don't have two files or the quirk defined twice in the same file.

Itay1787, 2024-12-18:
> `zigpy.exceptions.MultipleQuirksMatchException`
> 
> You have multiple quirks, make sure you don't have two files or the quirk defined twice in the same file.

I do have 2 files of quirks one for this device and one for another

prairiesnpr, 2024-12-18:
> I do have 2 files of quirks one for this device and one for another

Make sure this device isn't in both files, it's erroring on `TZE200_ya4ft0w4 Tuya`

Itay1787, 2024-12-18:
> > I do have 2 files of quirks one for this device and one for another
> 
> Make sure this device isn't in both files, it's erroring on `TZE200_ya4ft0w4`

Nope.` TZE200_ya4ft0w4 `is only in one file...

Itay1787, 2024-12-18:
> This is a restart or a reload, there is a known issue when reloading.

Reload...
is there a workaround?

prairiesnpr, 2024-12-18:
> > This is a restart or a reload, there is a known issue when reloading.
> 
> Reload... is there a workaround?

Should be able to add this to the top of the quirk, right below the last import. 

```python
from zigpy.quirks import DEVICE_REGISTRY

del DEVICE_REGISTRY._registry_v2[("_TZE200_ya4ft0w4", "TS0601")]

```

Itay1787, 2024-12-18:
> > 
> 
> Should be able to add this to the top of the quirk, right below the last import.
> 
> ```python
> from zigpy.quirks import DEVICE_REGISTRY
> 
> del DEVICE_REGISTRY._registry_v2[("_TZE200_ya4ft0w4", "TS0601")]
> ```

I added this and now the quirk doesn't work...

Itay1787, 2024-12-18:
> > > 
> > 
> > 
> > Should be able to add this to the top of the quirk, right below the last import.
> > ```python
> > from zigpy.quirks import DEVICE_REGISTRY
> > 
> > del DEVICE_REGISTRY._registry_v2[("_TZE200_ya4ft0w4", "TS0601")]
> > ```
> 
> I added this and now the quirk doesn't work...
> 
> log
> 
> `Logger: zhaquirks Source: /usr/local/lib/python3.13/site-packages/zhaquirks/**init**.py:481 First occurred: 21:19:50 (1 occurrences) Last logged: 21:19:50
> 
> Unexpected exception importing custom quirk 'tuya_motion_sensor_ya4ft0w4' Traceback (most recent call last): File "/usr/local/lib/python3.13/site-packages/zhaquirks/**init**.py", line 479, in setup spec.loader.exec_module(module) ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^ File "", line 1022, in exec_module File "", line 488, in _call_with_frames_removed File "/config/custom_zha_quirks/tuya_motion_sensor_ya4ft0w4.py", line 12, in del DEVICE_REGISTRY._registry_v2[("_TZE200_ya4ft0w4", "TS0601")] ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ KeyError: ('_TZE200_ya4ft0w4', 'TS0601')`

Fixed... needed to load without the del first then uncomment it.

Rabman416, 2024-12-21:
Mine started to fail on load and by following the above del tip, it worked. 

I put it is the other quirk ts0602_radar.py file. Not the new one I added for this presence sensor.

Thanks again for all your help!

closed 2024-12-23 by TheJulianJES

Rabman416, 2025-01-30:
UPDATE: The maximum range does not seem to be adjustable. When you drag the slider, it always snaps back to 9. This happens repeatedly. The other sliders are adjustable, but whether they actually change settings, or have any effect, hard to say.

DelBierzo, 2025-05-02:
> UPDATE: The maximum range does not seem to be adjustable. When you drag the slider, it always snaps back to 9. This happens repeatedly. The other sliders are adjustable, but whether they actually change settings, or have any effect, hard to say.

In "dp_id=3" and "dp_id=4"

add:

multiplier=0.01,

`
```
    .tuya_number(
        dp_id=3,
        attribute_name="detection_distance_min",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.METERS,
        min_value=0,
        max_value=8.25,
        step=0.75,
        multiplier=0.01,
        translation_key="detection_distance_min",
        fallback_name="Minimum range",
    )
    .tuya_number(
        dp_id=4,
        attribute_name="detection_distance_max",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.METERS,
        min_value=0.75,
        max_value=9.0,
        step=0.75,
        multiplier=0.01,
        translation_key="detection_distance_max",
        fallback_name="Maximum range",
    )

DelBierzo, 2025-05-02:
> UPDATE: The maximum range does not seem to be adjustable. When you drag the slider, it always snaps back to 9. This happens repeatedly. The other sliders are adjustable, but whether they actually change settings, or have any effect, hard to say.

add:  "multiplier=0.01," in "dp_id=4" or copy the entire code!

```
import math
from zigpy.quirks.v2 import EntityType
from zigpy.quirks.v2.homeassistant import UnitOfLength, UnitOfTime
from zigpy.quirks.v2.homeassistant.sensor import SensorDeviceClass, SensorStateClass
import zigpy.types as t
from zigpy.zcl.clusters.measurement import IlluminanceMeasurement, OccupancySensing

from zhaquirks.tuya import TuyaLocalCluster
from zhaquirks.tuya.builder import TuyaQuirkBuilder

class TuyaIlluminanceCluster(IlluminanceMeasurement, TuyaLocalCluster):
    """Tuya Illuminance cluster."""

class TuyaOccupancySensing(OccupancySensing, TuyaLocalCluster):
    """Tuya local OccupancySensing cluster."""

(
    TuyaQuirkBuilder("_TZE200_ya4ft0w4", "TS0601")
    .tuya_dp(
        dp_id=1,
        ep_attribute=TuyaOccupancySensing.ep_attribute,
        attribute_name=OccupancySensing.AttributeDefs.occupancy.name,
        converter=lambda x: True if x in (1, 2) else False,
    )
    .adds(TuyaOccupancySensing)
    .tuya_number(
        dp_id=2,
        attribute_name="move_sensitivity",
        type=t.uint16_t,
        min_value=0,
        max_value=10,
        step=1,
        translation_key="move_sensitivity",
        fallback_name="Motion sensitivity",
    )
    .tuya_number(
        dp_id=3,
        attribute_name="detection_distance_min",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.METERS,
        min_value=0,
        max_value=8.25,
        step=0.75,
        multiplier=0.01,
        translation_key="detection_distance_min",
        fallback_name="Minimum range",
    )
    .tuya_number(
        dp_id=4,
        attribute_name="detection_distance_max",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.METERS,
        min_value=0.75,
        max_value=9.0,
        step=0.75,
        multiplier=0.01,
        translation_key="detection_distance_max",
        fallback_name="Maximum range",
    )
    .tuya_sensor(
        dp_id=9,
        attribute_name="distance",
        type=t.uint16_t,
        divisor=10,
        state_class=SensorStateClass.MEASUREMENT,
        device_class=SensorDeviceClass.DISTANCE,
        unit=UnitOfLength.METERS,
        entity_type=EntityType.STANDARD,
        translation_key="distance",
        fallback_name="Target distance",
    )
    .tuya_switch(
        dp_id=101,
        attribute_name="find_switch",
        entity_type=EntityType.STANDARD,
        translation_key="find_switch",
        fallback_name="Distance switch",
    )
    .tuya_number(
        dp_id=102,
        attribute_name="presence_sensitivity",
        type=t.uint16_t,
        min_value=0,
        max_value=10,
        step=1,
        translation_key="presence_sensitivity",
        fallback_name="Presence sensitivity",
    )
    .tuya_dp(
        dp_id=103,
        ep_attribute=TuyaIlluminanceCluster.ep_attribute,
        attribute_name=TuyaIlluminanceCluster.AttributeDefs.measured_value.name,
        converter=lambda x: 10000 * math.log10(x) + 1 if x != 0 else 0,
    )
    .adds(TuyaIlluminanceCluster)
    .tuya_number(
        dp_id=105,
        attribute_name="presence_timeout",
        type=t.uint16_t,
        device_class=SensorDeviceClass.DURATION,
        unit=UnitOfTime.SECONDS,
        min_value=1,
        max_value=15000,
        step=1,
        translation_key="fading_time",
        fallback_name="Fading time",
    )
    .skip_configuration()
    .add_to_registry()
)
```

TheJulianJES, 2025-05-03:
The quirk is included since HA 2025.1.0. Are the issues with that?
If you're still using custom quirks for this device, you should delete them and let the built-in quirks take over.

TheJulianJES, 2025-05-19:
Please open a new issue, include your model, and just copy/paste that text + screenshot @krystian0201

DelBierzo, 2025-05-19:
As I said before, must be added, "multiplier=0.01," in "dp_id=4" that's all.

The published version is the only flaw it has.
With that change it works perfectly, that's why I still use the quirk.